cedricp/ddt4all #696: 32 pin r-link 2 connector pins diagram

myshuker, 2022-07-05:
Hej all
I need  32 pin r-link 2 connector pins diagram for renault Scenic 4 or Megane 4 
https://ibb.co/k52jqYJ

https://ibb.co/7rgVpgD

MellotronScratch, 2022-07-21:
Here you go. This is from a Kadjar, should be the same pinout.

Pin | Signal | Signal label
-- | -- | --
1 | 107W | DIAGNOSTIC SOCKET CANH MULTIMEDIA SIGNAL
2 | 107X | DIAGNOSTIC SOCKET CANL MULTIMEDIA SIGNAL
3 | 107G | HANDS FREE TELEPHONE MICROPHONE +
4 | 107F | MULTIMEDIA MICROPHONE SIGNAL
6 | 137H | CAN H SIGNAL
7 | 137G | CAN L SIGNAL
11 | 34EC | RIGHT HAND LINE IN + SIGNAL
12 | 34EE | LEFT HAND LINE IN + SIGNAL
13 | 34EM | LINE IN SIGNAL
20 | 107J | TELEPHONE MICROPHONE SCREENING
21 | 34KL | REAR CAMERA VIDEO SIGNAL
22 | 34KJ | - REAR CAMERA
23 | 34KK | + REAR CAMERA
24 | TB34 | REAR CAMERA SHIELDED CABLE
27 | TB37 | LINE IN SHIELDED CABLE
28 | 34ET | - LINE IN

endrjuk, 2023-02-11:
Hi, I guess that 12 pins connector (on left in the photo) is just power and speakers, but do you also have the 24 pins connector electric diagram for R-Link2? 
[image]

MellotronScratch, 2023-02-11:
@endrjuk This is all information I have. Your connector has more connections. You are right about the connector on the left on the photo. It's for power and speakers.

| Pin | Signal | Signal label               |
|-----|--------|----------------------------|
| 6   | 34AS   | RADIO SATELLITE IN0 SIGNAL |
| 7   | 34AT   | RADIO SATELLITE IN1 SIGNAL |
| 8   | 34AU   | RADIO SATELLITE IN2 SIGNAL |
| 9   | 34AP   | RADIO SATELLITE SIGNAL L0  |
| 18  | 34AQ   | RADIO SATELLITE SIGNAL L1  |
| 19  | 34AR   | RADIO SATELLITE SIGNAL L2  |
| 20  | 107V   | VOICE RECOGNITION          |
| 21  | 107O   | MASS VOICE RECOGNITION     |

endrjuk, 2023-02-11:
Thank you! I'm in the process of modifying the wiring harness to rlink-2 in Megane IV and any informations are helpful. Do you have any information about the multi-pin plug connected to the front panel? It's a Megane IV, so it must be something related to the air conditioning control and buttons for emergency lights/central locking. It's hard to get any electric diagrams in these generations of Renault cars...
[image]

EddyG70, 2023-03-11:
Can anyone tell me what all the connections are for from left to right? Please your help would be appreciated 😎

MellotronScratch, 2023-03-11:
From left to right:
Speakers, 12v+ and ground.
GPS Antenna (blue)
Telematic module (Brown USB)
USB Ports (Black USB)
Display (Blue USB)
[See my earlier post](https://github.com/cedricp/ddt4all/issues/696#issuecomment-1191154034) (32Pin white connector)
[Steering wheel controls](https://github.com/cedricp/ddt4all/issues/696#issuecomment-1426750362) (24Pin white connector)
FM2 antenna (Yellowish fakra connector)
FM1 antenna (White fakra connector
DAB+ (Brown fakra connector)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 not bad info rappot it to discutions!!!

closed 2023-06-02 by Furtif