cedricp/ddt4all #697: Problem with r link after using developer mode

saadbm1, 2022-07-09:
[image]

shrlnm, 2022-07-10:
What is your VIN?

saadbm1, 2022-07-10:
> What is your VIN?

how would that help in any way ?

AndyIreland, 2022-07-10:
I should imagine VIN number would match up as to what features you have in your car and the rlink can be then configured as to what and not features your particular car has got

saadbm1, 2022-07-10:
it’s more about if it’s possible to configure it or should i just give up, and if it’s possible then how ?

AndyIreland, 2022-07-10:
what is it an rlink2 is it?

saadbm1, 2022-07-10:
yes and the update version is 2.2.19.300

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif