cedricp/ddt4all #700: Automatically switch off my Trafic III dipped headlights by locking my vehicle.

f4gqk, 2022-07-10:
Hello,

I drive with my dipped headlights during the day with a renault trafic. On my comodo, I leave on the crossing position. When I get out of the vehicle, I have a beep sound, normal. I would like it to work like on my renault scenic III that when I get out of my car and lock it with my hands-free card, the lights go out automatically.

In DDT4ALL, where is this option?

Thanks

Sebastien

AndyIreland, 2022-07-10:
If you have the right EMM you could choose either scandinav running lights or swiss cf (or both even) 

I think Scandinav means headlights are on all the time as long as the ignition is on (or engine running) and Swiss cf I think keeps the headlights on until you open up the drivers door after a certain time limit after turning the ignition off . 

Better make sure that the SW version of yours is over SW13 I think it is because on a clio mk4 if you change lighting on SW13 and under I thing it can break the lighting ecu and you have to get it reprogrammed 
Good luck

f4gqk, 2022-07-12:
Thanks for response.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836