cedricp/ddt4all #703: I cannot clear 4 DTC's from my Body Control Module

AndyIreland, 2022-07-17:
**I have 4 DTC Errors on my BCM95 UCH on my Renault Zoe EV 2013 when I use PYREN. So when I went to clear them it did not show any errors ... but I chose number 2 anyway to clear - then selected 'yes' and I got the proper 'rsp54' or whatever it is (sorry doing this from memory) . 

Then chose 15 'scan all errors and still showed 4 errors on the BCM UCH again**

From Pyren ECU Log:
17:40:57.340;745;228006;7F2212;NR: SubFunction Not Supported
17:40:57.442;745;19023B;59 02 B9 95 7C 14 29 95 7D 14 29 95 7E 14 29 FF FF FF FF FF 
17:40:57.466;745;228004;7F2212;NR: SubFunction Not Supported
17:40:57.484;745;228005;7F2212;NR: SubFunction Not Supported
17:40:57.516;745;228002;7F2212;NR: SubFunction Not Supported
17:40:57.534;745;228003;7F2212;NR: SubFunction Not Supported
17:40:57.558;745;228000;7F2212;NR: SubFunction Not Supported
17:40:57.577;745;228001;7F2212;NR: SubFunction Not Supported
17:40:57.658;745;21AB;61 AB 42 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 74 FF FF FF 
17:40:57.691;745;2101;61 01 0F 0F FF FF FF 
17:40:57.794;745;210B;61 0B 00 00 6E 30 00 00 22 05 22 05 24 00 00 05 FF FF FF FF 
17:40:57.818;745;2136;61 36 80 
17:40:57.875;745;21FE;61 FE 37 30 31 31 52 04 30 30 31 34 37 30 37 52 14 0E 20 10 20 10 02 01 00 88 FF 
17:41:20.816;745;19023B;59 02 B9 95 7C 14 29 95 7D 14 29 95 7E 14 29 FF FF FF FF FF 
17:41:26.234;745;14FFFFFF;54 

**So, I decided to fire up DDT4ALL and done a scan of all ECU's on the CAN then selected BCM95 ECU and then 'clear DTC codes'  but get this error:**

[image]

**Here is the DDT4ALL ECU Log:**

17:56:25.465;745;10CF;7F1012;NR: SubFunction Not Supported
17:56:26.091;745;3E01;7E 
17:56:27.692;745;3E01;7E 
17:56:29.090;745;3E01;7E 
17:56:30.622;745;3E01;7E 
17:56:32.089;745;10C0;50 C0 
17:56:32.340;745;3E01;7E 
17:56:33.616;745;3E01;7E 
17:56:35.091;745;3E01;7E 
17:56:36.591;745;3E01;7E 
17:56:38.116;745;3E01;7E 
17:56:39.940;745;3E01;7E 
17:56:41.141;745;3E01;7E 
17:56:42.590;745;3E01;7E 
17:56:44.092;745;3E01;7E 
17:56:45.590;745;3E01;7E 
17:56:47.090;745;3E01;7E 
17:56:48.625;745;3E01;7E 
17:56:50.095;745;3E01;7E 
17:56:51.592;745;3E01;7E 
17:56:53.115;745;3E01;7E 
17:56:54.615;745;3E01;7E 
17:56:56.092;745;3E01;7E 
17:56:57.590;745;3E01;7E 
17:56:59.091;745;3E01;7E 
17:57:00.622;745;3E01;7E 
17:57:02.093;745;3E01;7E 
17:57:03.592;745;3E01;7E 
17:57:05.116;745;3E01;7E 
17:57:06.631;745;3E01;7E 
17:57:08.105;745;3E01;7E 
17:57:09.590;745;3E01;7E 
17:57:11.120;745;3E01;7E 
17:57:12.591;745;3E01;7E 
17:57:14.090;745;3E01;7E 
17:57:15.590;745;3E01;7E 
17:57:17.091;745;3E01;7E 
17:57:18.615;745;3E01;7E 
17:57:20.092;745;3E01;7E 
17:57:21.591;745;3E01;7E 
17:57:23.091;745;3E01;7E 
17:57:24.591;745;3E01;7E 
17:57:26.109;745;3E01;7E 
17:57:27.591;745;3E01;7E 
17:57:29.091;745;3E01;7E 
17:57:30.592;745;3E01;7E 
17:57:32.090;745;3E01;7E 
17:57:33.616;745;3E01;7E 
17:57:35.092;745;3E01;7E 
17:57:36.615;745;3E01;7E 
17:57:38.432;745;3E01;7E 
17:57:40.165;745;3E01;7E 
17:57:41.546;745;3E01;7E 
17:57:42.690;745;3E01;7E 

**I am using Python 38**

Thankyou

KarelSvo, 2022-07-17:
Did you activate the dump function?
Then give me your Pyren program folder so I can look at your problem in demo mode.

AndyIreland, 2022-07-17:
[1658075582_BCM95_SW25_V1.1_20150429T131442.txt](https://github.com/cedricp/ddt4all/files/9128528/1658075582_BCM95_SW25_V1.1_20150429T131442.txt)

here is the dump file Thanks @KarelSvo

KarelSvo, 2022-07-24:
My answer was

> Pyren program folder

AndyIreland, 2022-07-24:
I have 2

AndyIreland, 2022-07-24:
i have 2 folders in there:

[pyren.zip](https://github.com/cedricp/ddt4all/files/9176026/pyren.zip)
[pyren09p.zip](https://github.com/cedricp/ddt4all/files/9176027/pyren09p.zip)

KarelSvo, 2022-07-31:
ddt4all shows all errors that are possible for this ecu.
The clip database in Pyren filters the error codes. Only error codes relevant to your vehicle are displayed.
You can see the filter list with 'mod_ecu.py ECUID LANGUAGE'.

`python.exe  mod_ecu.py 11176 GB
CLIP DB : ..
DDT  DB : ./DDT2000data.zip
Loading language 
[XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX] 100% Done
Loading optimyzer
Loading defaults
Loading parameters
Loading states
Loading identifications
Loading commands
Loading mnemonics

Defaults

D335[5A60] UNDER-INFLATED
D336[5A72] FRONT LEFT-HAND WHEEL FLAT
D337[5A73] FRONT RIGHT-HAND WHEEL FLAT
D338[5A74] REAR RIGHT-HAND WHEEL FLAT
D339[5A75] REAR LEFT-HAND WHEEL FLAT
D158[5A80] RECOMMENDED PRESSURE SETTING
D006[5A82] FRONT LEFT-HAND WHEEL VALVE SENSOR
D007[5A83] FRONT RIGHT-HAND WHEEL VALVE SENSOR
D008[5A84] REAR RIGHT-HAND WHEEL VALVE SENSOR
D009[5A85] REAR LEFT-HAND WHEEL VALVE SENSOR
D065[9510] ELECTRIC WINDOW LIFT AUTHORISATION CONNECTION
D195[9520] RAIN/LIGHT SENSOR
D172[9530] UCH
D048[9533] PROTECTION CODE BETWEEN UCH AND EPS
D020[953B] FRONT INTERIOR AERIAL CIRCUIT
D022[953C] CENTRE INTERIOR AERIAL CIRCUIT
D021[953D] REAR INTERIOR AERIAL CIRCUIT
D297[953E] FRONT LEFT-HAND EXTERNAL AERIAL CIRCUIT
D298[953F] FRONT RIGHT-HAND EXTERNAL AERIAL CIRCUIT
D148[9540] REAR EXTERIOR AERIAL CIRCUIT
D170[9542] UCH
D029[9550] STEERING LOCK CIRCUIT
D002[9551] STEERING LOCK
D160[9552] STEERING COLUMN LOCK PROTECTION
D161[9553] STEERING COLUMN LOCK +APC CIRCUIT
D147[9554] STEERING COLUMN LOCK SAFETY CIRCUIT
D171[9556] UCH
D376[9557] COLUMN STEERING LOCK
D199[9563] BRAKE PEDAL CONTACT CIRCUIT
D200[9564] BRAKE PEDAL CONTACT CIRCUIT
D201[9565] BRAKE PEDAL CONTACT CIRCUIT
D202[9566] BRAKE PEDAL CONTACT CIRCUIT
D192[956D] HEATED REAR SCREEN CIRCUIT
D191[9570] STOP LIGHTS CIRCUIT
D151[9571] UCH 1 SUPPLY CIRCUIT
D132[9572] INDICATOR(S) SUPPLY CIRCUIT
D305[9573] STOP LIGHTS SUPPLY CIRCUIT
D175[9574] DOOR LOCKING CIRCUIT
D176[9577] BOOT OPENING CONTROL
D188[9579] REAR FOG LIGHTS CIRCUIT
D043[957F] RCH 1 RELAY CONTROL CIRCUIT
D044[9580] RCH 2 RELAY CONTROL CIRCUIT
D162[9581] HEATED REAR SCREEN RELAY CONTROL
D001[9582] UCH
D262[9585] UCH
D263[9586] UCH
D010[9590] COMBINED STALK UNDER STEERING WHEEL CIRCUIT
D316[9591] HORN
D185[95A0] REAR SCREEN WIPER CIRCUIT
D324[95B0] GLOVEBOX/CHILDPROOF LOCK CIRCUIT
D166[95C1] BATTERY DISCONNECTION UNIT SAFETY CIRCUIT
D012[95D0] RIGHT INDICATOR CIRCUIT
D013[95D1] LEFT INDICATOR CIRCUIT
D198[95D3] RAIN/LIGHT SENSOR
D196[95D4] RAIN/LIGHT SENSOR
D197[95D5] RAIN/LIGHT SENSOR
D193[95D6] RIGHT INDICATOR CIRCUIT
D194[95D7] LEFT INDICATOR CIRCUIT
D005[95E3] CARD
D167[95E5] CARD
D168[95E7] CARD
D169[95E9] CARD
D189[95F0] CARD READER CIRCUIT
D190[95F1] CARD READER SUPPLY CIRCUIT
`
17:40:57.442;745;19023B;59 02 B9 95 7C 14 29 95 7D 14 29 95 7E 14 29 FF FF FF FF FF
Your 3 code found 
[957C] 14 29
[957D]14 29
[957E] 14 29
are not in the filter list. These are therefore not relevant to your vehicle.
If the error status is PRESENT. Can't you erase these "ghost bugs" either.
The fault specification 14 means open circuit. So a part that is not connected causes the error message. Since this part is not intended for your vehicle, the Clip ECU database ignores this error message.

[image]

toll-n, 2023-04-15:
@KarelSvo  The screenshot of Builder Xml shown above is interesting.
Is it part of pyren3 or ddt4all?

KarelSvo, 2023-04-15:
ddt2000

toll-n, 2023-04-17:
Thanks for the reply. So not available to us ddt4all users. 
But it seems that the information can be found by searching through the ecu file.
Using the information above as an example I found that the ecu file is  
BCM95_SW25_V1.1_20150429T131442.xml
which can be found here:
...\ddt4all-master\ecus
Load into an editor eg Notepad++
Search for the fault code DT_ACC2 or the number 38268 
Line 27826   
  <Item Value="38268" Text="DTC_ACC2"/>     Gives the fault code in decimal.
Line 34941
  <Device Name="DTC_ACC2" DTC="38268" Type="4">
      <Mnemonic>DTC_ACC2</Mnemonic>
       <Test Type="20">
       <Description>DTC_ACC2SCGndOC</Description>
        </Test><Test Type="18">
        <Description>DTC_ACC2SCVBat</Description>
        </Test>
       </Device>
This gives me some fault information. 
Type 20 (0x14) Short cct to gnd or open cct
Type 18 (0x12) short cct to batt.

Thought this might be useful to other users. 
A question for @KarelSvo:

toll-n, 2023-04-17:
sorry pressed enter too soon.
The question is about identifying the fault as type 20 or type 18. 
Is the only way to do this to look into the pyren ecu log?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif