cedricp/ddt4all #706: Disable Stop-Start system on Dacia Lodgy

Kisszalimo, 2022-07-20:
**Car:** Dacia (Renault) Lodgy (2018)
**Type:** Diesel
**Engine:** Renault 1.5 blue dCi 115

***To be clear:*** I'm studying software engineering at the university so I have experience with computers, softwares, hardwares etc. However my knowledge of car programming is very basic. So I'm just an ordinary person who like computers :)

I would like to disable Stop-Start system on my Dacia Lodgy. Unfortunately there is no well-explained guides, just brief "click here - click there" tutorials and I think a lot of people (like me) don't know what they actually do, just try to follow the steps and hope they don't wreck their cars...

I found two tutorials on internet ([link1](https://lisandru.eu/automotive/ddt4all-hacks/super-changes-with-ddt4all-on-your-duster.html), [link2](https://lisandru.eu/automotive/ddt4all-hacks/dacia-logan-sandero-ddt4all-hidden-features.html)), but they are about other Dacia cars (Logan, Sandero, Duster, Dokker). Maybe they work on Lodgy too, but I'm not sure about that.

To make matters worse, I found ***STOPSTARTINHIBIT*** setting on multiple places and I don't know which is the right place.

For example

UCH >> T4_VS_BCM_DDT2000_SW... >> GEE >> GEE - Configuration:
[image]

UPC-EMM >> T4_EMM_DDT2000_SW... >> EMM_AlternatorManagement >> EMM_AlternatorManagement - Configuration:
[image]

Is there anybody who can tell me how to disable Start-Stop correctly? It would be appreciated if someone can explain me WHY we do what we do. :)

AndyIreland, 2022-07-20:
on a renault clio mk4 with stop-start there is an extra wire on the 12v battery under the bonnet, I have heard that once that is taken off (I would wrap some self adhesive insulation tape around it to stop shorts) then it disables stop-start, I have never tried it.

Kisszalimo, 2022-07-20:
Thank you for your answer. I am already aware of this solution, I found the wire under the bonnet. However I found another way to disable Stop-Start system: I installed a relay -connected to an Arduino- which generate an impulse to the S&S disable switch after ignition. So I could disable S&S with this "trick". 

I was just wondering if I could disable it with ddt4all... it would be an interesting experiment for me :)

AndyIreland, 2022-07-20:
is this video of any help its for the Megane , but it might help you still?:
https://youtu.be/7-vIlxGQkv0

Kisszalimo, 2022-07-22:
Maybe it can work, but I'm not sure about that. I'm afraid that I ruin my car if I give it a try. Or is it not dangerous?

AndyIreland, 2022-07-23:
yes always an element of danger when programming the computers/ecu's of your car with software

josefe17, 2022-07-26:
A the second left menu, select the UCH for your car (CMF1_BCM_DDT_10_Immo_20160208T143838). Then, navigate throuth the screens and look for GEE > GEE- Configuration one. If this exists and the field E_B_STOPSTARTINHIBIT_CF displays false at the READ column, setting it to true with the WRITE field should disable it. However, you should be very careful before writing anything as you may brick your BCM and require a reprogramming with the official tool. If yu have any doubt, post a screenshot of your GEE - Configuration screen.

JABICHOCU, 2022-10-09:
Hello Josefe17, do you know a similar solution for a Captur /Clio 4 2013 model? I can´t find any GEE options on my USM module to disable start-stop...thanx

josefe17, 2022-10-09:
Check other tabs as BMS > BMS - Configuration. The name should be the same.

JABICHOCU, 2022-10-10:
Sorry, can´t find the module you´re tellin me...these is all I have on my car, which one would it be??
[image]
[image]

AndyIreland, 2022-10-10:
BCM = UCH BCM

AndyIreland, 2022-10-10:
not in the BCM UCH instead of BMS? 

[image]

josefe17, 2022-10-10:
UCH stands for "Unité Centrale d'Habitacle" in French, which means Body Control Module, or BCM. And BMS stands for Battery Management System, which is a tab from the UCH options.

AndyIreland, 2022-10-10:
ah right so :

[image]

JABICHOCU, 2022-10-11:
OK, I´ve just checked with ddt4all on BCM, and the only one appearing in green is T4_BCM_DDT2000_SW11_0_20130128T102714. Unfortunately here there is no GEE options...I can find them on T4_BCM_DDT2000_SW13 , 14, 15...but these are all in RED, so I´m not going to risk bricking the module...my car is one of the first models from 2013, so maybe this is the reason

AndyIreland, 2022-10-11:
no thats right , in my experience if in read its not the right match for your car's ECU - I would always use the magnifying glass in DDT4ALL with exception , sometimes with a few ECU's that have to be selected manually if the magnifying glass does not find them. But that could be things like the DCM (TCU) ECU, TDB Clustre, Navigation ecu to name a few. - Its like on the Clio mk4 2013 with SW13 and under if you select 'follow me home' or something on the EMM lighting you have to be very careful because people have reported they have changed something under SW13 and its bricked the whole lighting and have either had to get the ECU re-programmed or replaced!

Furtif, 2023-06-02:
as some as pyren but good info rapport it as discuttions https://github.com/cedricp/ddt4all/issues/836