cedricp/ddt4all #709: Lights

JAzevedo351, 2022-07-31:
Hello all, i need  big hhelp and informatioa too.
How can i did turn off lights imediatli whene i turn off engine on my car?? My car is a grand scenic IV.
thank You for the help

AndyIreland, 2022-07-31:
hello, its one of the settings in the emm lighting ECU .. I think its the 'swiss CF' if that is set to '1' then the headlights stay on when the ignition turns off and only go out when the drivers door is opened.

[image]

JAzevedo351, 2022-07-31:
Hello, Thanks, first of all, the scenic 4 does not have, emm module, the light settings is in bcm, but I'll look better but thanks for your feedback

AndyIreland, 2022-07-31:
you havent got the scenic 4 ph2 have you by any chance only there is a setting in the UCH ecu for delay after engine turned off for the lights:
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

memo0616, 2026-02-03:
Hello, does your vehicle have an automatic headlight sensor?