cedricp/ddt4all #710: Clio as multisense car, instead of Megane Gt

mchbv, 2022-08-02:
Hello. I have a problem with a megane gt, the way that in multisense it is a renault clio regardless of what I select in the car type in radnav. I tried all the radnav variants 1.4, 1.4 to 2, 2 to 2.5, at best I get a talisman or most times a clio.

The only thing I can relate to is this
  Found ECU [Audio] (not perfect match) :RadNav 1.4 to 2.x DIAGVERSION [0D] SUPPLIER [CAP] SOFT [33] VERSION [16981 instead 16966]
And yes, the version is 3.3.16.981 updated from 2.2.15.703.
What else could I try?

shrlnm, 2022-08-07:
@mchbv  Try this
https://cloud.mail.ru/public/suUB/hfw7LQRcc

Do not remember to reload R-link2 after configuration change

And one for rare cases
https://cloud.mail.ru/public/mhZn/xqqbCrAjn

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif