cedricp/ddt4all #711: Android App crash while reading ecu.zip

Mierscheid, 2022-08-02:
Hi guys, 

when I unzip the ecus.rar from the link in Issue #56 and repack it as ecu.zip, the Android app crashes reproducible.
Is there another way, or a "real" ecu.zip as download?

Redmi Note 5, MIUI 11.03, Android 9
Can connect to the ELM-ODBII-BT-Dongle i.e. with "TWIZY", "Tune the T" or also "Torque".

TIA
Regards
Mierscheid

Mierscheid, 2022-08-02:
Could someone send me an ecu.zip that works with the app so I can verify and isolate the error?

AndyIreland, 2022-08-02:
hopefully this one will work for you it does with my ecu tweaker app:

https://filebin.net/8pco7lo5ts6uv0j8

Mierscheid, 2022-08-02:
<3 
Thanks a lot! I'll try it!

Mierscheid, 2022-08-02:
It works! Wow! Thank you!

AndyIreland, 2022-08-02:
great stuff glad to have helped

salihakin, 2023-02-03:
> hopefully this one will work for you it does with my ecu tweaker app:
> 
> https://filebin.net/8pco7lo5ts6uv0j8

I am interested in this file. Can you upload it again

salihakin, 2023-02-03:
> It works! Wow! Thank you!

Is everything ok for twizy.
I want to try it.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif