cedricp/ddt4all #713: Ecu.zip for android app

fineryx, 2022-08-16:
Hello, where can i get the ecu.zip database for the android app? Thanks

AndyIreland, 2022-08-16:
try this:
https://filebin.net/n9p7epmza9trispq

xtendpayaa, 2022-09-22:
Hi!
I'm interested too. Thx!

AndyIreland, 2022-09-22:
try here: 
https://we.tl/t-gQhbd2xuxp

xtendpayaa, 2022-09-22:
> try here: https://we.tl/t-gQhbd2xuxp

Thank you!

Jantjuh26, 2022-10-03:
I'm also interested! Someone got a link?

AndyIreland, 2022-10-03:
Try here:
https://we.tl/t-zZxCrR0F3v

AndyIreland, 2022-10-03:
> Thanks!

your welcome

svenvogel, 2022-12-30:
Hi there, the links are not more working. Can anyone send me a link to the ECU.zip too?

AndyIreland, 2022-12-30:
> Hi there, the links are not more working. Can anyone send me a link to the ECU.zip too? Where is normally the place where I can download them?

here is link (for 7days anyway) - I dont really know where  the normal place is for it to be honest.
https://we.tl/t-ioqrkphpq0

svenvogel, 2022-12-30:
@AndyIreland thank you very much! Where do you found them or where are updates for them?

AndyIreland, 2022-12-30:
> @AndyIreland thank you very much! Where do you found them or where are updates for them?

no problem - I found that ages ago - could be a year ago now. - Think I got it either off a car  forum or I might have download it from here for ddt4all with ecu zip included 
https://www.drive2.ru/b/498093336985338243/

and then just copied over the zip file

xtendpayaa, 2022-12-30:
Here is a bigger ecu file:
https://mega.nz/folder/nEsgHDCQ#UJt650dKyzVdUe4CLzH0uw

svenvogel, 2022-12-31:
@AndyIreland @xtendpayaa thanks!

cybersholt, 2023-01-13:
@xtendpayaa thanks for the upload. 

It would be great if the install docs link to a file or explained how to generate.

salihakin, 2023-02-04:
I'm also interested! Someone got a link?

xtendpayaa, 2023-02-04:
> I'm also interested! Someone got a link?

3 reply above to you the mega.nz file working nicely

salihakin, 2023-02-04:
> > I'm also interested! Someone got a link?
> 
> 3 reply above to you the mega.nz file working nicely
> 
> Or here the smaller one: https://mega.nz/file/NaFwGZIa#p713tInIHptfSkYwH2R3lP-wm3FFrkobeih_zM3bhTQ Need to rename the ecu.zip2 to ecu.zip to get working.

Thank you so much.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

radjool, 2023-09-14:
hi,
i am looking for the latest ecu.zip database that contains dacia duster 2023,  could those who have it share please?

MTPro94, 2024-05-08:
> Iată un fișier ECU mai mare: https://mega.nz/folder/nEsgHDCQ#UJt650dKyzVdUe4CLzH0uw

Link 2024 please

MTPro94, 2024-05-08:
ecu.zip 2024 valid link ? Please . Thank you

Furtif, 2024-05-08:
> ecu.zip 2024 valid link ? Please . Thank you

https://github.com/cedricp/ddt4all/discussions/845#discussioncomment-6144566