cedricp/ddt4all #714: RLink2 antitheft mode active

RPAuto, 2022-08-19:
Can someone help m remove antitheft mode on renault megane 4?
Unit change
Original dead 
thanks

MellotronScratch, 2022-09-15:
You need to change a byte in the eeprom. Not possible through OBD as far as I know.

rttr67, 2022-10-08:
I can help  
remove antitheft mode on renault

RPAuto, 2022-10-10:
Perfect what you need?

rttr67, 2022-10-10:
I answer you by email.

Rob69rt, 2023-01-23:
I have the same problem right now. I bought radio LANR16 from  Kadjar to my megane IV(with LANR14)and could someone help me with deactivation antitheft mode?

rttr67, 2023-01-23:
Rob69rt.
Leave an email and I will tell you options.

Rob69rt, 2023-01-23:
> 

> 
My email:
xiaomih1s@gmail.com
Thanks in advance

jurekogurek, 2023-02-02:
> Rob69rt. Leave an email and I will tell you options.

will you help me take off the antitheft? I bought a module for carplay lanr16. My email: knowak161@wp.pl

Grazi77, 2023-03-25:
c e una soluzione per sbloccare rlink2 con antitheft erroneamente attivato?

rttr67, 2023-03-25:
c'è sempre una soluzione

Grazi77, 2023-03-25:
mi puoi contattare?
g.torazza@gmail.com

Grazi77, 2023-03-26:
> Devi cambiare un byte nella eeprom. Non è possibile tramite OBD per quanto ne so.

hai maggiori informazioni su questo bit da modificare nel dump della eprom?
ho fatto il dump della eprom ma nn riesco a montare l immagine bin su l emulatore

ionitagelu, 2023-04-22:
Hello, for my Megane 4 I bougt a v9 unit (from another car), and I need to disable antitheft. Please help me. I have Eprom programmer.
My email: gelu.ionita@gmail.com
Thank you!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

MouadSoufiani, 2023-06-25:
Hi, can you send me also email for anti theft? mouadsoufiani@hotmail.be

ahs2184, 2023-06-29:
Could you please email regarding anti-theft Alex-simpson@blueyonder.co.uk

Serg797979, 2023-08-29:
Hello. help me remove the LANR14 anti-theft protection please. buditch_s@mail.ru

ricardomontenegro24, 2023-09-23:
Can you email me how to remove anti theft?

rjvmontenegro@hotmail.com

kzoid, 2023-09-23:
Someone already found out? I want to put an v9 unit in my Megane IV.

Kjaer123, 2023-11-22:
Rttr67

I need help to deactivate anti theft on a lanr16. 
I have changed vin and vehicle id. 
Mikael@kjhansen.dk

saadousfar, 2023-12-28:
hi i have the solution to desable the antitheft message by can for any information plz contact mee on saadousfar59@gmail.com

Kjaer123, 2023-12-29:
Thanks for your reply, I found the vehicle I'd in the dump from my old radio, and put it in the new radio. Then it was only a matter of adaption.
Now I works

Regards mikael

Sendt fra Outlook til iOS<https://aka.ms/o0ukef>

GiuseppeGullo67, 2024-03-06:
> Posso aiutarerimuovere la modalità antifurto su renault

GiuseppeGullo67, 2024-03-06:
Buonasera a tutti, io come ho letto anche altri sono in procinto per acquistare un Rlink usato, come faccio a non farlo bloccare , ho visto che con il programma ddt4all puoi cancellare e riscrivere il vin, ma prima di fare delle belinate vorrei essere sicuro, se qualcuno mi può dare consigli o aiuto ne sarei grato, la mia mail è ducatista67@libero.it , grazie anticipatamente

GiuseppeGullo67, 2024-03-06:
ducatista67@libero.it

Kjaer123, 2024-03-07:
Grazie per la risposta.
l'ho risolto da solo

Sendt fra Outlook til iOS<https://aka.ms/o0ukef>

shrlnm, 2024-03-07:
@GiuseppeGullo67 
https://www.drive2.ru/b/669234795749858633/

GiuseppeGullo67, 2024-03-07:
Grazie mille, un po’ complicato per me, ma ti darò aggiornamento in ogni caso

Giuseppe Gullo

lklik, 2024-03-29:
hi rttr67 I need your help to deactivate anti theft  with my Rlink  this's my email azizouaankour@yahoo.fr

munkun46, 2024-04-12:
> merhaba rttr67 Rlink'imle hırsızlığa karşı korumayı devre dışı bırakmak için yardımınıza ihtiyacım var bu benim e-posta adresim [azizouaankour@yahoo.fr](mailto:azizouaankour@yahoo.fr)

If you have an elm327 device, you can unlock it.

lklik, 2024-04-13:
Merhaba, Thanks to your reply, i appreciate your help , yes I do have the OBD2 device .Envoyé de mon iPhone

saadousfar, 2024-04-14:
contact mee

Arcadius82, 2024-04-26:
Hello. I have same problem with radio LANR14 replaced on second hand as oryginal unit not responding. Changed ID and VIN but its black screen still not working  and black screen. Before when i plug in second hand unit was ani theft message,

Can someone help me with this ?Please
aer.automotive@mail.com

andresantosmocas, 2024-05-16:
> I can help remove antitheft mode on renault

Hey, I also Bought a LANR16 and I have the anti-theft message. andresantosbraz@gmail.com

liciniofs, 2024-10-18:
Can you send me as well darth.design@gmail.com

johnkar87, 2025-03-27:
> Thanks for your reply, I found the vehicle I'd in the dump from my old radio, and put it in the new radio. Then it was only a matter of adaption.
> Now I works
> 
> Regards mikael
> 
> Sendt fra Outlook til iOS<https://aka.ms/o0ukef>
> […](#)

hi could you send me your email?

saadousfar, 2025-03-27:
I sell à Software to remove anti-theft via can saadousfar59@gmail.com

johnkar87, 2025-03-27:
my friend There is no way to pay to anybody I dint trust anybody who just
saying he have any software (not only you) if you want help people to forum
help them and not want to pay you

shrlnm, 2025-03-29:
@johnkar87 You may try it for free..  https://www.drive2.ru/b/669234795749858633/
Now qpyren3 contains this macro.
- Install qpyren3 and run it
- press button UPDATA and wait DONE message
- Go to MACRO
- select ./macro/rlink2  folder
- then select unlock.cmd

saadousfar, 2025-05-17:
i have a software to remove the anti-theft no need t odissampble the unit to read eeprom  saadousfar59@gmail.com

saadousfar, 2025-05-17:
[image]

MouadSoufiani, 2025-05-17:
This guy above me does it for 120eu. Very expensive

[image]

Renotech1, 2025-05-17:
Don't pay anyone 🙂
If they dont want to share there all scamming bullshitsss

ionitagelu, 2025-05-17:
> Don't pay anyone 🙂
> If they dont want to share there all scamming bullshitsss
> 
> Op za 17 mei 2025, 13:40 schreef Mouad Soufiani ***@***.***>:
> […](#)

You can use PyRen and script from russian site. It's free !

Renotech1, 2025-05-17:
Still not figured out how do I run the .cmd script ?

MouadSoufiani, 2025-05-17:
Offcourse not. Whole this community is free.

saadousfar, 2025-05-17:
> Ne payez personne 🙂 S'ils ne veulent pas partager leurs conneries d'arnaque Le 17 mai 2025, 13:40, Mouad Soufiani a écrit ***@***.***>:
> […](#)

TIME = money

MouadSoufiani, 2025-05-26:
> > Still not figured out how do I run the .cmd script ?
> 
> Run the pyren3 or qpyren3 launcher. Press UPDATE to download the latest macro including the unlock. https://gitlab.com/py_ren/pyren/-/blob/pyren3/pyren3/macro/rlink2/unlock.cmd?ref_type=heads Then click the MACRO button and select unlock.cmd.

Tried it on 2 units, but doesnt work. It reboots, but anti-theft is coming back

shrlnm, 2025-05-27:
> Tried it on 2 units, but doesnt work. It reboots, but anti-theft is coming back

May you capture all frames from the bus during reboot of the RL2?

saadousfar, 2026-08-27:
> You need to change a byte in the eeprom. Not possible through OBD as far as I know.

hi it possible to disable the antitheft via can by my software saadousfar59@gmail.com