cedricp/ddt4all #715: Cannot read data

zkmsh, 2022-08-20:
Hi, 

I cannot read data with DDT4ALL except in edition mode but then when I press "WRITE" nothing happens. Can you help me figuring this out please? I bought an OBD adapter that I modified, PIN 13 to PIN 6 and PIN 14 to PIN 12.

Here is my LOG.

Thanks for you help!

AndyIreland, 2022-08-20:
your car got the rlink2 yes?
Did you wire it up like this:
[image]

what are you trying to enable / disable on it and have you tried changing the CAN timeout ms to 600ms and then refresh?

zkmsh, 2022-08-20:
Hey,

So I make it work but my pin was a little bit damaged so I have a lot of error message. I order another adapter.

I don't have medianav (R-LINK) but I have the Carminat TOMTOM.

I'm trying to restore radio and bluetooth fonctionnality why was remove by Renault when I went to see them seeking for help.

zkmsh, 2022-09-09:
Hello, I have a weird issue, everytime I try to change ms timeout to 600 ms, ddt4all just closed itself. 

I think the issue is that my acoustic parameters have been reset. I'm still looking on how to get them back.

I have no idea why ddt4all closed itself tho...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

zkmsh, 2024-11-01:
Hello,

I gave up on this issue but I'm back at it.

If I understand correctly I have to find accoustic parameters and write them with DDT4ALL. I found these parameters: [https://gitlab.com/py_ren/pyren/-/blob/master/pyren/macro/megane3_fluence/radiosat_classic_bluetooth_accoustic.cmd](url), do you think it could work?