cedricp/ddt4all #717: Megane 3 - EOL Settings after reset configuration

wotekPL, 2022-09-05:
Hi, 

i reset the configuration on my Rlink in Megane 3 and now i get message "Please Add Settings" so can i repair without clip and reprog or  clip is the last chance?

AndyIreland, 2022-09-05:
did you press the soft reset or hard reset button?
Is it rlink 1 or rlink2 in your Megane?

wotekPL, 2022-09-05:
I press reset configuration

AndyIreland, 2022-09-05:
is it rlink1 tomtom or is it rlink2 rlink fitted to your megane ?

AndyIreland, 2022-09-05:
ah right thanks. I have repaired my rlink1 in my renault zoe once when it had EOL triangle before with DDT4ALL by filling in settings. 
Its a bit difficult to explain , but if you have a elm327 bluetooth OBD 2 and the specially wired cable to read the rlink through the OBD2 port and a windows laptop and can get an internet signal on your laptop inside your car I can log on over the internet and have a look remotely from my end to see what settings are filled in and see if i can see other settings missing if you like, if you want me to just let me know.

wotekPL, 2022-09-09:
Yes Sure! 
Now cannot turn on radio and navigation :( 

I have clip or ddt4all. 

Do you think we can do anything else?

If i change bic to old version screen can turn on but with information abaout settings. If i change bic to newer version then i cannot turn on and if i click eject disk then led blink 10x.

wotekPL, 2022-09-09:
Ok. Actually info about EOL is back.

wotekPL, 2022-09-15:
@AndyIreland how can i launch firmware from your post. I format sd card in FAT32 but software not launched?
I copy file tomtom.000 but canot launch this.

wotekPL, 2022-09-15:
@cedricp, @shrlnm 
[image]

Software Number is a 333C and ddt4all cannot autoidenty version MFD so how can i overwrite this row?

AndyIreland, 2022-09-15:
OK , so on my Renault Zoe I have a 'Home' button next to the touch screen itself - i dont know how the Megane 3 is configured with its buttons but i pressed the Home button on mine with 5 pushes with the tomtom.000 file on the card inserted and it started to load the firmware. My Zoe does not have a BIC unit in it like yours so you will have to make sure that is configured as well I think because sometimes when you press the 'reset software' in the mfd settings it can erase settings in BIC as well I think. 

other things you can try is that if the MFD does not auto find in DDT4ALL you have to set it manually choosing MFD 5.4 from the list then choose your settings then press home button 5 times or 'hard reset' within DDT4ALL to reboot the rlink

Another thing you could try is formatting a 8gb USB stick to Fat32 copy over the tomtom.000 file to it and try putting that into usb socket of car and see if that starts loading the software

wotekPL, 2022-09-15:
Ok could you show me screen with all data?

What can i generate VechicleIdentifier im BIC configuration?

Screen here https://github.com/cedricp/ddt4all/issues/92#issuecomment-457928308

wotekPL, 2022-09-15:
Yes, i try with pendrive format to FAT32 and new  card with format. Maybe i format card with map and then copy tomtom.000 to card? What do you think?

AndyIreland, 2022-09-15:
No, i would just copy the TomTom file to fresh SD card not the map card. 

Try use this tomtom.000 file its 11.346 version
https://we.tl/t-GCzQGyQoK3

wotekPL, 2022-09-16:
Eh, unfortunately but i put card or USB stick and nothing happend.

AndyIreland, 2022-09-16:
> Eh, unfortunately but i put card or USB stick and nothing happend.

ah right it must be because the radio is in EOL mode then it cannot boot SD card.

i cannot really think of anything else  then apart from manually selecting MFD 5.4 in DDT4ALL and filling out all the parameters that you can and see if that works - hard to tell what is going on without seeing for myself what is filled out and what is not , and I can only really look if I can log on remotely over the internet.

wotekPL, 2022-09-18:
I bought clip but also i have a problem. 
I cannot autoconfiguration because clip cannot this file on reprog or if i reprog rLinka by RUSB. So maybe somone knows what od RUSB on can clip :(

shrlnm, 2022-09-19:
If you  did reset configuration, then I may generate parameters for your VIN and auto configuration macro for pyren. I need your VIN and screenshot of ID screen from pyren.
Also I have to notice that  there are complaints that pyren can't write long commands to r-link devices. It is related to timeout parameters. It is probably will require to tune a code a bit for  implementing auto-configuration macro on rlink1.

wotekPL, 2022-09-19:
@shrlnm thank you!
Screenshot from pyren. 

Navi
[image]

BIC
[image]

AndyIreland, 2022-09-19:
@shrlnm if i were to supply a dump text file (.txt) of my 2012 radio file or dump of mfd5.4 txt dump file and sent it to someone with EOL and just changed the VIN section to their VIN and CRC then do a Hard Reset would that get their EOL disappeared andtheir r-link 1 working again?

if it did, what PIN code would they need to put in when starting up if it asked for it , would it be mine or theirs?

wotekPL, 2022-09-19:
@shrlnm 

line with $addr should be empty? 

`$addr = `

shrlnm, 2022-09-19:
@wotekPL it is a mistake. Insert an address here

wotekPL, 2022-09-20:
Which address can i write?

shrlnm, 2022-09-20:
TargetAddress from Uces.xml  for your ECU

    <EcuData name="11217" doc="MFD_EUR_X38PH2_0000_14_B" stdType="STD_B">
        <SessionId>genSession:GenSession?GenSession#11217</SessionId>
        <TrtId>diagServices:EcuDiag?EcuDiag#11217</TrtId>
        <ModelId>FG1011217.XML</ModelId>
        <OptimizerId>SG1011217.XML</OptimizerId>
        <ECUInformations>
            <Address **targetAddress="57"** toolAddress="F1" addressExtension="00" MType="0" TAType="0"/>

wotekPL, 2022-09-20:
@shrlnm - how can i thank you? You're made my day! 

T H A N K  Y O U ! ! !

Rlink is back!

closed 2022-09-20 by wotekPL

shrlnm, 2022-09-20:
Ok, glad to hear it. )
Could you tell me, which of the macro helped?

wotekPL, 2022-09-20:
@shrlnm cmd_10_4461_259205564R.cmd 
After use it i saw rlink logo.

shrlnm, 2022-09-21:
@wotekPL Thank you for feedback. One more question. Did you use USB-ELM or a BT/WIFI? If you saved a log could you share it? Your information may help others to repair their rlinks.
Thanks in advance

shrlnm, 2022-09-21:
Thank you. I've downloaded them. You may remove them from here.

catalup, 2022-10-05:
> If you did reset configuration, then I may generate parameters for your VIN and auto configuration macro for pyren. I need your VIN and screenshot of ID screen from pyren. Also I have to notice that [...]

@shrlnm could you please help me too? I am in the same situation. What screenshot is needed to send beside VIN? Thank you in advance!

shrlnm, 2022-10-06:
@catalup VIN may be  enough

catalup, 2022-10-06:
> @catalup VIN may be enough

@shrlnm the VIN is:

VF1BT17J650107677

Thank you a lot for help!

catalup, 2022-10-08:
> @catalup [link removed]

Could you please tell me more details? I don't know really what to do with the code lines that you've sent me. I will look a bit, but I'm not sure. 

Thank you a lot!

shrlnm, 2022-10-09:
@catalup I sent you pyren macro for repair your r-link.
Here is how to install the pyren
https://gitlab.com/py_ren/pyren/-/tree/pyren3
and wiki how to use a macro
https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen

catalup, 2022-10-10:
I succeeded using the instructions from this video:

https://youtu.be/h5IOZ-skiD8

Thank you a lot for your help! 
Have a beautiful week!

StepanenkoNadejda, 2022-10-14:
> > @catalupВИН может быть достаточно
> 
> @shrlnmВИН это:
> 
> VF1BT17J650107677
> 
> Большое спасибо за помощь!

Hello. Please could you help me?) I have a problem with my RLink " ...  unconfigured device please add eol settings ". VIN: VF1BZS00E52144364. PLEASE)

StepanenkoNadejda, 2022-10-14:
Hello, could someone help me) I have the same problem with my RLink " ...unconfigured device please add eol settings " VIN: VF1BZS00E52144364. PLEASE)

StepanenkoNadejda, 2022-10-14:
Hello. Please could you help me?) I have a problem with my RLink " ... unconfigured device please add eol settings ". VIN: VF1BZS00E52144364. PLEASE)

StepanenkoNadejda, 2022-10-14:
Please

StepanenkoNadejda, 2022-10-14:
Text

felixcastrofuentes, 2023-02-04:
@StepanenkoNadejda / @andrerochadev92 can you fix it?

vvvvitaliu, 2023-03-01:
> ah right thanks. I have repaired my rlink1 in my renault zoe once when it had EOL triangle before with DDT4ALL by filling in settings. Its a bit difficult to explain , but if you have a elm327 [...]

help me please restore r-link 1.

iliapeev, 2023-03-09:
@catalup 
Can you help me too?
I have the same problem.
vin:VF1BT17J652378561

shrlnm, 2023-03-10:
@iliapeev 
> vin:VF1BT17J652378561
https://cloud.mail.ru/public/SXiF/E8uvZRmB8

iliapeev, 2023-03-10:
@catalup thanks.it didn't work. only the logo of r-link

AlMeu, 2023-03-10:
The unit which have hardware problems you can't solve via software.

shrlnm, 2023-03-11:
@AlMeu Hi. Do you think it's hardware problem? I've never had RL1 so I not too aware with it. What do you think if I make a macro for radio module Radio_2012_V2.4_20130604T134716.xml too? Can it help?

iliapeev, 2023-03-11:
> Устройството, което има хардуерни проблеми, които не можете да разрешите чрез софтуер.

I don't think it's hardware. It broke when deleting errors.

shrlnm, 2023-03-11:
@iliapeev is your ELM able to connect to multimedia CAN?

iliapeev, 2023-03-11:
> @iliapeevвашият ELM може ли да се свърже с мултимедиен CAN?

yes.can

shrlnm, 2023-03-11:
@iliapeev 
https://cloud.mail.ru/public/fQNC/7H4Mcib8u
try to implement and save the log please.

iliapeev, 2023-03-11:
> 

thanks, i'll try now

iliapeev, 2023-03-11:
thanks a lot.it worked.now i just need to find the pin code

shrlnm, 2023-03-11:
Ok. Perfect.

iliapeev, 2023-03-11:
This is probably not the topic here, but does anyone know how to watch a video in motion?

AlMeu, 2023-03-11:
R-Link 1 doesn't have pin code.
You are sure that you speak about R-Link 1?

iliapeev, 2023-03-11:
> @iliapeev R-Link 1 няма пин код. Сигурни ли сте, че говорите за R-Link 1? Може би говориш за радио,
> 
> @shrlnm Много пъти с R-Link 1, когато се повреди, вие дори не можете да се свържете с него.

yes forr link 1 .first time it ask me for pin .already working

poupz13, 2023-03-11:
HI
For the same problem, can you help me please.
Vin : VF13FL00253820445

thanks a lot

AndyIreland, 2023-03-11:
> @iliapeev R-Link 1 doesn't have pin code. You are sure that you speak about R-Link 1? Maybe you speak about radio,
> 
>
pretty sure my rlink1 in my Zoe EV car has a security PIN code - had to enter it once when I disconnected battery overnight and when I fixed the EOL message on mine

AlMeu, 2023-03-11:
@AndyIreland disconnect radio only from connectors and you will see that requests pin code when you connect back.
If you have a screen where you insert it is not by force always for that. You know something, but you don't know how is really working. 
To be a user, doesn't mean you are an engineer also.

AndyIreland, 2023-03-11:
> @AndyIreland disconnect radio only from connectors and you will see that requests pin code when you connect back. If you have a screen where you insert it is not by force always for that. You know [...]

Thank you for that info - but all I am saying is that when I disconnected my battery and when I fixed my EOL message when I restarted it I am pretty sure I had to put a PIN in on the rlink touch screen to get it working

shrlnm, 2023-03-13:
@poupz13 
> Vin : VF13FL00253820445
Try t.me/pyren_bot
then command 
/vin VF13FL00253820445

[16 more replies]