cedricp/ddt4all #721: R-link reset

gabrielcozma33, 2022-09-17:


gabrielcozma33, 2022-09-17:
Hello , i just delete all setup on the r link . I have a talisman 2016  and for the First time using ddt4all . I presa the Button on hard reset i think ! Now screen is black , no sound from speackers and all setup are gone . I try to reconfigure the settings but once i go out and return , everything is delete . Is there any option to configure the rlink? I dont want to go to dealership 🙈 sorry for my language

AndyIreland, 2022-09-17:
hello - its more than likely 'soft reset' button you have pressed - have you not even got a screen with triangle saying "EOL" on it?

gabrielcozma33, 2022-09-17:
Hello , thank you for the comment , no i dont have anithing , its all black . No sound ... if i go in radio setings all is not present , if i go in navigation all is deletet , i dont know what to do .

gabrielcozma33, 2022-09-17:
I will put some photos with what i did .

AndyIreland, 2022-09-17:
if you have windows laptop and bluetooth OBD2 adaptor and can get Internet in your car will you let me take over and look at your settings of your rlink over the Internet and try a couple of things?

gabrielcozma33, 2022-09-17:
Yes , tomorow i will write you and we can try it if you want

AndyIreland, 2022-09-17:
sure - I shall be available from 2pm onwards (UK Time)

gabrielcozma33, 2022-09-18:
Hello . [image]

This is what i did.. configuration reset or key on reset . I dont remember 🙈

AndyIreland, 2022-09-18:
Hello , it looks like you have the Rlink2 in your car would i be correct? - does it look something like this: 

[image]

I was thinking you had a Rlink1 for some reason , and I was going to look at the files on my rlink1 and try them but that will be no good if you have a rlink2 unit

gabrielcozma33, 2022-09-18:
Hello , i have fix it today . I Found a guy with program (peyton) i think 🥴 and he set the original File . Now Everithin is ok

AndyIreland, 2022-09-18:
ah yes  right thats good PyREN he most probably used and a dump file

gabrielcozma33, 2022-09-18:
Thank you for your help !👍👍👍

AndyIreland, 2022-09-18:
thats OK, glad you got it working in the end

hugoreispt, 2023-03-09:
Hey Andy. I messed up my rlink1 on Renault zoe. Need eol settings something like that. Could you help me? Got Windows 10 computer with ethernet connect and obd single, i just dont know how to operante pyren. 
Mu e-mail: hugoreispt1@gmail.com

hugoreispt, 2023-03-09:
> thats OK, glad you got it working in the end

Hey Andy. I messed up my rlink1 on Renault zoe. Need eol settings something like that. Could you help me? Got Windows 10 computer with ethernet connect and obd single, i just dont know how to operante pyren.
Mu e-mail: hugoreispt1@gmail.com

Aviv5510, 2023-04-22:
> Hello , i have fix it today . I Found a guy with program (peyton) i think 🥴 and he set the original File . Now Everithin is ok

hey, can u send me this ?

AndyIreland, 2023-04-22:
> > thats OK, glad you got it working in the end
> 
> Hey Andy. I messed up my rlink1 on Renault zoe. Need eol settings something like that. Could you help me? Got Windows 10 computer with ethernet connect and obd single, i just dont know how to operante pyren. Mu e-mail: [hugoreispt1@gmail.com](mailto:hugoreispt1@gmail.com)

did you get it sorted/working?

hugoreispt, 2023-04-23:
No. Havent spent time on it. Do you have a solution?

AndyIreland, 2023-04-23:
you can try copying this TomTom.000 file onto the root of a 8gb SD Card formatted to fat32 (not your map card) and see if it gets your rlink1 working. If not, put the file onto a 8gb usb stick formatted to fat32 and see if that gets your rlink1 working again:
https://we.tl/t-a87B6HD4wJ

Aviv5510, 2023-04-23:
> you can try copying this TomTom.000 file onto the root of a 8gb SD Card formatted to fat32 (not your map card) and see if it gets your rlink1 working. If not, put the file onto a 8gb usb stick formatted to fat32 and see if that gets your rlink1 working again:
> https://we.tl/t-a87B6HD4wJ

You have a solution like this for Rlink 2 ?

AndyIreland, 2023-04-24:
> > 
> You have a solution like this for Rlink 2 ?

No I havent , I just done this on my Rlink1 - there is a command script I think you can use with PyREN and try get rlink2 working i think

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif