cedricp/ddt4all #722: Adaptative Cruise Control Clio 5

ALBALOCHI, 2022-09-19:
Hello 
I have a Clio 5 2019 1.0 TCE equipped with front camera and radar so I already have the functions like Distance Warning (DW). 
I belive that the car is most likly adapted to activate the ACC by changing the steering wheel with one has the ACC fucntion. 
after a small research I found that the related paramters are in sertain ECUs like 
- ABS 
- FCAM "Front Camera"
- FRRAD "Front Radar" 

I injected the ABS with new software version which is coming from a Clio 5 with ACC from the factory, but I still can't have the ACC function. 
Does anyone had already succeeded to activate this fuction ? maybe we could dicuss the possibility. 
Thank you . 

[image]

[image]

Erkanaga, 2023-03-03:
Bende deniyorum tuşlu direksiyon buldum taktım k.k tuşları hatası verdi kohm ları enjeksiyon beyni kabul etmiyor uğraşıyorum bir sonuç bulursam paylaşırım

ALBALOCHI, 2023-04-13:
UP

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

dawnwhitewater, 2024-02-28:
Up

dawnwhitewater, 2024-02-28:
> Hello
>
> I have a Clio 5 2019 1.0 TCE equipped with front camera and radar so I already have the functions like Distance Warning (DW).
>
> I belive that the car is most likly adapted to activate the [...]

Do you have any improvement bro.
Maybe we can talk for that 
capoeira_35@hotmail.com