cedricp/ddt4all #723: Clio V Multi Sense

MarVentura, 2022-09-20:
In the past few days I've seen that is possible to activate multi sense on Clios V that don't bring it from factory. Does anybody know the correct path to do it ?

AndyIreland, 2022-09-20:
not so much that, but will you be able to unlock the security gateway with a token to enable it??

https://myrenaultzoeblog.wordpress.com/2021/05/25/changing-infotainment-mmi-settings-on-zoe-ze50-ev-with-ddt4all/

MarVentura, 2022-09-20:
@AndyIreland As you can see in this video this shop gave that option to be enabled https://www.instagram.com/p/Cc3h1bDqfqk/

AndyIreland, 2022-09-20:
[image]

[image]

MEX – stands for Renault Multi-sense – a choice between driving modes (depending on vehicle) – loads can be changed on these pages to do with modes like eco mode / race etc and ambient lighting and sounds like engine sounds.

MarVentura, 2022-09-20:
Can you say that way to that page, I'm new to DDT4ALL and I'm still learning the sections that go to certain pages.

AndyIreland, 2022-09-20:
well - it depends what ECU/multimedia unit it finds when it searches - but say its a A-IVI v1.9

[image]

AndyIreland, 2022-09-20:
How do you plan unlocking the CGW gateway then for making adjustments with DDT4all on the Clio 5 ?

MarVentura, 2022-09-20:
I think you can do it like in this video. https://www.youtube.com/watch?v=MaLw8DO3gTo

AndyIreland, 2022-09-20:
Thanks, the hardware in the video looks expensive and it still looks like you have to register as an independent repairer to unlock

MarVentura, 2022-09-20:
You have another way ?

AndyIreland, 2022-09-20:
no, not at all. And I have not even a Mk5 or later Renault with CGW to experiment on either. 
So some people on this thread say it can be done by bypassing the OBD2 socket and wire up straight to the BCM module ? - some say they are unlocked up to 100km - some say that if you get a renault garage to unlock the gateway it holds until you switch off the engine and restart it - have a read through this thread if you have time

MarVentura, 2022-09-20:
Gonna try to test it with the adapter when I get it, my Clio V 1.0 TCE has 22k km at the moment. If it doesn't work right away I know someone at Renault and maybe they can get it unlock.

AndyIreland, 2022-09-20:
That's good. Good luck with it, let me know how you get on

jordirs2004, 2022-10-15:
> You have another way ?

Connect your obd adapter on the easylink unit or on the bcm unit to bypass the gateway and after use ddt4all 
[image]

KarelSvo, 2022-10-16:
> 
>  and after use ddt4all !

The safe way is to use Pyren mod_DDT. There you have the option of automatically making a backup of the ECU configuration beforehand. In the event of problems, you can use the rollback function to restore the original status.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

eldimoni1, 2023-07-09:
yo lo tengo también activado como en la fotografía de echo hay cosas que tengo activadas , qué estaban ocultas [image] [image] [image] [image] [image] [image]

matzel82, 2023-07-09:
And than ? What can you change ?
I think so...nothing.

eldimoni1, 2023-07-09:
Eso no es nada? Está en menú oculto

matzel82, 2023-07-09:
I want to say. Now...at this moment. What can you change at your car. Normally it's connected with interiorlightning and engine or suspension. The configuration from France is another....the software an configuration at the can Gateway too...

I don't think so that's doesn't work correctly

eldimoni1, 2023-07-10:
Si tu lo dices, aquí te mando una fotografía de lo que se usa para conectar al módulo bcm o al Easylink evitando sgw… saludos…
[image]

eldimoni1, 2023-07-10:
Para liberar el sgw de renault necesitaría el token de renault y la cámara clip esto evita el sgw, y activas algunas cosas

matzel82, 2023-07-10:
The token works only with Original clip with Garage License

AndyIreland, 2023-07-10:
> The token works only with Original clip with Garage License

so you couldnt just get your normal elm327 and DDT4ALL and buy the token and start programming the Clio5 etc which has the gateway? - you have to have a Renault CAN CLIP and can clip software?

matzel82, 2023-07-10:
> > The token works only with Original clip with Garage License
> 
> 
> 
> so you couldnt just get your normal elm327 and DDT4ALL and buy the token and start programming the Clio5 etc which has the gateway? - you have to have a Renault CAN CLIP and can clip software?

NO... it doesn't work. Yes I have original software and Token

matzel82, 2023-07-10:
> No lo sé, me comentaron que hay que tener el token renault y el can clip original, igual si se desbloquea con el token a lo mejor con el dd4ttal y el elm 327 se podría,estoy intentando conseguir un token y probarlo

Correct you need it. And No... it doesn't work with elm. You won't get a token. It's only from Renault possible

KarelSvo, 2023-07-10:
> > The token works only with Original clip with Garage License
> 
> so you couldnt just get your normal elm327 and DDT4ALL and buy the token and start programming the Clio5 etc which has the gateway? - you have to have a Renault CAN CLIP and can clip software?

PassThru J2534 interface stand alone or built into diagnostic tool
Buy Unlock Gateway module and PHYSICAL TOKEN and Token Ticket at ASOS Renault.

matzel82, 2023-07-10:
That's only available for "Pros" also Garages

DrPleb, 2023-10-04:
On Clio V 2020 how do you know on easylink side which cable is can low and can high?

AndyIreland, 2023-10-04:
> On Clio V 2020 how do you know on easylink side which cable is can low and can high?

will this article be of any help to you:
https://myrenaultclioblog.wordpress.com/2023/05/22/add-wireless-android-auto-wireless-car-play-to-renault-clio-mk5-bypassing-the-security-gateway/

DrPleb, 2023-10-05:
> > On Clio V 2020 how do you know on easylink side which cable is can low and can high?
> 
> will this article be of any help to you:
> https://myrenaultclioblog.wordpress.com/2023/05/22/add-wireless-android-auto-wireless-car-play-to-renault-clio-mk5-bypassing-the-security-gateway/

Are you sure that the easylink can high and can low are in the same position for the Zoe and got the Clio V? Because originally this guide was for the zoe...

eldimoni1, 2023-10-05:
No por mucho madrugar... Amanece más temprano

DrPleb, 2023-10-05:
> Are you sure that the easylink can high and can low are in the same position for the Zoe and got the Clio V? Because originally this guide was for the zoe...

The problem is that with my odb2 vgate wireless i cannot get it powered on and the pin on my obd are not the same that the guide says and it doesn't get powered on.

Could the pin scheme be this?

[image]

The black is the negative, the yellow is the +12v red can high and blu can low. The only difference in this photo is that on the other side there is my easylink and not that module

AndyIreland, 2023-10-05:
> > > Are you sure that the easylink can high and can low are in the same position for the Zoe and got the Clio V? Because originally this guide was for the zoe...

No, I am sorry i cannot be absolutely sure because I dont own a Clio Mk5 nor a Zoe ZE50 just an older Zoe without Security Gateway. 

But the Easylink on the ZE50 & Clio 5 look exactly the same unit to me

DrPleb, 2023-10-06:
> No, I am sorry i cannot be absolutely sure because I dont own a Clio Mk5 nor a Zoe ZE50 just an older Zoe without Security Gateway.
>
> But the Easylink on the ZE50 & Clio 5 look exactly the same unit [...]

Thank you! I did it! I tried with DDT4ALL to get Android Auto Wireless but my ECU database is not updated, the Navigation ECU on the database is not updated and it doesn't recognize the ECU on my easylink. Does anyone have the database updated to 2022 ecu's?

I'm trying to use Pyren but i don't understand it and I don't know how to enable aa wireless with it

AndyIreland, 2023-10-06:
> Thank you! I did it! I tried with DDT4ALL to get Android Auto Wireless but my ECU database is not updated, the Navigation ECU on the database is not updated and it doesn't recognize the ECU on my [...]

I dont know where you can get a more up to date database but have you tried manually selecting any of the AVI (I think it is for easylink) selecting it and see if they 'read' - if they are in red the parameters then obviously you chose the wrong ecu or the gateway is still blocking it. 

Yeah if you used PYREN and set CAN2 (mind you I dont know if you still have to do that now on the later Renault cars with gateway) that might find your easylink

AndyIreland, 2023-10-06:
@DrPleb  - i have this xml in my folder , but it says 2021

[image]

AndyIreland, 2023-10-06:
This may be latest database?:
https://github.com/cedricp/ddt4all/discussions/845#discussioncomment-6144566

KarelSvo, 2023-10-07:
v2.8 - 06/10/2022
Added new Config parameters :
CFG_ADAS_REAR_UPA_SETTINGS,CFG_ADAS_RVC,CFG_ADAS_MVC,CFG_ADAS_AVM,CFG_ADAS_SHIELD,CFG_ADAS_ISA_CUSTOM_SHIELD,CFG_ADAS_WDA_CUSTOM_SHIELD,CFG_ADAS_ELKS_CUSTOM_SHIELD,CFG_ADAS_ELKA_CUSTOM_SHIELD,CFG_ADAS_BSW_CUSTOM_SHIELD,CFG_ADAS_WDA,CFG_ADAS_ELKS,CFG_ADAS_ELKS_DEF,CFG_ADAS_UTA_CUSTOM_SHIELD,CFG_Fake_NAV,CFG_CockpitSliderSize,CFG_DaySliderSize,CFG_ADAS_ELKA_OTO-Config items added newly in $2110
part_authentication-DID$4187 added Newly
CFG_TypeOfCluster,ADAS_OSP_ISA -Config items added newly in $2108
3- Config Items Byte position modified in $DAB1
Updated Config parameters:
Welcome sequence added in $2108
18-10-2022:
Added new vehicle project XJAPh2
New Auto-ident 5807
2022-10-19  DiagVersion=16, 144,177,Supplier=BOSCH, Soft=I_PRJ_RN_AIVI_21.15V07, Version=5807_220215
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_21.3V23, Version=5623_220121
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_21.3V24, Version=5624_220226
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_21.3V26, Version=5626_220328
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_21.3V27, Version=5627_220425
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_21.3V28, Version=5628_220511
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_21.3V30,Version=5630_220621
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_22.2V07, Version=6107_220621
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_22.2V10, Version=6110_220731
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_22.2V11, Version=6111_220804
2022-10-19 DiagVersion=16,144,177, Supplier=BOSCH, Soft=AI_PRJ_RN_AIVI_22.2V16, Version=6116_221005

DrPleb, 2023-10-07:
> ```
> v2.8 - 06/10/2022
> Added new Config parameters : [...]

Is this a release of ddt4all? Because i never tried saw XJAph2, used only XJA. Mine is a Clio V E-TECH...may be this is the problem.

I cannot use the latest version of DDT4ALL because when it tries to connect to OBD wireless it starts loading 192.168.0.10:35000 but it stays here. With the previous versions it works...can you help me with this?

eldimoni1, 2023-10-07:
> > > > En Clio V 2020, ¿cómo se sabe en el lado de easylink qué cable es bajo y alto?
> > > 
> > > 
> > > ¿Este artículo le será de alguna ayuda? 
> > > [https://myrenaultclioblog.wordpress.com/2023/05/22/add-wireless-android-auto-wireless-car-play-to-renault-clio-mk5-bypassing- la-puerta-de-seguridad/](https://myrenaultclioblog.wordpress.com/2023/05/22/add-wireless-android-auto-wireless-car-play-to-renault-clio-mk5-bypassing-the-security-gateway/)
> > 
> > 
> > ¿Estás seguro de que el easylink puede alto y bajo están en la misma posición para el Zoe y el Clio V? Porque originalmente esta guía era para la zoe...
> 
> El problema es que con mi odb2 vgate wireless no puedo encenderlo y el pin de mi obd no es el mismo que dice la guía y no se enciende.
> 
> ¿Podría ser este el esquema de pines?
> 
> [image]
> 
> El negro es el negativo, el amarillo es el +12v, el rojo puede ser alto y el azul es bajo. La única diferencia en esta foto es que del otro lado está mi easylink y no ese módulo

[image]

KarelSvo, 2023-10-07:
> I cannot use the latest version of DDT4ALL because when it tries to connect to OBD wireless it starts loading 192.168.0.10:35000 but it stays here. With the previous versions it works...can you help me with this?

https://github.com/cedricp/ddt4all/pull/659/files

AndyIreland, 2023-10-07:
> > I cannot use the latest version of DDT4ALL because when it tries to connect to OBD wireless it starts loading 192.168.0.10:35000 but it stays here. With the previous versions it works...can you help me with this?
> 
> https://github.com/cedricp/ddt4all/pull/659/files

that a WiFi ELM327 then you have? - try making sure that your laptop is disconnected from your home wifi

DrPleb, 2023-10-07:
> that a WiFi ELM327 then you have? - try making sure that your laptop is disconnected from your home wifi

Yes, is this one here

[image]

It works with Pyren and DDT4ALL 2019
Could the error be something related to Python? Because the old 2019 version uses Python 2 and this one uses Python 3

AndyIreland, 2023-10-07:
yeah, can you not try using the old 2019 DDT4ALL version but with an updated ECU zip database? 
https://cloud.mail.ru/public/4mtR/2PLQ7JcmG

DrPleb, 2023-10-13:
> yeah, can you not try using the old 2019 DDT4ALL version but with an updated ECU zip database? https://cloud.mail.ru/public/4mtR/2PLQ7JcmG

Yes, I tried but it cannot find the correct ECU matching my EasyLink version

[image]

Any ideas where I can find that right version?

AndyIreland, 2023-10-13:
do you feel brave enough to use AIVI_ 2.6 if you have it showing up in DDT4ALL list of ECU's selecting it manually?

Your ECU (AIVI21 seems to have a date of 06/12/2021 and AIVI_2.6 looks like it has a date of 02/09/2021 

only if your brave and want to take the chance - if you read only and see everything in RED backgrounds then you will now its not compatible at all.

DrPleb, 2023-10-13:
Yeah, I will try in a few hours. Can I also try with 2.7 and 2.8 that are from 2022 or too much risk?

AndyIreland, 2023-10-13:
if that were me I would try - although could be dangerous ... not actually reading the ECU because you are only reading and not changing parameters. 

Lets put it this way I havent got a ZE50 EV just got a ZE20 EV with an rlink1 (and no security gateway like the ZE50 and Clio5) - well when I was adding Android Auto to the r-link1 I chose a much later Navigation and it all went swimmingly and fine. 
It might be different on the Clio5 easyconnect I wouldnt know, I havent one to test it out on. 

make notes of your settings - or try making a backup dump file first with PYREN before changing any settings (if Clio5/ZE50 easyconnect works with PYREN, I think it should)  - it might just get you out of trouble later on if you can load the settings back on to get you back to how you was should you brick it

eldimoni1, 2023-10-13:
> ¿Te sientes lo suficientemente valiente como para usar AIVI_ 2.6 si aparece en la lista DDT4ALL de ECU seleccionándolo manualmente?
> 
> Su ECU (AIVI21 parece tener una fecha del 12/06/2021 y AIVI_2.6 parece tener una fecha del 09/02/2021
> 
> solo si eres valiente y quieres arriesgarte: si solo lees y ves todo en fondos ROJOS, ahora no será compatible en absoluto.

de donde se puede sacar la ecu A IVI 2.1? yo tengo la AIVI. 2.6

eldimoni1, 2023-10-13:
> Sí, lo intentaré en unas horas. ¿Puedo probar también con 2.7 y 2.8 que son del 2022 o mucho riesgo?

primero antes  de cambiar nada, hazle una fotografia a los parametros, y luego intenta cambiarlos, asi si pasa algo lo puedes volver a grabar, si te sale No Data, es, o que no se puede cambiar o que no lo puede leer...
de donde has sacado las del 2022?
yo tengo la del 2021, lo lee pero no lo graba, solo la camara trasera, los demas parametros si, me han comentado que ha de ser con la can clip y el token de renault

AndyIreland, 2023-10-13:
> de donde se puede sacar la ecu A IVI 2.1? yo tengo la AIVI. 2.6

you could try this 'bigger' ecu list:
https://github.com/cedricp/ddt4all/issues/713#issuecomment-1368113934

AndyIreland, 2023-10-13:
> primero antes de cambiar nada, hazle una fotografia a los parametros, y luego intenta cambiarlos, asi si pasa algo lo puedes volver a grabar, si te sale No Data, es, o que no se puede cambiar o que [...]

yes more than likely if you can read but not change parameters either the security gateway is blocking it (as fitted on later  Renault cars) or chosen wrong ECU. 

You need token to unlock or some people reckon they have bypassed gateway by wiring OBD2 directly to BCM or easyconnet radio wiring at the back of the dash.

eldimoni1, 2023-10-13:
> > > Sí, lo intentaré en unas horas. ¿Puedo probar también con 2.7 y 2.8 que son del 2022 o mucho riesgo?
> > 
> > 
> > primero antes de cambiar nada, hazle una fotografía a los parámetros, y luego intenta cambiarlos, asi si pasa algo lo puedes volver a grabar, si te sale No Data, es, o que no se puede cambiar o que no lo puede leer.. .de donde ha sacado las del 2022? yo tengo la del 2021, lo lee pero no lo graba, solo la camara trasera, los demas parametros si, me han comentado que ha de ser con la can clip y el token de renault
> 
> Sí, lo más probable es que, si puede leer pero no cambiar los parámetros, la puerta de enlace de seguridad lo esté bloqueando (como se instala en los automóviles Renault posteriores) o haya elegido la ECU incorrecta.
> 
[image]
[image]
[image]
[image]
[image]
[image]
y en el BCM, luces diurnas traseras activadas
[image]
[image]

DrPleb, 2023-10-13:
> if that were me I would try - although could be dangerous ... not actually reading the ECU because you are only reading and not changing parameters.
>
> Lets put it this way I havent got a ZE50 EV just [...]

is this correct?

[image]

eldimoni1, 2023-10-13:
> > Si fuera yo, lo intentaría, aunque podría ser peligroso... en realidad no leer la ECU porque solo estás leyendo y no cambiando parámetros.
> > Digámoslo de esta manera: no tengo un ZE50 EV, solo compré un ZE20 EV con un rlink1 (y sin puerta de enlace de seguridad como el ZE50 y el Clio5). Bueno, cuando estaba agregando Android Auto al r-link1, elegí una Navegación y todo salió a la perfección y bien. Puede que sea diferente en el Clio5 easyconnect. No lo sé, no tengo nadie para probarlo.
> > tome notas de su configuración, o intente crear un archivo de copia de seguridad primero con PYREN antes de cambiar cualquier configuración (si Clio5/ZE50 easyconnect funciona con PYREN, creo que debería hacerlo); podría sacarlo de problemas más adelante si puede cargar Vuelva a activar la configuración para que vuelva a ser como estaba en caso de que la bloquee.
> 
> ¿es esto correcto?
> 
> [image]

[image]
este es el mio en el A IVI 1.10

DrPleb, 2023-10-13:
On latest version of ddt4all that can't connect to my OBD, there is this section and i think that this is my ecu[image]

If i try a-ivi 2.8 with ddt4all 2019 that can connect to my obd and i change to aa wireless...wireless on the car os enabled but aa doesn't work, it crashes, and also wifi on the car keep loading to infinite. I tried to set up Renault RS screen and doesn't happen anything, i think i have to make this ddt4all version to work...but i don t know why it doesn't connect, i m using it in demo mode

AndyIreland, 2023-10-13:
> > this correct?
> 
> [image]

yep that looks good - green ones are the 'READ ECU'

did you press 'Hard Reset' afterwards?

DrPleb, 2023-10-13:
> > > this correct?
> > 
> > [image]
> 
> yep that looks good - green ones are the 'READ ECU'
> 
> did you press 'Hard Reset' afterwards?
> 
> 

Of course, and the easylink reboots. And my edits works because the wifi section on the car appears but if I try to connect android auto wireless it crashes and also my phone keep searching for android auto device

AndyIreland, 2023-10-13:
> Of course, and the easylink reboots. And my edits works because the wifi section on the car appears but if I try to connect android auto wireless it crashes and also my phone keep searching for [...]

What android version is installed on your Android phone? - I think wireless android auto works on android phones with android version 11 or 12 only. 
If you have got correct version on your phone go into settings for wireless android auto and make sure that is all set up correctly. 

Try going into android auto app settings  on the phone and in the storage settings , then press 'force stop' then 'clear cache' then 'clear data'  - then setup android auto again on the phone

jordirs2004, 2023-10-13:
> Of course, and the easylink reboots. And my edits works because the wifi section on the car appears but if I try to connect android auto wireless it crashes and also my phone keep searching for [...]

What is the version of your easylink ? Because android auto wireless only work on the last update of the easylink only available in the Renault garage.
But be careful, activate all you want on the easylink link before the update because after, the easylink will be locked and you can't modify anything [image]

[22 more replies]