cedricp/ddt4all #724: Noob questions

xtendpayaa, 2022-09-22:
Hello everyone!
I have a 2018 Renault Kadjar and a 2010 Renault Grand Scenic.
I don't have garage so when it's raining outside all the night my Kadjar folding mirrors don't wanna really work. When no rain it's working perfectly. So I wanna switch off the automatic mirror folding I don't need it (if I need it anyway there is a button for it). So somebody know that how to do that? Because if it's not working perfectly now because the rain then 100% it's not going to work in -25°C too 😂
If somebody know in Scenic too it's a big plus!
Second thing is just an option not so important but is that possible to disable start-stop?
And where I can find database file for Renault?
Thank you!

AndyIreland, 2022-09-22:
Hiya - i dont know what ECU's DDT4ALL finds on your Kadjar - but on my Clio mk4 2016 its on the UPC EMM ECU

[image]

here is details for disabling stop start system , should be similar instructions for your car>
https://youtu.be/7-vIlxGQkv0

xtendpayaa, 2022-09-22:
Thank you very much! I'm gonna try

xtendpayaa, 2022-09-24:
I tried today the Android version ddt4all but no luck.
[image]
I've got only initializing CAN protocol... and that's it. I have iLC bluetooth obd2 scanner what is working perfectly with Torque pro but maybe not good with that software?

What is the cheapest known working device for PC or phone?

AndyIreland, 2022-09-24:
I much prefer using the PC version of DDT4ALL 

This is one I got about a year ago , Kingbolen Bluetooth OBD2 works with DDT4ALL with me
£6.99 UK pounds

AndyIreland, 2022-09-24:
If you have a windows laptop and can get an internet signal in your car and want me to check your obd2 and settings remotely over the internet just let me know

xtendpayaa, 2022-09-24:
> If you have a windows laptop and can get an internet signal in your car and want me to check your obd2 and settings remotely over the internet just let me know

Thank you! I just order it from official website but 17.99USD. I don't know when it's arrive so now I need to wait patiently.

xtendpayaa, 2022-10-23:
Hi again!
So my Kingbolen bluetooth obd reader arrived. I'm connected to Windows then start the ddt4all program. I've got Com3 and Com4 as port. If I choose com4 i got "permission denied" message if I choose com3 then start testing com speed but nothing more happening. I'm checked in Windows the com3 is incoming port with obd name, and com4 is outgoing with obd slave name. 
So what I'm fu*ked up? Or first start need to wait lot of time? Thank you!

AndyIreland, 2022-10-23:
hiya glad you got it delivered - i would need to see how your laptop is configured and i cannot do that blindly, all i can do is log onto your laptop remotely and see how its all configured sorry

xtendpayaa, 2022-11-15:
Hi again!
So ddt4all still not working with computer but working the Android program.
I have switched off start/stop and the folding mirrors with this:
https://github.com/cedricp/ddt4all/issues/131#issuecomment-457033526

But.. when open the car there is a led light in the tails until I start the car because then only DLR lighting in the front (of course there is light in the rear if switch on or auto switch the car because start to be dark outside). I think this is the welcome light? But where I can enable this led tail lights always on when I start the car? Because now in Finland starts to all the they "grey" weather but when starts little bit darkening then the "normal" tail lights switching on. I hope you understand what I want 😂

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif