cedricp/ddt4all #725: 4ws megane iv

radugns, 2022-09-23:
Hello.
Can anyone help me to reconfigure a new 4ws (4control) motor for my megane 4?
Since I bought a used one as replacement, renault dealer does not want to calibrate it for my car. I think that I need to set the vin number according to my car on the new 4control module.
That can be done using ddt4all, right?

AndyIreland, 2022-09-23:
if you need to input VIN its in this ECU - however the VIN number doesn't seem like it can be changed, well not when DDT4ALL is in demo mode anyway. 
The message at the top in french means "parameter values ​​written in memory (speed <2km/h)" 

[image]

radugns, 2022-09-23:
But do I need to really change anything to it? Or just connect the new one and thats it. I just don't want to ruin and to have no possibility to put back the older one.

radugns, 2022-09-23:
And having no vin, I assume its simple a matter of change in parts.

AndyIreland, 2022-09-23:
i could not advise on any of that I am sorry because I dont know. Hopefully someone else can advise you

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif