cedricp/ddt4all #726: Megane 4 OBD2 can't connect via wifi

webmaster750, 2022-10-01:
Hello

I can't connect ddt4all with my adapter(on youtube the same adapter works and can do coding)

The error is : no ELM327 or OBlink detected on com port 192.168.0.10 :35 000

Maybe it is adapter fault?! But adapter works with apps like Torque and Csr scanner.

If You guys have working adapter that works, please post the name or link.

Thank You
[image]

AndyIreland, 2022-10-01:
DDT4ALL normally requires a OBD2 Adaptor with a PIC18F25K80 chip in it - maybe yours not has it in it?

Here is mine that works with DDT4ALL , but its a bluetooth one:
https://www.amazon.co.uk/Kingbolen-Bluetooth-Scanner-Diagnostic-Compatible/dp/B07LC8LKRP

webmaster750, 2022-10-01:
> DDT4ALL normally requires a OBD2 Adaptor with a PIC18F25K80 chip in it - maybe yours not has it in it?
> 
> Here is mine that works with DDT4ALL , but its a bluetooth one: https://www.amazon.co.uk/Kingbolen-Bluetooth-Scanner-Diagnostic-Compatible/dp/B07LC8LKRP

Yea maybe they changed the chip. I was looking this video and he use the same adapter and it works. https://youtu.be/8_MwY3U1klw
I will not open it to search what chip is it as I will return it, and buy what You recommend.
Thanks

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif