cedricp/ddt4all #730: x90 Logan/Sandero

erhsntrk, 2022-10-08:
What changes can we make to this vehicle? ECU is French. I can understand with limited translation. Can you help me anyone?

AndyIreland, 2022-10-08:
what year is the car, if it has not got a security gateway to it there are loads you can change. 
when you say the ECU is in French - do you mean the dashboard text is in french?

KarelSvo, 2022-10-08:
Which engine?  are you from, what is your language?

erhsntrk, 2022-10-08:
> Which engine? Where are you from, what is your language?

1.5dci k9k 792 my engine. Im from Turkey.

AndyIreland, 2022-10-08:
ah right - I reckon you meen the Language of the DDT4ALL program is actually in french and you want it in Turkey - well I dont think you can . the only thing you can rely on is putting the french into Google Translate and do it that way, laborious I agree, but thats what I have had to do on my Zoe EV when DDT4ALL is in french, after a while you pick up certain things without having to google translate - when I first started using DDT4ALL I did not know what  Ecrans and Lecture meant ... but I do now LOL :D

AndyIreland, 2022-10-08:
> mk2 logan's ecm/uch doesn't look like. and its interface is very different for them.

by the way , better to do a search on the ECU's with the magnifying glass when connected rather than pick the ECU manually from the list!

erhsntrk, 2022-10-08:
> ah right - I reckon you meen the Language of the DDT4ALL program is actually in french and you want it in Turkey - well I dont think you can . the only thing you can rely on is putting the french into Google Translate and do it that way, laborious I agree, but thats what I have had to do on my Zoe EV when DDT4ALL is in french, after a while you pick up certain things without having to google translate - when I first started using DDT4ALL I did not know what  ecriture and Lecture meant ... but I do now LOL :D 

i can use the program ^^ , is there any feature that i can turn on in the ECU, uch etc... of the car.  e.g;  like activating the trip computer

KarelSvo, 2022-10-09:
The trip computer can only be changed directly in the instrument panel memory chip. The instrument panel has no connection to the OBD port. Search for the Logan Firmware Editor 2.6
Read international Logan forums. There are very few changes in X90 vehicle type. Such as the flashing noise or the automatic locking of the door while driving. I recommend Pyren for reading data and diagnosing errors on all ECUs. There is also a Turkish language file for this software, which automatically translates some parts of the ddt4all file.

erhsntrk, 2022-10-09:
> The trip computer can only be changed directly in the instrument panel memory chip. The instrument panel has no connection to the OBD port. Search for the Logan Firmware Editor 2.6
> Read [...]

Thanks for your answer have a nice day 👍

AndyIreland, 2022-10-09:
Have you tried manually selecting TDB cluster ECU for a X52 Logan and see if that reads any parameters from your dash after you are connected with DDT4ALL?

[image]

KarelSvo, 2022-10-09:
I already wrote, X90 has no connection between TDB/CLUSTER and OBD port.

AndyIreland, 2022-10-09:
> I already wrote, X90 has no connection between TDB/CLUSTER and OBD port.

ah right thanks - sorry

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif