cedricp/ddt4all #732: Change revision days on cluster Clio 4

JABICHOCU, 2022-10-09:
Hello to all of you...i´m trying yo change revision days form 365 to 720 on my Captur/Clio 4 cluster and it doesn´t allow me to save changes...after writing the change cluster resets but it is back to 365 days. Can anybody tell me if this is possible???  I tried many times without success...thanx

[image]

KarelSvo, 2022-10-09:
Give the log files from the ddt4all/logs folder to see what the ecu responded to.

ohayyam, 2022-10-09:
KarelSvo please email me for some question :)  omern7100@gmail.com

JABICHOCU, 2022-10-09:
Well, to tell you the truth this was actually tried with Ecu Tweaker. on android, and have no way to check log...this is the error while writing data>>>
[image]

JABICHOCU, 2022-10-09:
sorry...I just found out there´s a way to download the log...this is what it says:

New session with ECU Cluster X98_X87_X82_xHA_X98ph2_X87ph2_v7.4
CAN SENT: [08-10-08:35:21] 10C0
CAN RECV: [08-10-08:35:21] 50C0
CAN SENT: [08-10-08:35:21] 220201
CAN RECV: [08-10-08:35:21] 6202017530
CAN SENT: [08-10-08:35:21] 22020C
CAN RECV: [08-10-08:35:21] 62020C0168
CAN SENT: [08-10-08:35:21] 22020E
CAN RECV: [08-10-08:35:21] 62020E0012
CAN SENT: [08-10-08:35:21] 220202
CAN RECV: [08-10-08:35:21] 62020202D0
CAN SENT: [08-10-08:35:22] 22012C
CAN RECV: [08-10-08:35:22] 62012C0F
CAN SENT: [08-10-08:35:22] 22020D
CAN RECV: [08-10-08:35:22] 62020D6E33
CAN SENT: [08-10-08:35:22] 22020B
CAN RECV: [08-10-08:35:22] 62020B7530
CAN SENT: [08-10-08:35:22] 22020F
CAN RECV: [08-10-08:35:22] 62020F0F
CAN SENT: [08-10-08:35:22] 220204
CAN RECV: [08-10-08:35:22] 620204017A
CAN SENT: [08-10-08:35:22] 220203
CAN RECV: [08-10-08:35:22] 6202036E33
CAN SENT: [08-10-08:35:24] 220201
CAN RECV: [08-10-08:35:24] 6202017530
CAN SENT: [08-10-08:35:24] 22020C
CAN RECV: [08-10-08:35:24] 62020C0168
CAN SENT: [08-10-08:35:24] 22020E
CAN RECV: [08-10-08:35:24] 62020E0012
CAN SENT: [08-10-08:35:24] 220202
CAN RECV: [08-10-08:35:24] 62020202D0
CAN SENT: [08-10-08:35:25] 22012C
CAN RECV: [08-10-08:35:25] 62012C0F
CAN SENT: [08-10-08:35:25] 22020D
CAN RECV: [08-10-08:35:25] 62020D6E33
CAN SENT: [08-10-08:35:25] 22020B
CAN RECV: [08-10-08:35:25] 62020B7530
CAN SENT: [08-10-08:35:25] 22020F
CAN RECV: [08-10-08:35:25] 62020F0F
CAN SENT: [08-10-08:35:25] 220204
CAN RECV: [08-10-08:35:25] 620204017A
CAN SENT: [08-10-08:35:25] 220203
CAN RECV: [08-10-08:35:25] 6202036E33
CAN SENT: [08-10-08:35:51] 10C0
CAN RECV: [08-10-08:35:51] 50C0
CAN SENT: [08-10-08:35:52] 1103
CAN RECV: [08-10-08:35:52] 5103
CAN SENT: [08-10-08:36:01] 10C0
CAN RECV: [08-10-08:36:01] 50C0
CAN SENT: [08-10-08:36:01] 1103
CAN RECV: [08-10-08:36:01] 5103
[08-10-08:56:19]  Log file created

AndyIreland, 2022-10-09:
Did you press the 'prise en complet.." button at the bottom to finalise the setting and reset the TDB Clustre? - when you press that the dash should go blank for a second or 2 as it resets and if you have dials on your cluster they may flicker/buzz as the dash is reset

[image]

KarelSvo, 2022-10-09:
According to the data log, point 2 was not carried out.

JABICHOCU, 2022-10-09:
Well, I pressed "prise en comple.." but I didn´t press "duree initiale" before...could that be the mistake??

AndyIreland, 2022-10-09:
yeah i think you would have to do it in that order to make it stick

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif