cedricp/ddt4all #734: Problem with RLINK

StepanenkoNadejda, 2022-10-14:
Hello. Please could you help me?) I have a problem with my RLink " ... unconfigured device please add eol settings ". VIN: VF1BZS00E52144364. PLEASE)

shrlnm, 2022-10-15:
Hello
https://cloud.mail.ru/public/CkBw/gSBsghLKW

StepanenkoNadejda, 2022-10-15:
Hi. Спасибо)

StepanenkoNadejda, 2022-10-15:
> Hello Ссылка в почте

Hello, is it possible to do within CLIP (on Android)?

AndyIreland, 2022-10-15:
what do you do? change the .cmd to .txt and then choose 'rollback' in PyREN @shrlnm

StepanenkoNadejda, 2022-10-15:
> что вы делаете? измените .cmd на .txt, а затем выберите "откат" в PyRen @shrlnm

Я новичок работы в PyRen). Я пытался все сделать как в ролике (https://www.youtube.com/watch?v=h5IOZ-skiD8). Установил все, пытался подключить свой USB ELM327 (чип CH340T), но все время выдает не найдены порты моего ELM327 именно в PyRen Laucher. Подумал, что как-то не так устанавливаю, и думаю провобать через Андроид с ELM327 BT) ВОт и хотел узнать может данную процедуру можно через CLIP проделать)

StepanenkoNadejda, 2022-10-15:
> > I am newbie working in PyRen). I tried to do everything as in the video (https://www.youtube.com/watch?v=h5IOZ-skiD8). I installed everything, tried to connect my USB ELM327 (CH340T chip), but all the time it gives out the ports of my ELM327 were not found in PyRen Laucher. I thought that somehow I was installing it wrong, and I was thinking of trying through Android with ELM327 BT) So I wanted to know if this procedure can be done through CLIP)

AndyIreland, 2022-10-15:
you may need OBD2 adapter with a PIC18F25K80 chip in it to use PyREN - or the OBD2 USB ports have not been added correctly in windows device manager

StepanenkoNadejda, 2022-10-15:
Thank  you) Could you write a few important steps for release this issue). May be recommend a source of information - a video or a website?

AndyIreland, 2022-10-15:
not the best at explaining I would have to log onto your laptop over the internet and see for myself what is going on and what ports etc  is listed in the device manager in your windows if you want me to do that let me know. 

Apart from that maybe youtube is more than likely your best bet for a video

StepanenkoNadejda, 2022-10-15:
> не лучший в объяснении. Мне пришлось бы войти на ваш ноутбук через Интернет и лично убедиться, что происходит, какие порты и т. Д. Перечислены в диспетчере устройств в ваших окнах, если вы хотите, чтобы я это сделал, дайте мне знать.
> 
> Кроме того, возможно, YouTube, скорее всего, лучший выбор для видео.

StepanenkoNadejda, 2022-10-15:
> not the best at explaining I would have to log onto your laptop over the internet and see for myself what is going on and what ports etc is listed in the device manager in your windows if you want [...]

Thank you. I'll try to do myself and if I can't did it I ask you and when you will free time to help me via internet)

AndyIreland, 2022-10-15:
sure @StepanenkoNadejda , have a try yourself but if you do need my help tomorrow Sunday after 4pm (16:00) UK time I will be OK to have a look

AndyIreland, 2022-10-15:
I presume you have specially wired OBD2 splitter cable or specially wired OBD2 pins so that the r-link of your Megane 3 can be read through the OBD2 socket of the car? 

[image]

Also when you pressed virgin or soft reset it is possible that this may have erased settings on the BIC module of the car too.

shrlnm, 2022-10-16:
> Hello, is it possible to do within PyCLIP (on Android)?

You may use qpyren3 instead of pyclip on Android
https://mega.nz/file/93gVGC4C#ViRwRDkHNpIQRpuUMh_xBd2_itgc7ZZcVNZb-dduHOA
This apk don't contains DB. DB you may copy from pyclip.

Any way I suggest to use  laptop and a USB-ELM for configuring ECUs

StepanenkoNadejda, 2022-10-16:
> You may use qpyren3 instead of pyclip on Android https://mega.nz/file/93gVGC4C#ViRwRDkHNpIQRpuUMh_xBd2_itgc7ZZcVNZb-dduHOA This apk don't contains DB. DB you may copy from pyclip.
>
> Any way I suggest [...]

Ooo) Thank you )

AndyIreland, 2022-10-16:
> You may use qpyren3 instead of pyclip on Android https://mega.nz/file/93gVGC4C#ViRwRDkHNpIQRpuUMh_xBd2_itgc7ZZcVNZb-dduHOA This apk don't contains DB. DB you may copy from pyclip.
>
> Any way I suggest [...]

I wouldnt mind giving qpyren3 a try out but I am stuck with the Database , went to PyCLIP - downloaded PyCLIP tar gz  - but what folder contains the Database to get qpyren3 working can you tell me please?

[image]

StepanenkoNadejda, 2022-10-17:
> Я полагаю, у вас есть специально подключенный кабель-разветвитель OBD2 или специально подключенные контакты OBD2, чтобы r-link вашего Megane 3 можно было считывать через разъем OBD2 автомобиля?
> 
> [image]
> 
> Кроме того, когда вы нажали virgin или soft reset, возможно, что это могло привести к удалению настроек и на модуле BIC автомобиля.

> I presume you have specially wired OBD2 splitter cable or specially wired OBD2 pins so that the r-link of your Megane 3 can be read through the OBD2 socket of the car?
>
> [image]
>
> Also when you [...]

Hello. Now I am ordering an ELM327 USB with a chip (PIC18F25K80). I'm waiting for it to come and try to first install PyRen correctly on the laptop (Win 11), then I hope the laptop will detect ELM327 USB), then I'll insert the Macro that you sent and really hope for a miracle) Thanks)

AndyIreland, 2022-10-17:
try this  version of PyREN because its got its own installer and puts a shortcut on your dektop and most importantly included the database in with the pyren installer:

https://cloud.mail.ru/public/91uz/nYXACBs38

StepanenkoNadejda, 2022-10-17:
> > try this version of PyREN because its got its own installer and puts a shortcut on your dektop and most importantly included the database in with the pyren installer:
> > https://cloud.mail.ru/public/91uz/nYXACBs38
> 
> Thank you. Will try)

Please tell me if it is necessary to install on the CAN CLIP (as in this video https://www.youtube.com/watch?v=h5IOZ-skiD8&t=302s )?)

StepanenkoNadejda, 2022-10-17:
> try this version of PyREN because its got its own installer and puts a shortcut on your dektop and most importantly included the database in with the pyren installer:
> 
> https://cloud.mail.ru/public/91uz/nYXACBs38

Please tell me if it is necessary to install on the CAN CLIP (as in this video https://www.youtube.com/watch?v=h5IOZ-skiD8&t=302s )?)

AndyIreland, 2022-10-17:
with PyREN you just use it with a ELM327 diagnostics adaptor (bluetooth/wifi/USB) no CLIP needed - if you need to backup/read/write parameters to the Rlink you have to modify/wire up a Y splitter cable for CAN - H

StepanenkoNadejda, 2022-10-17:
> with PyREN you just use it with a ELM327 diagnostics adaptor (bluetooth/wifi/USB) no CLIP needed - if you need to backup/read/write parameters to the Rlink you have to modify/wire up a Y splitter cable for CAN - H

"...modify/wire up a Y splitter cable for CAN - H" it is mean change connections 6 to 13, 14 to 12?

AndyIreland, 2022-10-17:
> > > "...modify/wire up a Y splitter cable for CAN - H" it is mean change connections 6 to 13, 14 to 12?

yes that is correct. Only if you want to change stuff on the rlink. other ECU's do not need no modification to the ELM327

StepanenkoNadejda, 2022-10-17:
> > > > "...modify/wire up a Y splitter cable for CAN - H" it is mean change connections 6 to 13, 14 to 12?
> 
> yes that is correct. Only if you want to change stuff on the rlink. other ECU's do not need no modification to the ELM327

Cool) Thanks)

StepanenkoNadejda, 2022-10-17:
> > > > "...modify/wire up a Y splitter cable for CAN - H" it is mean change connections 6 to 13, 14 to 12?
> 
> yes that is correct. Only if you want to change stuff on the rlink. other ECU's do not need no modification to the ELM327

One more question. Do I understand correctly. That the file you sent me should be placed in (C:\Program Files (x86)\PyRen\pyren\macro\) and rename it to .txt. Next, in the Marco folder, remove the file init.txt and leave your own (in .txt). When starting the PyRen program, click on Macro?

StepanenkoNadejda, 2022-10-17:
Friends, is it correct that during the procedure of changing parameters and so on through PyRen, the car engine should not be turned on, but only the entire panel is turned on (I have a start button)?

AndyIreland, 2022-10-17:
> One more question. Do I understand correctly. That the file you sent me should be placed in (C:\Program Files (x86)\PyRen\pyren\macro) and rename it to .txt. Next, in the Marco folder, remove the file init.txt and leave your own (in .txt). When starting the PyRen program, click on Macro?

OK, i will see if I can explain, that file I sent you to , to download and put a shortcut on your desktop and includes the database - well when you start that up, the 'MACRO' button will not be there. So go to the Gitlab site and download PyREN from there

https://gitlab.com/py_ren/pyren

 - then unzip the zip file, then copy all the files into the C:\Program Files (x86)\PyRen folder and when it says overwrite then say yes - then when done , click on the PyREN shortcut on desktop and then when it starts up you should see 'MACRO' button as well as a DDT Mode button

AndyIreland, 2022-10-17:
> Friends, is it correct that during the procedure of changing parameters and so on through PyRen, the car engine should not be turned on, but only the entire panel is turned on (I have a start button)?

Do you know what - I don't know what the official like is on it , but I would definitely have the engine running ... I would hate r-link timing out and switching off whilst your in the middle of programming it

StepanenkoNadejda, 2022-10-18:
> OK, i will see if I can explain, that file I sent you to , to download and put a shortcut on your desktop and includes the database - well when you start that up, the 'MACRO' button will not be [...]

Thanks for the explanations) The file that you sent me earlier (https://cloud.mail.ru/public/91uz/nYXACBs38), it already contains the MACRO and DDT buttons)

StepanenkoNadejda, 2022-10-18:
> Do you know what - I don't know what the official like is on it , but I would definitely have the engine running ... I would hate r-link timing out and switching off whilst your in the middle of [...]

Thank you)

shrlnm, 2022-10-18:
Hi. qpyren use ordinary  preparsed and zipped DB with name pyrendata*.zip as in pyren . Just place it in qpython/projects3/qpyren3/pyren3
[image]

shrlnm, 2022-10-18:
> One more question. Do I understand correctly. That the file you sent me should be placed in (C:\Program Files (x86)\PyRen\pyren\macro) and rename it to .txt. Next, in the Marco folder, remove the file init.txt and leave your own (in .txt). When starting the PyRen program, click on Macro?

Place the macro I sent you to the PyRen\pyren\macro folder. 
Do not rename it and do not remove init.txt. Init.txt is a father of all macro 
Engine may be turned ON as so you will configure not engine ECU and also it will ensure good  power during configuration.
The ELM in your case should be re-wired as @AndyIreland  wrote as so the relevant ECU is on the multimedia CAN bus canH="13" canL="12" 

Good luck

StepanenkoNadejda, 2022-10-18:
> Place the macro I sent you to the PyRen\pyren\macro folder. Do not rename it and do not remove init.txt. Init.txt is a father of all macro Engine may be turned ON as so you will configure not engine [...]

Thank you)

StepanenkoNadejda, 2022-10-18:
Hurray! Everything works. Thank you very much)

StepanenkoNadejda, 2022-10-18:
> Hurray! Everything works. Thank you very much)

Everything is in accordance with the instructions. Only when I open the PyRen (https://cloud.mail.ru/public/91uz/nYXACBs38), a MACRO opens there in the format .cmd is not required to translate into .txt.

closed 2022-10-18 by StepanenkoNadejda

AndyIreland, 2022-10-18:
glad you got it sorted / working

StepanenkoNadejda, 2022-10-18:
> glad you got it sorted / working

Thank you for your help) How can I thank you?)

AndyIreland, 2022-10-18:
> > glad you got it sorted / working
> 
> Thank you for your help) How can I thank you?)

That's OK - you done most of the work yourself along with good man @shrlnm offering advice and file .
Its just good you recovered it

StepanenkoNadejda, 2022-10-18:
Thank you all so much for your help!)

AndyIreland, 2022-10-20:
> @AndyIreland
> 
> > I wouldnt mind giving qpyren3 a try out but I am stuck with the Database
> 
> Hi. qpyren use ordinary preparsed and zipped DB with name pyrendata*.zip as in pyren . Just place it in qpython/projects3/qpyren3/pyren3 [image]

Thanks @shrlnm - I think I have got the database working but have hit this hurdle it says 

choose model:

so I put in X10 and it goes choose model again, then I try ZOE and again it says choose model: 
I must be near to it working , it just looks like its missing something.

[image]

shrlnm, 2022-10-22:
@AndyIreland 
it means that the DB is not in appropriate format. 
Which DB did you use?

shrlnm, 2022-10-22:
@AndyIreland 
try this
https://mega.nz/file/4r4DnAAa#eEL9_87BLFFIXcMOgdkqw_1vtbKtUF8_JE2AytMHupE

Baum62, 2022-11-09:
Hello. Please could you help me ? Ich habe problemme mit R-Link  Unconfigured device please add eol settings ". VIN: WME4533421K034605 . Bitte . Ich brauche hilfe . Meine auto Smart 453 .

shrlnm, 2022-11-09:
The vin is absent in db

Zaiac5215273, 2023-03-02:
Hello. Please could you help me ? Ich habe problemme mit R-Link Unconfigured device please add eol settings ". VIN: VF1KT12C652410846