cedricp/ddt4all #738: Pb config R2_08_v3_81

BuM3GT, 2022-10-18:
Bonjour à tous,

J'ai besoin d'aide pour la configuration de R2_08_v3_81 avec Tomtom
L'autoradio ne fonctionne plus suite à une mauvaise manip avec DDT4all et un ELM BT.
Les ECU sont bien reconnus en CAN2.
J'ai réussi à faire disparaitre le message "connect diag unit" mais aucune commande de l'autoradio ne répond.
J'ai bien essayé de faire la configuration acoustique (est-ce les bonnes valeurs que j'ai?), vérifier le VIN et CRC, configurer le NFA mais rien ne change...je suis perdu.
Si une âme charitable peux m'aiguiller, je lui serait très reconnaissant.

BuM3GT, 2022-10-20:
Hi @AndyIreland, sorry for my basic english...
Can you help me?
I'm trying to configure my R2-08_v3_81 ECU, here are the values ​​entered from a macro for Pyren but I doubt the composition...
#Radio reception
2E2B2D00011E1422241A14280D0A081F0F020206020408300301010001001430121220121400800100640A050F011402080A043C0F0F140F321E3250050F0009120A8CE10A5A5A0384300101063C961401C80000050000000100000000
#Accoustic Whole car
2E2B2C3E9B522F0001FFFAFDDA0164FE002C973E9B511E0001FFF4FDD302CCFC00164C00003E9B732D0001FFF4FE7803E0FE002C973E9B5B490001FFECFE1104FFFC00164C0000000040000001000000004000C0004000000040000001000000004000C0004000000040000001000000004000C0004000000040000001000000004000C00040000000400040000001000000004000C0004000400040000001000000004000C0004000400040000001000000004000C0004000400040000001000000004000C000400000002D43400000010000FFE9006FFF8021043FE8400000010000FFD200EFFF0014F63FDA40000001FFFFFFB5018EFE001C223FBE400000010044FF7D0044FF8041C23F0140000001FFE2FE1507A1F80024513DB440000001FF88FBBE0D0EF00024C900003D5B40000001FE33FB362B9EE000182B3DD740000001FEB6FC2E262BC0000EC836A140000001F0F8F1D166ABC0001F234F8C400000010BC6FC463076E000547B3FE440000001004BFFCB0122FF005FD73FE640000001FFFFFFCE03B8FE0017BE00002C4740000001000000004000C00040003FEC400000010000FFD90167FF00166C3FF5400000010000FFEC0052FF8017093DC940000001FF98FBD30BC3F0001CFF00003DB040000001FEE1FBA01F18F0001B043E1040000001FED3FC6B26CAE00013293FBE400000010044FF7D0044FF8041
#Accoustic Driver
2E2B2B3E9B522F0001FFFAFDDA0164FE002C973E9B511E0001FFF4FDD302CCFC00164C00003E9B732D0001FFF4FE7803E0FE002C973E9B5B490001FFECFE1104FFFC00164C0000000040000001000000004000C0004000000040000001000000004000C0004000000040000001000000004000C0004000000040000001000000004000C00040000000400040000001000000004000C0004000400040000001000000004000C0004000400040000001000000004000C0004000400040000001000000004000C000400000002D43400000010000FFE9006FFF8021043FE8400000010000FFD200EFFF0014F63FDA40000001FFFFFFB5018EFE001C223FBE400000010044FF7D0044FF8041C23F0140000001FFE2FE1507A1F80024513DB440000001FF88FBBE0D0EF00024C900003D5B40000001FE33FB362B9EE000182B3DD740000001FEB6FC2E262BC0000EC836A140000001F0F8F1D166ABC0001F234F8C400000010BC6FC463076E000547B3FE440000001004BFFCB0122FF005FD73FE640000001FFFFFFCE03B8FE0017BE00002C4740000001000000004000C00040003FEC400000010000FFD90167FF00166C3FF5400000010000FFEC0052FF8017093DC940000001FF98FBD30BC3F0001CFF00003DB040000001FEE1FBA01F18F0001B043E1040000001FED3FC6B26CAE00013293FBE400000010044FF7D0044FF8041
#Phone acoustic parameters
2E2B250101705AC7400401701800029402950064E0A00073C0003333030B00005AA10000000810082E2D38DD6A86000000000FFF00310004B0239C31E6E0B775239C6798E6E039961F941E0E3D36B70CD6A096531E0E76B1FF80B76A37F6CA7D3EB7412E37F6403B070461FD7F40FFFF07043B0061FD900000000000000502950000747EC3402C00CCCC0003027F8DCEFFFF000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
I will receive a USB ELM so as not to saturate the BT one.
I can give you my VIN if you want.

BuM3GT, 2022-10-20:
@AndyIreland , thanks for your fast answer
I don't work with Pyren but DDT4all, I extract the value from a macro and try to write with DDT4all.

BuM3GT, 2022-10-20:
Hi @srlnm
I saw you have helped somebody for the same problem #376 
Can i take a part of your time to resolve mine?

BuM3GT, 2022-10-25:
I am progressing slowly in my research and my tests but my problem persists.
The Cbox and the Tomtom work but I have no sound or radio...message "connect diag tools"
I get fault code DTC934155 (DF030) in ECU R208_X95_X62_X61_0000_64_B and can't write VIN and CRC anymore.
Can anyone tell me how to get this all back up and running? @shrlnm ?

shrlnm, 2022-10-25:
@BuM3GT Hello. What is your VIN?

BuM3GT, 2022-10-25:
My VIN is VF1KZ1T0644629491
Thanks

BuM3GT, 2022-10-25:
@shrlnm 
thank you, I will try this macro tonight.

BuM3GT, 2022-10-25:
@shrlnm 
The vehicle configuration has been taken into account but the acoustic part blocks writing.
The VIN and CRC is correctly registered.
Here are the log files, I don't understand why writing is not possible.

[ecu_log.txt](https://github.com/cedricp/ddt4all/files/9863671/ecu_log.txt)
[elm_log.txt](https://github.com/cedricp/ddt4all/files/9863672/elm_log.txt)

shrlnm, 2022-10-26:
@BuM3GT Thank you for logs.
Actually commands was not implemented correctly. 
The problem is in slow response of ELM
The adaptive timing algorithm do not decrease waiting timeout. I don't know why

>[19:02:43.853]105D2E2B2D00011E
<[**0.297**]105D2E2B2D00011E
3001008181818181

It takes about 297 milliseconds to send every part of long command. As a result whole command is sending about 10 seconds and it is too long for ECU.

I may try to correct the pyren  a bit for decreasing the timer by software.

BuM3GT, 2022-10-26:
Thanks for your return @shrlnm 
I have an OBD link SX, do you think it will resolve the problem?

shrlnm, 2022-10-28:
@BuM3GT Let's try. Send me the new logs of implementing OBD link SX please.

BuM3GT, 2022-10-28:
@shrlnm you are The Men of the situation!!!
It worked with the OBDlink SX, writing speed is faster.
I do not know how to thank you
The car was in the Renault garage and even the technician was unable to find a solution
[](url)
[ElmOBDLinkSX.txt](https://github.com/cedricp/ddt4all/files/9890448/ElmOBDLinkSX.txt)

shrlnm, 2022-10-30:
Ok) Thank you for log.

closed 2022-10-30 by BuM3GT

casianb2, 2024-01-31:
> @BuM3GT [link removed]

hello. I've got the same problem. Can you provide the config macro again please? VIN: VF1VY5KGNUC448306

shrlnm, 2024-01-31:
> I tried to ask the pyren telegram bot for the macros, but he did not recognise my VIN

I use exactly the same database, and so if the bot can't find your vin, I won't be able too ((
In which year your car was manufactured?

casianb2, 2024-01-31:
> > I tried to ask the pyren telegram bot for the macros, but he did not recognise my VIN
> 
> I use exactly the same database, and so if the bot can't find your vin, I won't be able too (( In which year your car was manufactured?

its a renault koleos from 2012

casianb2, 2024-01-31:
i screwed with the ECU configs and now I don't even know what equipment the car has, to be able to correclty set the config. Is there some way to find out what equipment the car has? Specifically, I don't know what to set into ECU Configuration.SWRCtype. 
[image]
In Multimedia Interface Unit also the same problem: PTT, President

shrlnm, 2024-01-31:
> its a renault koleos from 2012

Unfortunately the H45 platform  is absent in reprog DB. 
The most realistic way is to ask someone with similar car to get a dump of configuration and configure your unit with that dump

casianb2, 2024-01-31:
> > its a renault koleos from 2012
> 
> Unfortunately the H45 platform is absent in reprog DB. The most realistic way is to ask someone with similar car to get a dump of configuration and configure your unit with that dump

damn... so unfortunate. Anyway, thanks so much for helping. I love your work. Keep on keeping on

shrlnm, 2024-02-01:
@casianb2 
there is another way, but it is more difficult
R2_08 is installed on such platforms

"X38"
"X43"
"X47"
"X61"
"X62"
"X91"
"X95"

Using pyren3/doc_maker and the allvin option, you can create tables of all VINs for these platforms. From the tables, you need to select the vin configuration of which is as similar as possible to yours. Make macros for the selected vins and try to apply them.
For searching in these tables you need to use the data shown by pyren3 in menu "ACI : Auto Config Info" like here

https://github.com/cedricp/ddt4all/issues/815#issuecomment-1527890620

casianb2, 2024-02-01:
:O that is a great idea. i'll try it later today.