cedricp/ddt4all #739: using DCM12_L4CB1_v9_3_2009...xml does not read any data

octavp, 2022-10-19:
I am trying to read ECU data (or any data for that matter) from the in title mentioned circuit on a Renault Symbol under Window 10. I see in the logs that the BUS init returns OK but afterwards any screen I try (for example VIN frame) returns NO DATA. Any suggestion on how to debug further?

KarelSvo, 2022-10-19:
Where are your log files?

octavp, 2022-10-19:
[ecu_ddt_snippet_01.txt](https://github.com/cedricp/ddt4all/files/9824012/ecu_ddt_snippet_01.txt)
[elm_ddt_snippet_01.log](https://github.com/cedricp/ddt4all/files/9824014/elm_ddt_snippet_01.log)
[screens_snippet_01.txt](https://github.com/cedricp/ddt4all/files/9824016/screens_snippet_01.txt)
log files attached

KarelSvo, 2022-10-20:
The log says Elm doesn't give a positive response when requesting the Fast Init routine. (AT FI = ?)
No Elm test to see. The answer to AT PPS would be interesting.
No scanning for ecus seen. It would be interesting to see if another ecu responds positively.
If you are using the Python 3 version, I recommend testing the Python 2 version or Pyren. The Python 3 version is not entirely bug-free. https://www.drive2.ru/users/zmiterm/

octavp, 2022-10-20:
I am using python 3.8 (which comes as packaged together with ddt4all release). is there a guide somethere on how to change the python version? I tried using 3.10 python but without success so far. can I send the AT PPS (what does it stand for) request by hand somehow?

octavp, 2022-10-20:
I used pyren and when scanning the x35 ECU it checks every port speed up to 2000000 and it comes back with the ELM not responding. I start to believe that my ELM is broken.

octavp, 2022-11-01:
I tried the pyren_183 and 184 installer from the link you mentioned above and the result was a lot of timeouts during the scanning process (33 error timeout and 30 error question). For the slow init the timeouts were 89. While the car was running I saw 3 or 4 green leds coming on for some small amount of time (I associated them with the scanning of the 35 ecus pyren had in its database). shall I conclude that my ELM is broken?

octavp, 2022-11-01:
would you be able to share  a link to one ELM 327 that works for you?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif