cedricp/ddt4all #740: No picture from rear camera on RLink(( Please help the experts)

StepanenkoNadejda, 2022-10-19:
Hello,
Who faced the situation. I connected the rear-view camera to the RLINK (yellow and black wires) Megane 3, activated it via ddt4all, the camera is connected to the rear lights. The camera turns on, but there is no picture on the screen. I've heard what additional configurations need to be installed. It's possible?

AndyIreland, 2022-10-19:
have you wired up the composite video in? :

[image]

AndyIreland, 2022-10-19:
Did you reboot the rlink after making your changes?

AndyIreland, 2022-10-19:
I wonder if the the wires under the seat are connected up to the back of the radio - in any case i thought the wire for the reverse camera were up  in the footwell area , not under the seat? 

there is also a CVBS (video in) ground as well on pin 19

StepanenkoNadejda, 2022-10-19:
> I wonder if the the wires under the seat are connected up to the back of the radio - in any case i thought the wire for the reverse camera were up in the footwell area , not under the seat?
> 
> there is also a CVBS (video in) ground as well on pin 19

Yes. 19 and 21, they are the same turns out (grounding).

StepanenkoNadejda, 2022-10-19:
[image]
[image]

StepanenkoNadejda, 2022-10-19:
[image]
[image]

AndyIreland, 2022-10-19:
if that were me i would pull off the trim , take out the rlink screen and put the wires straight to the back of the multi connector and see what happens. - its a good sign the screen goes black but backlit when you put car in reverse .
Can you see any reversing guidelines on the screen?

StepanenkoNadejda, 2022-10-19:
> if that were me i would pull off the trim , take out the rlink screen and put the wires straight to the back of the multi connector and see what happens. - its a good sign the screen goes black but backlit when you put car in reverse . Can you see any reversing guidelines on the screen?

I also connected the R Link to the screen. Nothing. When the reverse speed is turned on, a black screen with guide lines appears, but there is no picture.

AndyIreland, 2022-10-19:
and you are absolutely sure the camera is working - if the camera is 12v , try putting the positive up to a constant positive feed and not on the reverse light bulb socket

StepanenkoNadejda, 2022-10-19:
> and you are absolutely sure the camera is working - if the camera is 12v , try putting the positive up to a constant positive feed and not on the reverse light bulb socket

Thank you! I'll try for the sake of experiment, maybe something will change. But as the camera has LEDs, they will be constantly lit as soon as the car will turn on.

AndyIreland, 2022-10-19:
> Thank you! I'll try for the sake of experiment, maybe something will change. But as the camera has LEDs, they will be constantly lit as soon as the car will turn on.

are they infra red LED's though - the ones that help the camera see in the dark? - if they are you cannot see them lit up just with eye, but if you point your phone camera at them you will see them lit up

StepanenkoNadejda, 2022-10-20:
No, I have a camera with ICE lighting, not infrared. Does it matter if the camera is with an ССD matrix or without an ССD?

StepanenkoNadejda, 2022-10-20:
[image]

AndyIreland, 2022-10-20:
so , when you have it wired up to the reverse bulbs when you put the car into reverse the LED lights are lit? - or do they only light up when its dark outside?

StepanenkoNadejda, 2022-10-20:
> are they infra red LED's though - the ones that help the camera see in the dark? - if they are you cannot see them lit up just with eye, but if you point your phone camera at them you will see them [...]

No, I have a camera with LED's lighting, not infrared. Does it matter if the camera is with an ССD matrix or without an ССD?

StepanenkoNadejda, 2022-10-20:
> so , when you have it wired up to the reverse bulbs when you put the car into reverse the LED lights are lit? - or do they only light up when its dark outside?

LED is constantly glowing when I turn on the reverse gear. It doesn't matter day or night, since there is no light sensor on the camera. I also tried to connect the other 2 cameras. Nothing(

StepanenkoNadejda, 2022-10-20:
> LED is constantly glowing when I turn on the reverse gear. It doesn't matter day or night, since there is no light sensor on the camera. I also tried to connect the other 2 cameras. Nothing(

They also write that you need to configure: multimedia, a parking sensor unit — so that the car understands what we want from it. To do this, the record of the type of your parking sensors (standard for example RDAVAR, RDPRAR, SRDPRO) should simply be replaced with RDAR01.

StepanenkoNadejda, 2022-10-20:
[image]

StepanenkoNadejda, 2022-10-20:
> [image]

But I don't know how do it?)

AndyIreland, 2022-10-20:
> LED is constantly glowing when I turn on the reverse gear. It doesn't matter day or night, since there is no light sensor on the camera. I also tried to connect the other 2 cameras. Nothing(

Can you make a screenshot of your settings for these parameters:

[image]

AndyIreland, 2022-10-20:
and a screenshot of yours for this:

[image]

StepanenkoNadejda, 2022-10-21:
[image]
[image]

StepanenkoNadejda, 2022-10-21:
> and a screenshot of yours for this:
> 
> [image]

Yes, of course)

StepanenkoNadejda, 2022-10-21:
> Yes, of course)
> 
> [image] [image]

Yes, of course)

[image]
[image]

AndyIreland, 2022-10-21:
thanks for that , looks all ok - have you got reverse parking sensors on your car? 

Also where it says trunk handler message set that to not fitted and  the trunk open handler disable rvc set that to not fitted

StepanenkoNadejda, 2022-10-21:
> thanks for that , looks all ok - have you got reverse parking sensors on your car?
> 
> Also where it says trunk handler message set that to not fitted and the trunk open handler disable rvc set that to not fitted

Yes, I have rear parking sensor).

StepanenkoNadejda, 2022-10-22:
> thanks for that , looks all ok - have you got reverse parking sensors on your car?
> 
> Also where it says trunk handler message set that to not fitted and the trunk open handler disable rvc set that to not fitted

can there be a decoder in the car that does not allow the image to be displayed on the screen? 3 cameras tried to connect. Nothing(

BarektE, 2023-02-19:
> [image]

This camera does not work on R-Link.

AndyIreland, 2023-02-19:
> > [image]
> 
> This camera does not work on R-Link.

why not? - are you giving it 12v and not the 6v supply from the rlink?

Furtif, 2023-06-02:
good post need a ref .. but clean in ... please refer it at discutions 
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Dorinmomo, 2024-05-16:
I have the same problem with Rlink 1, no image from rear cam. Did anyone fixed it? Can you tel me how?

Dorinmomo, 2024-05-17:
I'm back !! I did it !!! OMG !! I have image on my rear camera. R link 1 renault captur. Activated it with ddt4all , bought a chinese connection cable  that said "for renault" but it did not work.... I thought it was the camera, tested all the wires, 👎 .... The thing was that the wires where in the wrong place(pins). I did not check if it was on the 18, 19 pins. Took all out and reconect the wires and it worked. The camera was aftermarket, like in the pictures. 
[image]
Thank you for posting here .

Dorinmomo, 2024-05-18:
> I wonder if the the wires under the seat are connected up to the back of the radio - in any case i thought the wire for the reverse camera were up in the footwell area , not under the seat?
> 
> there is also a CVBS (video in) ground as well on pin 19

Probably the wires are not up to the screen, mine had not. You need to go with the wires up to the back of he screen. That's why you do not have image. 
[image]
[image]
[image]
[image]
[image]

AndyIreland, 2024-05-18:
if any help to anyone here is wiring diagram to the back of the MFD (multi function touchscreen display) 
[image]

R-3MY, 2024-06-09:
> I'm back !! I did it !!! OMG !! I have image on my rear camera. R link 1 renault captur. Activated it with ddt4all , bought a chinese connection cable that said "for renault" but it did not work.... I thought it was the camera, tested all the wires, 👎 .... The thing was that the wires where in the wrong place(pins). I did not check if it was on the 18, 19 pins. Took all out and reconect the wires and it worked. The camera was aftermarket, like in the pictures. ![17159652722611039177459834047726](https://private-user-images.githubusercontent.com/170031235/331677223-2ada513b-92e0-407a-9436-0dea787225c6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5Mjg2MTgsIm5iZiI6MTcxNzkyODMxOCwicGF0aCI6Ii8xNzAwMzEyMzUvMzMxNjc3MjIzLTJhZGE1MTNiLTkyZTAtNDA3YS05NDM2LTBkZWE3ODcyMjVjNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOVQxMDE4MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hY2I2ZjZiMzI5OWRkNDhiMDkyZjE1MGM0ZTQ2YWVjYzljNjQzMjQ1ZWQ1MTJlZjAxOWNiNjM4ZjM2ZDliMjljJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.R2KvuBtsqQt3yxsCRmI-2N9O5nO7ir5_C7keNNLh0Gg) Thank you for posting here .

Thank you, you literally saved me!

I tested several 12v cameras and I still have the same problem (black screen) however when I put my finger in front of the camera I do have a "display" only it seems to be very noisy. I briefly see guide lines but no display of the image that should come from the camera.

I have a Kangoo II ph 2 with rear sensors.

Regards, Rémy.