cedricp/ddt4all #743: ECU tweaker source code

cedricp, 2022-10-20:
Hi all,
I decided to open source the Android port - **ECU tweaker**, you can find repo here :
https://gitlab.com/DrQuark/android_can
I have no time for it anymore.

JABICHOCU, 2022-10-20:
This has been a very useful app for me...thanx for all your efforts :)

David050893, 2023-01-30:
hello... thank you very much in advance for contributing your knowledge to the community,

on the other hand I don't know if this is an appropriate question,
I am doing a university work and I have used much of your material specifically the ddt4all repository, which I hope to share if all goes well, only I still have doubts and I would like you to help me clarify them, because after identifying the variable in the hecadeximal data string in a PID request I have had difficulty seeing the value in our decimal system, is there a specific formula or what should I do, you could guide me please ...

cedricp, 2023-01-30:
Please describe what's not clear, I'll try to explain as much as I can

AlMeu, 2023-01-31:
Where can be found this edition mode?

cedricp, 2023-01-31:
Select **edition mode** from the first window (ELM selection one)

AlMeu, 2023-01-31:
In my 5.5.0 isn't
[image]
There is another version?

cedricp, 2023-01-31:
Latest verion : 

[image]

Demo should work, though

AlMeu, 2023-01-31:
There is somehere this last version for download?

AlMeu, 2023-01-31:
I downloaded latest version v2 and try to run in Win10 x64 with OBD EX interface.
I installed the drivers for this and run it in connected mode, but in scan didn't detect anything.
I tried also in Edition mode and I have got this fault:
[image]

David050893, 2023-02-01:
Thank you very much for your help, I really appreciate it!!!

AndyIreland, 2023-02-01:
why am i seeing ECU tweaker source code in the title ... but its DDT4ALL ???

cedricp, 2023-02-01:
Check the link in the first post, I released ECU tweaker source code

AlMeu, 2023-02-06:
I managed to make start Edition mode but when I scan I get this error and quits:
[image]
Same issue also with same version in Connected mode.
Any ideea?

Sartascn, 2023-03-13:
Complimenti per il progetto Ecu Tweaker. Riesco a leggere tutti i parametri ma, non riesco a modificarli. C'è qualche suggerimento utile? Grazie

Sartascn, 2023-03-13:
Scusami Cedric... Quando avvio Ecu Tweaker non mi compare l'icona del lucchetto. Di conseguenza, non posso inviarti il codice di attivazione. Potresti spiegarmi meglio? Grazie infinite!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Larrypao1, 2023-11-18:
hola cedric me gustaria poder tener la licencia de ecu tweaker y gracias por tu gran aporte

billigizer, 2024-03-19:
Hi guys, i just found out about ecu tweaker. Is it to late to try? Is the elm dongle wifi/Bluetooth still compatible? I bought a autel device for 1 100 dollars just to find out this, which is almost free, is more useful for me so i feel devestated and happy at the same time. I guess money can't always buy happiness 🙁😂 thank you in advance Cedric, what amazing work (the part that is understandable for me)