cedricp/ddt4all #745: ESCL present change error

burakcankose, 2022-10-25:
Hi, I have clio 4 phase 2 vehicle. the steering wheel failed and we realized that it was from the UPC control unit and we changed this control unit. After about 4 months, it gave this error again and the vehicle does not work (by the way, I am using an emulator instead of a lock) right now I do not want to change the control unit again and I want to turn off the steering column lock system from the HFM module. When I try this with clip, I get programming error and command impossible warnings. When I want to do it with ddt4all, I get a programming error as in the picture. How can I find a solution about this? If I make the change from the MTC file and write the setting again, will it be a solution? Also, does anyone know the code for this option in the MTC file? My car is not working right now, I am waiting for your answers.

AndyIreland, 2022-10-25:
have you tried using a different session setting instead of 10C0 say changing the settings in aftersales session I dont know if it will work or not and also you can try adjusting CAN timeout to 600ms see if that helps

burakcankose, 2022-10-25:
> have you tried using a different session setting instead of 10C0 say changing the settings in aftersales session I dont know if it will work or not and also you can try adjusting CAN timeout to 600ms see if that helps

I tried different session but not working, How can change CAN timeout ?

AndyIreland, 2022-10-25:
this down here at bottom of page:

[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif