cedricp/ddt4all #746: RLINK 2 SCREEN STUCK BLACK WITH BUTTONS

fraan15, 2022-10-25:
Hi, since two months ago , im having problems with my rlink 2 , the first problem is the screen still on black, but i could resolve this after 3-4 days and the screen waked up again , then i update to new version and all was working good, but at the night , the screen reset and freeze on menu, and now my rlink screen is on and i can see the menu , but if i press any button, it stay freeze and dont work correctly , i have all logs of the rlink but i dont see anything, if anyone can help me ... i can pay if someone help me to resolve, thank u guys.

attached video of the problem and the folder with the logs

[logfiles_mm2014.zip](https://github.com/cedricp/ddt4all/files/9863772/logfiles_mm2014.zip)

https://user-images.githubusercontent.com/102795195/197873510-45b652c3-fae8-403b-8653-6aa39b6a1dfe.mp4

rttr67, 2022-10-26:
Hello, I will advise only by e-mail

fraan15, 2022-10-26:
> Hello, I will advise only by e-mail

franredondo.informatica@gmail.com

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif