cedricp/ddt4all #747: R-LINK antitheft activated

kovo5555, 2022-10-31:
I have pyren, ddt4all and alot of chip reading tools.
Renault Scenic IV, rlink is the horizontal one
I had to fit another unit because my dashboard was flashing both colors, no image on rlink display no matter what I did and there was no sound. 
Now another unit I installed and it has antitheft activate with padlock on screen. 
I can read eeproms or whatever chips you need me to or solder pins. Tell me what to do to fix this, please I'm desperate. Thank you.

kovo5555, 2022-10-31:
my email is kovo5555@gmail.com

closed 2022-10-31 by akakv123

shrlnm, 2022-11-01:
@akakv123  are you able to get access to the android file system? You need to delete /data/system/acode.dat file.