cedricp/ddt4all #752: Renault talisman: screen, radio not working what to do ?

talisman0007, 2022-11-16:
Hi! Suddently, the emergency download message popped up on my Renault talisman screen. I figured that need to update the system through the official website, using a USB. While doing this the system just showed “downloading: 4 min left” for about 2 hours. I had to turn of the car, and since the screen is black and nothing works. I can’t seem to figure why this has happened. Some threads link the issue to a burned screen unit, but how is this possible and what the best and easiest fix for this? Any ideas ?

rttr67, 2022-11-17:
Hello, I will advise only by e-mail

talisman0007, 2022-11-17:
Hello,

Yes please!

rttr67, 2022-11-17:
email

talisman0007, 2022-11-17:
My mail is: ***@***.***

Thank you !

rttr67, 2022-11-17:
My mail is: ***@***.***       email I don't see mail

talisman0007, 2022-11-17:
Oh sorry ill try again: 

316349@via.dk

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif