cedricp/ddt4all #753: How to run on Mac OS without issues

th3mrx, 2022-11-17:
**Step 1 :** Install Python 3 latest version from the official website : [](https://www.python.org/downloads/macos/)
**Step 2 :** Open Terminal and type this commands

pip3 install PyQt5
pip3 install PyQt5-sip
pip3 install PyQtWebEngine

**Step 3 :** Edit the file "options.py"  located inside the folder "ddt4all-2.0"
in the line 51 you will find this line of code : 

t = gettext.translation(filename, 'locale', fallback=True, codeset="utf-8")

**change it to :** 

t = gettext.translation(filename, 'locale', fallback=True)

**Step 4 :** Save and open the folder "ddt4all-2.0" inside the terminal and then type : 
python3 ddt4all.py

**The software should popup in a minute without any issues.
GG**

tinbert, 2022-11-27:
Thanks a lot, th3rmx! 
It starts w/o problems on my Intel-Mac, however this won't work on M1 Mac.
It looks like there are some binaries necessary which do not run on M1, so there should be a way to run it under Rosetta, though I wasn't successful. (see https://stackoverflow.com/questions/68317410/how-to-install-pyqt5-on-macos)

Furtif, 2023-02-21:
This is solved in https://github.com/cedricp/ddt4all/pull/779

Furtif, 2023-02-21:
Sync you clone repo and close issue

bruvv, 2023-04-07:
Still an issue @Furtif 
```
Traceback (most recent call last):
  File "/Users/niels/Documents/scripts/ddt4all/main.py", line 18, in <module>
    import dataeditor
  File "/Users/niels/Documents/scripts/ddt4all/dataeditor.py", line 7, in <module>
    import ecu
  File "/Users/niels/Documents/scripts/ddt4all/ecu.py", line 12, in <module>
    import elm
  File "/Users/niels/Documents/scripts/ddt4all/elm.py", line 19, in <module>
    _ = options.translator('ddt4all')
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/niels/Documents/scripts/ddt4all/options.py", line 52, in translator
    t = gettext.translation(filename, 'locale', fallback=True, codeset="utf-8")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: translation() got an unexpected keyword argument 'codeset'
```
with a fresh pull

nmschulte, 2023-05-30:
Also on GNU/Linux.

`python3 --version`
```
Python 3.11.2
```

````
Successfully installed PyQt5-5.15.9 PyQt5-Qt5-5.15.2 PyQt5-sip-12.12.1 PyQt5-stubs-5.15.6.0 PyQtWebEngine-5.15.6 PyQtWebEngine-Qt5-5.15.2 crcmod-1.7 pyserial-3.5 pyusb-1.2.1
Runs app ...
Traceback (most recent call last):
  File "/home/nmschulte/src/ddt4all/./main.py", line 18, in <module>
    import dataeditor
  File "/home/nmschulte/src/ddt4all/dataeditor.py", line 7, in <module>
    import ecu
  File "/home/nmschulte/src/ddt4all/ecu.py", line 12, in <module>
    import elm
  File "/home/nmschulte/src/ddt4all/elm.py", line 19, in <module>
    _ = options.translator('ddt4all')
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nmschulte/src/ddt4all/options.py", line 52, in translator
    t = gettext.translation(filename, 'locale', fallback=True, codeset="utf-8")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

InsanityM, 2023-06-02:
Hi, i have the same problem like @nmschulte . I can't run the program. I use ubuntu 23.04.

Furtif, 2023-06-02:
> Still an issue @Furtif
> 
> ```
> Traceback (most recent call last):
>   File "/Users/Documents/scripts/ddt4all/main.py", line 18, in <module>
>     import dataeditor
>   File "/Users/Documents/scripts/ddt4all/dataeditor.py", line 7, in <module>
>     import ecu
>   File "/Users/Documents/scripts/ddt4all/ecu.py", line 12, in <module>
>     import elm
>   File "/Users/Documents/scripts/ddt4all/elm.py", line 19, in <module>
>     _ = options.translator('ddt4all')
>         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/Documents/scripts/ddt4all/options.py", line 52, in translator
>     t = gettext.translation(filename, 'locale', fallback=True, codeset="utf-8")
>         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> TypeError: translation() got an unexpected keyword argument 'codeset'
> ```
> 
> with a fresh pull

🈂️ odd normarly ok

Furtif, 2023-06-02:
rapport all questions at https://github.com/cedricp/ddt4all/discussions, because bugs py etc ins not bug as code

closed 2023-06-02 by Furtif