cedricp/ddt4all #754: Changement de VIN dans l'UCH

Tomag1975, 2022-11-17:
Hello to everyone! Please help me to find a solution to renew BCM or change the VIN on my BCM Dacia Sandero III (Magneti Marelli).
I have seen here on the site some print screen with DDT4 in edit mode procedure, but after putting all the necessary string (reference code followed by VIN and then CRC) it gives me the error that the ECU is not unlocked!
Is it possible that I need a security code to unlock it?

Thank you and I look forward to your reply!

Best regards,

Georgetoma1975@gmail.com

KarelSvo, 2022-11-17:
Where is the photo of your error message?

Tomag1975, 2022-11-18:
Hi, tomorrow I'll try again and I'll come back with the printscreen with the error that appears after sending the request.

Thank you,

AlMeu, 2023-01-03:
> Hello to everyone! Please help me to find a solution to renew BCM or change the VIN on my BCM Dacia Sandero III (Magneti Marelli). I have seen here on the site some print screen with DDT4 in edit mode procedure, but after putting all the necessary string (reference code followed by VIN and then CRC) it gives me the error that the ECU is not unlocked! Is it possible that I need a security code to unlock it?
> 
> Thank you and I look forward to your reply!
> 
> Best regards,
> 
> [Georgetoma1975@gmail.com](mailto:Georgetoma1975@gmail.com)

As I know there is CAN Gateway in network and can't write anything without unlock the CGW.

Tomag1976, 2023-02-12:
Hello AlMeu

> As I know there is CAN Gateway in network and can't write anything without unlock the CGW.

Hello AlMeu,and what would be the solution, or do I have to get a security code from the dealership?

AlMeu, 2023-02-12:
I never changed VIN of BCM, but if this behind a Gateway, must unlock it first and for this need Renault token/dongle for sure.
There are several ECU where can't be changed the VIN and I think the BCM is one of these.
Now depends about car we spoke. In Sandero III is for sure.

Tomag1976, 2023-02-12:
hi,
I have Renault Token but in Renault Clip have one  single choice: to secure the gateway not unsecure or unlock it!

AlMeu, 2023-02-12:
Then you don't have enough rights to use that token.

Tomag1976, 2023-02-12:
Possible....

Tomag1976, 2023-02-12:
I had a new BCM that got erased when writing and then stopped communicating at all.It's only my fault that I didn't write via LAN and only via WIFI and because I didn't have good internet traffic it stopped writing and then froze and I couldn't "resuscitate" it.
The token/dongle is fully licensed.

Tomag1976, 2023-02-12:
The Clip platforme is limited, the token unlocks modules for reading and configuration, but there is no possibility to code/unlock GW with the Clip.
I can write or reconfigure some modules online,but the BCM is from another car and has a different VIN no.,that's why the problem is that the media nav. tablet has the anti-theft turned on.

AlMeu, 2023-02-12:
No one process from any SW of writting configuration is erasing all in ECU before write a configuration as I know.
Anyway via WiFi is dangerous configure anything. Now you know it.
I agree that Clip is limited, but with enough rights have enough power to rewrite configurations and it is designed to do minimal damaged even if you have minimal knowledge.
The MediaNav it is the last problem. With this I can help you. This I can fix it if I have the unit on my table. 
The main problem is BCM which need to integrate with all other ECU's. And anyway I think VIN in BCM you can't change.

Tomag1976, 2023-02-12:
On this Sandero 3, the navigation media consists of a tablet and a unit, which one should I send, if you can help!?

On the BCM I also have the keys, and on the ECU injection I do Immo Off (EMS3140) with emulator, until a solution is found to rewrite the VIN.

Regards,

Tomag1976, 2023-02-12:
I also have knowledge of electronics, I did change the controller from the old BCM to the second hand one, but it seems to have been destroyed by water and I have no possibility to find the type of controller (it has no data to match what is on the market) and no writing software for booting.

Tomag1976, 2023-02-12:
I talked to some of the colleagues on the MHH Auto forum (Automotive/Gabi Gafu) and they couldn't help me, although I sent them the BCM, they didn't manage to connect it (the second hand one that works).

Maybe in the future they will implement in the RLk program, for now Abritus says they can virginize and rewrite VIN on such BCM, but it is not worth investing for one more car like this!
I have already tried with what I have:RLK V1.99 and Easy Renault V.x10,besides Clip V224 and VCI (RNM) with token full licence(from a friend,I have also but limited),Pyren,DDT 4.0.9.2,DDT4 All and Autel IM508,but if you don't have access to information and no advanced programming knowledge it's quite difficult!

AlMeu, 2023-02-12:
In Sandero 3 the tablet as you call, have attached the back and it is a simple unit actually with the screen attached like all MediaNav's. That you need send it.
I have sent you an email you shown up, I think you are the same guy @Tomag1975 & @Tomag1976. Check the SPAM folder.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif