cedricp/ddt4all #756: Black Screen Renault Koleos 2018 Full Full

Youssef-711, 2022-11-23:
Can you help me 
Car model Renault Koleos 2018 Full Full 
I turned on Nervision mode 
The screen is black
VF1RZG007JC301840 
![16691684193407822674649682441085](https://user-images.githubusercontent.com/118640514/203455374-d60a88af-41cc-4889-b40d-8b577db
[image]
68e56.jpg)

poli27, 2022-11-24:
pyren ;)

poli27, 2022-11-24:
https://www.blogarama.com/technology-blogs/1353906-medianav-skins-maps-wince6-micon-manager-ddt4all-navitel-igo-7way-sevenway-hacks-supermod-mn1-mn2-mn3-dacia-renault-blog/43071970-link-black-screen-repair

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif