cedricp/ddt4all #762: Edit and save problem ?

EnisBerke, 2022-12-26:
-When I log in to add hours and degrees to the dial via ddt4all tdb-ucm on my -2022 Sandero Stepway, I see the vehicle information appearing, but when I edit and print, nothing changes. How can I fix this?
- Is the Sandero Stepway 3 obd connection the same as Clio 5?

AndyIreland, 2022-12-26:
more than likely protected by a locked security gateway stopping any parameters being changed :

https://github.com/cedricp/ddt4all/issues/404

EnisBerke, 2022-12-27:
can we use a bypass cable for locked security gateway?

AndyIreland, 2022-12-27:
the link here will be of help:

[https://github.com/cedricp/ddt4all/issues/404](url)

EnisBerke, 2022-12-27:
Is it available for Sandero Stepway 2022. #404 for Clio. I didn't see any information about Sandero Stepway 22 if I'm not mistaken.

AndyIreland, 2022-12-27:
but it could be the same issue why you cannot change the settings on your Dacia, seeing as Dacia is part owned by Renault and that you will have to bypass the security firewall on the OBD2 socket to make changes by either buying a token or bypassing the gateway by wiring up directly to the Dacia BCM ECU.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif