cedricp/ddt4all #764: DDT4ALL VERY laggy in connected mode

Sennadachi, 2023-01-03:
I am using a USB to ELM327 cable (PIC18F25K60) on my 2016 Renault Twingo SCe and in connected mode the DDT4ALL is very slow and laggy. My car ran out of battery and couldn't start because I was waiting so long for it to scan. I am not ruling out the cable as an issue because in Edition Mode the DDT4ALL program is fine. Does anyone else have this issue and is there a fix?

AndyIreland, 2023-01-03:
you can keep renault alive I think (ie stop the timeout) by  putting keycard into slot on dashboard and putting car into D (drive on automatic gearbox) with engine off and then a long press on the start button on the dash (if your twingo has  a start button) if your Twingo is manual gearbox you can put it in 1st gear with engine off and long press on the start button on the dash. I think this is technically called 'park mode' and I think in that mode the car does not 'time out'

Sennadachi, 2023-01-03:
My twingo is manual and there is no start button. What I am saying is the DDT4ALL software is behaving so slowly that me sitting there with the heater on drained the battery enough to where the car wouldnt start. I am just emphasising how long I waited for the thing to scan.

cedricp, 2023-01-03:
The scan is indeed slow, but all addresses are scanned, you can save the scan result (in the file menu) to skip it next time you use it, though

AndyIreland, 2023-01-03:
> My twingo is manual and there is no start button. What I am saying is the DDT4ALL software is behaving so slowly that me sitting there with the heater on drained the battery enough to where the car wouldnt start. I am just emphasising how long I waited for the thing to scan.

you dont also happen to have note what port speed you were set to in the programme ?

Sennadachi, 2023-01-08:
> The scan is indeed slow, but all addresses are scanned, you can save the scan result (in the file menu) to skip it next time you use it, though Keep in mind it's an effort to get something working :)

It was a dodgy malfunctional Chinese ELM cable. I am sorry for complaining

closed 2023-01-08 by Sennadachi