cedricp/ddt4all #766: Eco mode not working for Renault clio 4 diesel

sbelesiotis, 2023-01-06:
Hello everybody. I bought 3 months ago a 2016 Renault  clio diesel 75hp with an eco button installed. For some reason once I press the eco button it seems to work by reducing engine power, but neither the eco button light switch nor the eco indication light in in instrument cluster are lighting on. Does anybody have any idea of what is happening? Is there any option that I could activate through ddt4all in order to make it functioning?

AndyIreland, 2023-01-06:
As far as I know the eco button does not light up ... ever! 
and as far as the other light not lighting up on the dash have you tried levering out the eco button out of the center console trim to actually see if its wired up?

sbelesiotis, 2023-01-08:
The eco button in the central console is wired up since the eco sign is lighting up green when I turn the lights on and also once I press the button the engine changes its behaviour as it reduces power. However the eco indication on the instrument cluster is not lighting up.

AndyIreland, 2023-01-08:
maybe the eco 'bulb' in the dash has blown then - as a test with PyREN you can check all the dash light warning bulbs are working

sbelesiotis, 2023-01-09:
I performed the test and the eco bulb is not lighting up. Also the economy bar is not lighting up as well.

sbelesiotis, 2023-01-09:
I will try to remove the dash as to have access to the instrument cluster to check whether the eco bulb is present or not. I have see a video that shows the economy var working.

AndyIreland, 2023-01-09:
> I will try to remove the dash as to have access to the instrument cluster to check whether the eco bulb is present or not. I have see a video that shows the economy var working.

Better disconnect the 12v battery before you remove the dash cluster unit (then have the code for your medianav radio)

AndyIreland, 2023-01-09:
There is a setting also in DDT4ALL for enabling eco meter - but the fact that the bulb is not lighting on the PyREN test makes me think you havent got the bulb in yours too:

[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

milvasms, 2023-07-28:
> Hello everybody. I bought 3 months ago a 2016 Renault clio diesel 75hp with an eco button installed. For some reason once I press the eco button it seems to work by reducing engine power, but [...]

Hi there, 
I haver the exacto same problem for 3 years and have not been able to solve it...
Have you find any solutions?

AndyIreland, 2023-07-28:
the eco button never lights up - the green eco word should light up on the instrument cluster should though

milvasms, 2023-07-29:
> the eco button never lights up - the green eco word should light up on the instrument cluster should though

It does light up when you turn you headlights on, as some other buttons do, but its just a "presence" light. My problem is that I'm aware of when it's on even if the "ECO" light doesn't turn up on the dashboard, do to the effect on thortle response and acelleration, etc.

AndyIreland, 2023-07-29:
ah right - yes button lights up when headlights are on. i thought you meant the button doesnt light up when eco is on

AndyIreland, 2023-07-29:
I can do a test remotely that checks ALL your dashboard cluster ,lights are working if you have a bluetooth or usb ELM327 OBD2 adaptor and a windows laptop with bluetooth and capable of receiving Internet signal when inside you car: 

[image]

milvasms, 2023-07-29:
Was doing that exactly just a few minutes ago! :D
Every light comes up, except the ECO light. It's probably missing the led inside I guess...

[image]

AndyIreland, 2023-07-29:
> Was doing that exactly just a few minutes ago! :D Every light comes up, except the ECO light. It's probably missing the led inside I guess...
> 
> [image]

Ah right yes - and on yours you have the stop start bulb whereas on mine I havent (although my clio4 does not have stop start feature so that most probably why) 

The bulbs would hardly be replaceable on the reverse of this cluster I would say because I reckon they would just be SMD LED's on a board. 

The eco-bar / driving style LED is missing on my cluster as my bar does not go red/orange/green on my cluster - it looks like its missing on yours too