cedricp/ddt4all #767: It's not issue more help request

otpisani, 2023-01-06:
Can someone help me what I need to change in multi function unit  (named from Renault) for Megane II, they are the same old and new one, but in Renault workshop mention alternator configuration, which is mention couple times in ddt4all. Can someone share what need to be changed so that I can start engine, or point me where can I find information, thanks in advance.

MellotronScratch, 2023-01-12:
I don't know what you mean with Multi Function Unit. Do you mean UCH?

AndyIreland, 2023-01-12:
I class the MFU unit (multifunction unit) as being the entertainment unit, be that a Renault Medianav or an R-Link depending on model of car . 

And the parameters are called RES (remote engine start) on a petrol or diesel car

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif