cedricp/ddt4all #768: Hi i go in test mode in media nav. And i press clear virgin now its gone... it says in black screen connect diag tool. Please help me.

JPMIKE111, 2023-01-13:
Hi i go in test mode in media nav. And i press clear virgin now its gone... it says in black screen connect diag tool. Please anyone can help me?  Many thanks

JPMIKE111, 2023-01-13:
I believe I need restore software, any one with this software
?

AlMeu, 2023-01-13:
There is not such software for MediaNav.
You erased all parameters inside unit.

AndyIreland, 2023-01-13:
hi @JPMIKE111 

I used this on mine once. - it all depends if your medianav can still read from a USB stick, if it cannot then the boot section has been erased and then I think you will have to get it re-flashed on the bench with someone. 

My medianav was version 9.1.3 (EVO) , I dont know what version you have. It might not work on another version of medianav. 
So, if your medianav still recognises a plugged in USB download the file (in RAR format) then extract and put onto root of 8GB USB stick . 

then go in car and start the engine and keep engine running , plug in USB stick and see if it reads the USB stick - then hopefully after a while it starts reading ULC2_recovery - and (if you can see some dialog on screen) it might say "loading - do not turn of engine" , let it go to end it might be long and take around 20-30 minutes to complete.

Do this at your own risk , its possible it might make matters worse - I dont know at what stage your medianav is or what version it is (you can pull out medianav radio and the version number should be printed on the serial number label) - good luck

Here is the RAR file:
https://we.tl/t-gDvI4WuvOx

JPMIKE111, 2023-01-13:
Hello, Thanks for your message. Yes the usb stick reads the usb sticker, I will be try tomorrow. Is quit dark now, but thank you.
Joe

AndyIreland, 2023-01-13:
hope it works - only use if feel confident . it worked on mine and gave it some extra features afterwards ...  but as I say min is 2016 clio mk4 medianav evo V9.1.3 - if yours is earlier medianav it might not work

shrlnm, 2023-01-14:
@JPMIKE111 if recipe provided by @AndyIreland will not help, post your VIN here I'll try to help you to restore parameters

JPMIKE111, 2023-01-14:
@AndyIreland  thank you. My it's early version 2014 4.0 it's only media nav, yours is media nav evul. so unfortunately stay same message.

JPMIKE111, 2023-01-14:
The radio are not original from vehicle, the vin number not will be help.

AndyIreland, 2023-01-14:
> @AndyIreland thank you. My it's early version 2014 4.0 it's only media nav, yours is media nav evul. so unfortunately stay same message.

Thats a shame - try this update from here put it on a USB stick  (the lgu file) if is reading from usb stick :
choose 4.1.0 

https://www.gps-rlink.com/medianav/update-software-medianav/

AlMeu, 2023-01-14:
@AndyIreland don't keep giving false indication tinking that you know.
All what you are indicating are guessing, randomly chose as solutions when actualy don't know you too what is or how it works.

@JPMIKE111 
Internal parameters are never written by software upgrade or some magical boutons.

AndyIreland, 2023-01-15:
> @AndyIreland don't keep giving false indication thinking that you know. All what you are indicating are guessing, randomly chose as solutions when actualy don't know you too what is or how it works. Thanks producers of MediaNav that didn't allow to upgrade wrong versions on different units.
> 
> @JPMIKE111 Internal parameters are never written by software upgrade or some magical boutons.

sorry, thought I was being of some help. 
I think in the past i fixed my 9.1.3 Medianav with that file on the USB stick - I am pretty sure of it. Thought it would help in this instance. was not giving false information on purpose.

AlMeu, 2023-01-15:
@AndyIreland 
I am 300% that you didn't fix this problem in your unit with any files what you indicate.
Your unit was not in "connect diag tool".

JPMIKE111, 2023-01-15:
Unfortunately doesn't working.

AndyIreland, 2023-01-15:
> 

 maybe not, it was a long time ago i did it. I thought it said connect diag tool on the screen. 
 
 this guy a whizz kid on all things renault medianav @JPMIKE111 
 https://www.youtube.com/@lisandruybf/about
 
 but as others have said you may have erased all the settings of the medianav by pressing the virgin mode . 
 
 If you have a bluetooth OBD2 Adaptor and a specially wired OBD2 cable (or OBD2) where it can read the medianav through the CAN socket, and a windows laptop and get internet in the car I can maybe log on and see if all the settings are still there maybe in your medianav . 
 
 Other than that as the other guys say, maybe it needs sending off and re-programming to get it working again

JPMIKE111, 2023-01-15:
I doing again and happen again

AlMeu, 2023-01-15:
Keep trying ... :-1:

haldun88, 2023-05-07:
> hi @JPMIKE111
>
> I used this on mine once. - it all depends if your medianav can still read from a USB stick, if it cannot then the boot section has been erased and then I think you will have to get [...]

hello,
I have a Clio 4 icon and media nav evo. i have same problem. i was on micom test screen then accidentally pressed some button. now my screen says "system not configured please connect diag tool. my system is "9.1.3" as yours. I wanted to try the file which you had shared here but the link you had given is dead. Can you please renew the link. I live in a small city. there is no one who can solve my problem. thanks by now

KarelSvo, 2023-05-08:
https://www.dacianer.de/resources/funktionen-mit-ddt4all-freischalten-konfigurieren.40/updates#resource-update-107

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

Furtif, 2023-06-02:
good stuff but...

closed 2023-06-02 by Furtif

samir00718, 2023-08-28:
> 

hey, 
can you please renew the link ?
i have same 9.1.3 version
and i have same issue.
i would be grateful if you renew the link.
i have searching for the solutions since last 2 months.
please Help

samir00718, 2023-08-28:
> Hi i go in test mode in media nav. And i press clear virgin now its gone... it says in black screen connect diag tool. Please anyone can help me? Many thanks

if you still have file what @AndyIreland gave  you by link , please post it again or mail it to me on may gmail :- mansurisamir00718@gmail.com

sekkel93, 2024-06-20:
> hi @JPMIKE111
>
> I used this on mine once. - it all depends if your medianav can still read from a USB stick, if it cannot then the boot section has been erased and then I think you will have to get [...]

Hello please i have same issue in my media display version 6.0.9.6 can you advise me please 

thank you

sekkel93, 2024-06-20:
Hello please i have same issue in my media display version 6.0.9.6 can you advise me please

thank you

Furtif, 2024-06-20:
> Hello please i have same issue in my media display version 6.0.9.6 can you advise me please
> 
> thank you

This is no longer an issue, you should visit the discussions section

AlMeu, 2024-06-20:
> Hello please i have same issue in my media display version 6.0.9.6 can you advise me please
> 
> thank you

For you unit it is not so simple procedure, there it is also a Gateway.

Furtif, 2024-06-20:
> > Hello please i have same issue in my media display version 6.0.9.6 can you advise me please
> > thank you
> 
> For you unit it is not so simple procedure, there it is also a Gateway.

please rade section discutions https://github.com/cedricp/ddt4all/discussions/1146

StefanZivkovikj, 2024-11-09:
Hey i have this same exact problem: system not configured please connect diag tool
[image]

But my multimedia is Rlink2 the cat is renault megane 2017 can somebody explain me the steps on how to fix this?