cedricp/ddt4all #772: backup megane 1.5 dci

killuadk, 2023-01-21:
Hi gyus 
while i was looking for DRL reset option because i replaced my LED controller (water get inside ) i did change somme parameter that i don't remember (i did get in ABS setting and many others too)
so in result : 
the STOP-START is always desactivated 
Cruise also desactivated 
light AUTO doesn't Work also 

how can i backup to default setting or if someone has the backup of the same car as mine

shrlnm, 2023-01-22:
Hi 
I may help only with default settings in format of pyren dump. Then you may compare them with yours by mod_ddt and find differences. For making the default dump I need the VIN

Ibratk, 2023-01-25:
@shrlnm Hi, can you help me with the sound problem after a new Radio unit for my Megane 4 2016 ?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836