cedricp/ddt4all #773: Sound problem

Ibratk, 2023-01-23:
Good morning everyone, I have a Renault Megane 4 from 2016. I replaced my Radio unit from HW 1. with a used Radio unit from HW 3. I did this to be able to use Carplay. Now the Carplay works but I have no sound anymore. I have tried with cable and bluetooth but there is no sound. Not even the parking sensors. Will it help if I mount the old original radio unit and make a backup (dump) and put it back with the new radio unit? Does anyone know how I can fix this?

(Translated from Dutch)

AlMeu, 2023-01-23:
I think your unit is bricked, remains to change with other one.
I can help with some new units or old unlocked. If you need leave an email for contact.

Ibratk, 2023-01-23:
I've heard it might be because of Bose edition. Should I encode it maybe? My email is: ibow-58@hotmail.com

AndyIreland, 2023-01-23:
have you got a bluetooth OBD2 and specially wired cable where you can read the radio through the obd2 socket? 
Also is  it a medianav or rlink you have? 
Medianav= USB socket to right of touchscreen.
Rlink = USB socket next to map SD Card reader down by the gear shifter.

Ibratk, 2023-01-23:
I have an R link down by the gear shifter. I have an elm327 cable that allows me to use ddt4all but I don't know where to see the values ​​and what to do with them.

AlMeu, 2023-01-23:
@AndyIreland don't lead again to wrong ways. 
If not sound at all, in any way ... no radio, no bluetooth, no USB .... that are not parameters
But you can try whatever you want, hope always dies last, but what's dead will stay dead.

AndyIreland, 2023-01-23:
> I have an R link down by the gear shifter. I have an elm327 cable that allows me to use ddt4all but I don't know where to see the values ​​and what to do with them.

Ah right - you mean you have the map SD card down by the gear shifter? 

Your screen looks like this?:

[image]

and a wired up cable (or ELM327 adaptor) like this ? 
[image]

Ibratk, 2023-01-23:
Yes SD card is down by the gear shifter. I have a R link 2 big screen. And yes i have a ELM327 adaptor like the right one on the picture. You know what i have to do ?

[image]

AlMeu, 2023-01-23:
So @AndyIreland what was his question and what you answer it .... :-D

AndyIreland, 2023-01-23:
ah right sorry both - for some reason I thought it was  a rlink 1 (evolution) and I might of had an idea if it were that.  

If it similar to rlink 1 you should still be able to go into sound parameters using DDT4ALL and check them settings, see if any adrift. 

if you have a windows laptop and can get internet signal  when in the car @Ibratk  I could take a look over the internet and see if anything jumps out at me even though I havent got an rlink2

shrlnm, 2023-01-26:
@Ibratk what is your VIN and LGU version of new box?

Ibratk, 2023-01-26:
Vin = VF1RFB00356620149 and Lgu version on the radio unit is: hw 3.11,4.4 and sw 3.3.16.946. but since I didn't have a radio I updated it to 9.0.37 and it still doesn't work.

AlMeu, 2023-01-26:
@shrlnm pay attention there is not sound at all in anything, bluetooth, radio, etc and moreover now free upgrade from 3.3 to 9.0, "cherry from the cake"

Ibratk, 2023-01-26:
Now I tried to downgrade again because some functions were not working but suddenly I got black screen. I have no image anymore. It's getting worse...

shrlnm, 2023-01-28:
@Ibratk ohh. If you did an upgrade from LGU v3 on LGU v9 then I can't help you. Sorry. Your r-link 2 is dead.

Ibratk, 2023-02-17:
Hi @shrlnm . I bought a new Radio for my car but i need to remove/change the VIN number of the radio. I've read in other threads that you can change in Virgin mode but I'm not good at it. Can you help me with Teamviewer or something else?

shrlnm, 2023-02-18:
Hi @Ibratk what version of LGU is on the new radio? What VIN number do you have?

Ibratk, 2023-02-18:
@shrlnm The HW ver: 3.10.5.4 and SW ver: 7.0.24.136. My vin number: VF1RFB00356620149

shrlnm, 2023-02-18:
@Ibratk https://cloud.mail.ru/public/wNpb/ZhskospaT

Ibratk, 2023-02-18:
@shrlnm What should I do with this cmd file?

AlMeu, 2023-02-18:
Run it as macro in pyren.

Ibratk, 2023-02-18:
So i must first activate virgin mode and then this cmd file run it on Pyren?

shrlnm, 2023-02-19:
@Ibratk if you need to change the vin then you have to virginize the box first

Ibratk, 2023-02-19:
@shrlnm Ok thank you!

closed 2023-03-01 by Ibratk