cedricp/ddt4all #774: Rlink2 configuration reset

DmitriGoncharick, 2023-01-26:
Dear sirs, hope for your help,
I've pressed unknowingly "configuration reset" in diagnostic ddt4all. Of course screen became black and all parameters has been resetted. Seems that I was able to turn back part of functionality of the system, but unfortunately I can't set-up parameters for phone (I guess this parameter is responsible for voice calls) and acoustic parameters, maybe something else like guiding lines calibration... 
Could you please help me with parameters restoration? 
Vehicle is Renault Talisman 1,5 dci
Rlink2 SW 9.0.37
VIN: VF1RFD00460923411
Contact e-mail: dgoncharick@gmail.com
Thank you!

AlMeu, 2023-01-26:
Your unit original was not v9.0.
Perhaps a v3.3 or a v7.0 in a 2018 car.
I answered you by email.

DmitriGoncharick, 2023-01-27:
Hi, thank you for reply! Exactly, it was 7.0, about one year ago I did update to 9.0 and maybe a week ago to 9.0.37. But update was not the cause of this issue. Is there any chance to restore acoustic parameters? Unfortunately I can't find your answer in mailbox, please check if you sent to correct email from my first post.

AlMeu, 2023-01-27:
I have tried twice and all the time your mailbox reject my emails.
"_Sorry, we were unable to deliver your message to the following address.
<[dgoncharick@gmail.com](mailto:dgoncharick@gmail.com)>:
552: 5.7.0 This message was blocked because its content presents a potential
5.7.0 security issue. Please visit
5.7.0  [https://support.google.com/mail/?p=BlockedMessage ](https://support.google.com/mail/?p=BlockedMessage)to review our
5.7.0 message content and attachment content guidelines. e6-20020ac84e46000000b003a690cc75e8si2017713qtw.564 - gsmtp_"

I will write here.
Here you have your original configuration as your car come out from facory.
Use pyren and apply it as macro.

[RLink_3411_283C26774R.zip](https://github.com/cedricp/ddt4all/files/10516179/RLink_3411_283C26774R.zip)

Good luck

DmitriGoncharick, 2023-01-27:
Thank you very much, I will try to run the script today evening and give a feedback. I'm afraid to ask you, but maybe you have backup parameters for the rest of ECUs in my car? Sometimes I have issue with sleep mode and I think that it's related to wrong parameters configured. Car go for sleep mode only after reset of USM ecu. Have a good day!

AlMeu, 2023-01-27:
My friend, if you play in all ECU parameters you made mess.
It is not safe to just load a set of paramaters everywhere.
If you have more issues better go to a Renault service and they will adjust safely, I hope.

DmitriGoncharick, 2023-01-27:
Thank you very much for configuration script! Everything was restored exept of phone calls (hands-free), I can hear music with bluetooth, but doesn't hear voice during call. In dump I can see bytes configured according to script, but if I try to read it via DDT it says NO DATA, and in log I have WRONG RESPONCE from ELM. DDT Screens in PyRen also can't read Phone paremeters. What do you think, could it be related to version 9.0 ? Is it possible make safe downgrade to 8 or 7? 
Regarding other backups, I've bought car as a second hand and it already had a problem with sleep mode. I understand that just restore of all configurations by script is not the safe way. I've done all dumps for ECUs and would like to check byte to byte with factory configuration (byte descriptions I have in ddt DB). Just to be sure that configuration is correct. 
Anyway thank you very much again for your help! Have a good weekend!

shrlnm, 2023-01-28:
Hi
@AlMeu you were right when noticed that the RL2 have not original version. It means that default configuration  generation have to be a bit more complicated. 
It have to be generated by other configuration file.
1) read the new configuration file reference in the pyren ID screen (ID028)
```
ECU : 11547   (RT:0.0000 RR:  0.0)   RadNav_XFC_0000_0D_UDS
Screen : ID : Identifications
ID020  FACTORY CONTROL PART NUMBER (F187)                 281155430R            
ID021  NAME OF THE COMPUTER SUPPLIER (F 18A)               CAP                   
ID022  SOFTWARE NUMBER (F194)                             90                    
ID023  VDIAG NUMBER  (F1 A0)                               0d                    
ID024  SOFTWARE VERSION (F195)                            34                    
ID025  HARDWARE NUMBER (F191)                             285J95418R            
ID028  CONFIGURATION FILE REFERENCE LINK F188             281155430R            

```
The configuration file reference is 281155430R   
2) now we have to find it in BVMEXTRACTION/XFD/REF.DAT
3) For example, we have found it in line
`0000023292:1:237013604S 237613599S;2:476607355R 478550230R;3:285J24407R 285J22160R;4:284B34752R 284B34318R;5:985109088R 285A41183R;7:285470124R 285474407R;8:275317828R 275319304R;13:285F22726R 285078770R;18:285040459R 285044327R;20:284B63349R 284A97679R;32:249991930R 259C46848R;41:259865181R 259867887R;50:259C58908R 259C73791R;65:285P87045R 285P92648R;70:284A67660R 284A61669R;76:281155430R 283C21455R;77:282759572R 283C01648R;90:283C32096R 283C61669R;91:280630844R 283C49133R;`
4)  We have to get whole record for family 76 there. It's "76:281155430R 283C21455R"
5) Now open the file MTCSAVE/VIN/refdata.txt which were created after first run of acf.py, and replace appropriate substring on the new one here "76:281155430R 283C21455R".
6) Now run acf.py again and select "Read data from file" now.
7) Now the configuration should be appropriate to the new LGU version

shrlnm, 2023-01-28:
@DmitriGoncharick the problem with hands-free is well known. I faced with it twice on my r-link2. The recipe is quite long.

You need 
1) good ELM. It's better to use the USB one. Port speed 230400 and --cfc option
2) LGU V7 full-version with a size about 1.8GB (not 1.3GB)
3) CAN CLIP or the pyren acf macro like the one @AlMeu sent you (but it have to be generated according to your MICOM and LGU V7 version) 

Now the recipe
1) Downgrade your RL2 to V7 with full-version  image
2) Reinstall maps if they not work after downgrade
3) Read ID028 from ID screen of pyren
4) Post it here for generating new configuration for you
5) When config will be generated you have to check that all commands from it can be implemented well. Just run the macro and check that the log not contains negative responses
6) Highly likely that on this stage the problem with hands-free will be stil there.
7) If so, then we have to virginize the RL2 and re implement the autoconfiguration macro.
8) If on step 5 all commands were implemented well then Virginize RL2 (be brave and patien as so after that the screen will became black)
9) Wait about 10 minutes while RL2 reloads completely (TDB reset their skin. RL2 will be black) Do not worry 
10) Configure RL2 from the scratch by CLIP autoconfiguration or by the pyren acf  macro that was generated on step 5
11) Now RL2 should be in OK state and you may install V9, but from now you can't change anyting in configuration by the ELM 

Good luck

AlMeu, 2023-01-28:
In step 5 ID will be the same if he don't change the unit phisically.
In my check I have fealling the original unit was a v3.3 or even lower and thats could be a problem.

shrlnm, 2023-01-28:
I suppose that the REF depends on MICOM version. So if the box was upgraded from earlier versions to v7 then the MICOM had to be upgraded too, it is why I noticed it.

AlMeu, 2023-01-28:
I will double check, but I'm pretty sure that REF depends by part number and not by MICOM version.

AlMeu, 2023-01-28:
@shrlnm I doubled checked and I have a unit with part number 281151014R and this it's ID
[image]
So ID028 it is identical with ID020 always.

shrlnm, 2023-01-28:
@AlMeu May be you are right. My knowledge are based only on facts about my r-link2.
But below is a snippets of my dumps before and after upgrade my r-link2 in a garage from v3.3 on v7.0 and MICOM 5615

22F188:62 F1 88 32 38 31 31 35 35 32 35 32 52 (281155252R)
22F188:62 F1 88 32 38 31 31 35 35 34 33 30 52 (281155430R)

shrlnm, 2023-01-30:
I suppose it is quite logical to link these references to MICOM version as MICOM is responsible for processing CAN commands, check their length and transfer them to the android app. It is why very important to recover configuration of  RL2 on that LGU version which is compliant with MICOM. When you install LGU v9 on MICOM 5615 you are breaking interoperability among CAN BUS and the android app. Android app will work fine but you lose an ability to change some setting thru CAN BUS.
One more argument about ID028 is related to CAN CLIP auto configuration scenarios scen_auto_config_11300.xml  This scenario is reading ID028 during autoconfiguration procedure

DmitriGoncharick, 2023-01-30:
Good morning dear sirs!
Thank you very much for your kind help. Looks like I was able to recover hands free function. I did next steps:
- First of all I updated my ELM327 with last 2.2 firmware (with pickit3) and updated bluetooth module with dual band eeprom. It gave me extended buffer and support of additional AT commands, just to be sure that I dont have problems from ELM side.
- Rollback to V 7.0.24.166 and make reset configuration to default (DDT diagnosis menu => ECU Reset => Configuration reset)
- Run configuration script sent by @AlMeu
- After downgrade from 9 to 7 I've got a problem with bluetooth connection at all, but as far as I understood it's related to different bluetoth driver version. I've done reset to default settings from RLink HMI. Bluetooth connection restored.
- Set Phone configuration via DDT Manual command.
I have no possibility check all function of the system, as I did late in the night :) but at least now I have phone calls. I will check if ID028 has been changed during downgrade and let you know.

I would like to ask you once more if you have possibility provide me dumps of other ECUs in my vehicle. I would like to compare it with my current settings before visit to renault service, as I have problems with sleep mode. 
Thank you!

shrlnm, 2023-01-31:
@DmitriGoncharick 
https://cloud.mail.ru/public/kpxw/2s434qgxc

DmitriGoncharick, 2023-02-13:
Good morning! Thank you for help, finally issue with sleep mode has been solved. I've deactivated parameter CF_Passing_inhibition according to backup dumps. R-link has been upgraded to 9.x without any issues as well.

closed 2023-02-13 by DmitriGoncharick

Proton9999, 2024-02-08:
Good evening. I downgraded the rlink firmware and then installed the newest one. The menu button is now inactive on the screen. Resets when pressed. Who can help?

sdade86, 2024-04-26:
@DmitriGoncharick Hello mate, i have some question about V9, maybe you can help me.

- I have update to V9 and using Carplay the phone button on steering whell not working for aswer/hung up calls. Do you have the same problem?
- If i want to downgrade to V8 is it possible? or downgrade is only to V7?
- Where you have found LGU for rollback to V7? need any special procedure or only put in USB stick and install?

Thanks in advance!

sdade86, 2024-05-01:
> @DmitriGoncharick the problem with hands-free is well known. I faced with it twice on my r-link2. The recipe is quite long.
>
> You need
>
> 1. good ELM. It's better to use the USB one. Port speed 230400 [...]

@fusz20 btw i can’t find full LGU, do you have? For sure need to put forceupdate.txt in rlink folder

fusz20, 2024-05-01:
Unfortunately, I don't have the full version.
Does anyone have any ideas?
Where can it be purchased?

sdade86, 2024-05-01:
In this forum have various rlink firmware 

https://www.renaultkadjarforum.com/forum/renault-r-link/736-all-r-link-firmwares.html

but use at your risk, i don’t know if is full or not, if work or not

matzel82, 2024-05-01:
One part is install with an USB tool, directly over the USB port, the other part ist to install over can directly to the ECU from radio…

please…don’t touch it

fusz20, 2024-05-01:
How can I turn off the check?
When I put in the V7. Then write it out.
Switching between generations is not supported

sdade86, 2024-05-01:
Have put forceupdate.txt in rlink folder?

fusz20, 2024-05-01:
Yes, I placed it
Unfortunately, my V7 version is only 1.1 GB.
I couldn't find a bigger size anywhere

sdade86, 2024-05-01:
Maybe @DmitriGoncharick or @shrlnm  can help you…

fusz20, 2024-05-02:
Thank you very much!
Is this good for V8 to V7?
Because I have 9.0.37.

sdade86, 2024-05-02:
I can't give you a reliable answer, I've never tried it. As I told you previously you can use at your risk

sdade86, 2024-05-03:
@fusz20 please if you decided to try create a new discussion and make an how to helpfull for other user

fusz20, 2024-05-04:
I would try.
But I have V9.0.37 on the car.
So I don't want to run from V8 to V7. The system might crash if I run it.

johnkar87, 2025-02-09:
> I have tried twice and all the time your mailbox reject my emails. "_Sorry, we were unable to deliver your message to the following address. <[dgoncharick@gmail.com](mailto:dgoncharick@gmail.com)>: [...]

hi can you contact with me my email is giannis.karaoglanis.87@gmail.com

johnkar87, 2025-02-11:
hi.i tried to update 2.2. to 3.3 but failed and know i think i have bootloop.i tried  with pyren with no lucky.Maybe i am doing something wrong but how i get this screen on pyren to generate a file?

> ECU : 11547   (RT:0.0000 RR:  0.0)   RadNav_XFC_0000_0D_UDS
> Screen : ID : Identifications
> ID020  FACTORY CONTROL PART NUMBER (F187)                 281155430R
> ID021  NAME OF THE COMPUTER [...]

AlMeu, 2025-02-12:
@johnkar87 if you have the R-Link 2 in bootloop, there is nothing to do anymore to your unit.
Just to replace with another unit. 
If you leave an email for contact, I can help you with this.

johnkar87, 2025-02-12:
my email is that you sent me email you How you help me?

johnkar87, 2025-02-12:
giannis.karaoglanis@gmail.com 
my email

> [@johnkar87](https://github.com/johnkar87) if you have the R-Link 2 in bootloop, there is nothing to do anymore to your unit. Just to replace with another unit. If you leave an email for contact, I can help you with this.

AirySeeker113, 2025-05-08:
Hi, i have a 2016 Kadjar 1.5 dci BOSE ver. that was originally 2.2, when updating to 3.3 i saw in the update log that it factory reset itself and now the unit is stuck in bootloop. The update was successful and MICOM version is 5446. Can somebody help me fix it?
VIN: VF1RFE00356680417
E-mail: airyseeker@gmail.com
Thank you!