cedricp/ddt4all #778: Win11x64 and ELM327 Bluetooth - permission error

ezshua, 2023-02-13:
Hello,

I try "Check ELM" on Windows11 x64 with paired OBDII Elm327 bluetooth version. And has this message^

RU version:
Пытаемся открыть портCOM5 @ 38400
Serial<id=0x22323edc880, open=True>(port='COM5', baudrate=38400, bytesize=8, parity='N', stopbits=1,
 timeout=5, xonxoff=False, rtscts=False, dsrdtr=False)
Пытаемся открыть портCOM5 @ 38400
Error:could not open port 'COM5': PermissionError(13, 'Отказано в доступе.', None, 5)
ELM не подключен или неверный COM порт COM5
ELM reset...
****************************************
*       Связь с ELM потеряна

ENG version:
Trying to open port COM5 @ 38400
Serial<id=0x22323edc880, open=True>(port='COM5', baudrate=38400, bytesize=8, parity='N', stopbits=1,
  timeout=5, xonxoff=False, rtscts=False, dsrdtr=False)
Trying to open port COM5 @ 38400
Error:could not open port 'COM5': PermissionError(13, 'Permission denied.', None, 5)
ELM not connected or wrong COM port COM5
ELM reset...
****************************************
* Communication with ELM lost

I run ddt4all as administrator, used Installer v2 and latest source ddt4all-master.zip for update

Maybe I need some option for Win11x64 ?

Furtif, 2023-02-13:
uses one other speed and try again.

ezshua, 2023-02-14:
**Exactly the same result at other speeds:**

Trying to open portCOM5 @ 115200
Serial<id=0x1cd8ef7f970, open=True>(port='COM5', baudrate=115200, bytesize=8, parity='N', stopbits=1
, timeout=5, xonxoff=False, rtscts=False, dsrdtr=False)
Trying to open portCOM5 @ 38400
Error:could not open port 'COM5': PermissionError(13, 'Отказано в доступе.', None, 5)
ELM not connected or wrong COM port COM5
ELM reset...
****************************************
*       Connection to ELM was lost

**if press "ELM benchmark" after that again:**

Trying to open portCOM5 @ 38400
Error:could not open port 'COM5': OSError(22, 'Additional connections to this remote computer 
are not possible now because their number has reached the limit.', None, 71)
ELM not connected or wrong COM port COM5
ELM reset...
****************************************
* Connection to ELM was lost

**and the interface hangs until the task is canceled...**

ezshua, 2023-02-14:
**It's strange, but Pyren gives diagnostics after clicking "Check ELM":**

Opening ELM
v1.0   ATZ                      [OK]
v1.0   ATE1                     [OK]
v1.0   ATL0                     [OK]
v1.0   ATI                      [OK]
v1.0   AT@1                     [OK]
v1.0   ATAL                     [OK]
v1.0   ATBD                     [OK]
v1.0   ATBI                     [OK]
v1.0   ATCAF0                   [OK]
v1.0   ATCFC1                   [OK]
v1.0   ATCP 01                  [OK]
v1.0   ATCS                     [OK]
v1.0   ATCV 1250                [OK]
v1.0   ATD                      [OK]
v1.0   ATDP                     [OK]
v1.0   ATDPN                    [OK]
v1.0   ATH0                     [OK]
v1.0   ATI                      [OK]
v1.0   ATIB 10                  [OK]
v1.0   ATIB 96                  [OK]
v1.0   ATL1                     [OK]
v1.0   ATM0                     [OK]
v1.0   ATMA                   [TIMEOUT]
v1.0   ATMR 01                [TIMEOUT]
v1.0   ATMT 01                [TIMEOUT]
v1.0   ATNL                   [TIMEOUT]
v1.0   ATPC                     [OK]
v1.0   ATR1                     [OK]
v1.0   ATRV                     [OK]
v1.0   ATSH 001122              [OK]
v1.0   ATSH 012                 [OK]
v1.0   ATSP A6                  [OK]
v1.0   ATSP 6                   [OK]
v1.0   ATCM 123                 [OK]
v1.0   ATCM 12345678            [OK]
v1.0   ATCF 123                 [OK]
v1.0   ATCF 12345678            [OK]
v1.0   ATST FF                  [OK]
v1.0   ATSW 96                  [OK]
v1.0   ATSW 34                  [OK]
v1.0   ATTP A6                  [OK]
v1.0   ATTP 6                   [OK]
v1.0   ATWM 817AF13E            [OK]
v1.0   ATWS                     [OK]
v1.1   ATFC SD 300000           [OK]
v1.1   ATFC SH 012              [OK]
v1.1   ATFC SH 00112233         [OK]
v1.1   ATFC SM 1                [OK]
v1.1   ATPP FF OFF              [OK]
v1.1   ATPP FF ON               [OK]
v1.1   ATPPS                    [OK]
v1.2   ATAR                     [OK]
v1.2   ATAT1                    [OK]
v1.2   ATDM1                  [TIMEOUT]
v1.2   ATIFR H                [TIMEOUT]
v1.2   ATIFR0                   [OK]
v1.2   ATIIA 01                 [OK]
v1.2   ATKW0                    [OK]
v1.2   ATMP 0123              [TIMEOUT]
v1.2   ATMP 0123 4            [TIMEOUT]
v1.2   ATMP 012345            [TIMEOUT]
v1.2   ATMP 012345 6          [TIMEOUT]
v1.2   ATSR 01                  [OK]
v1.3   ATCRA 012                [OK]
v1.3   ATCRA 01234567           [OK]
v1.3   ATD0                     [OK]
v1.3   ATFE                     [OK]
v1.3   ATJE                     [OK]
v1.3   ATJS                     [OK]
v1.3   ATKW                     [OK]
v1.3   ATRA 01                  [OK]
v1.3   ATS1                     [OK]
v1.3   ATSP 00                  [OK]
v1.3   ATV0                     [OK]
v1.4   ATCEA                    [OK]
v1.4   ATCEA 01                 [OK]
v1.4   ATCV 0000                [OK]
v1.4   ATIB 48                  [OK]
v1.4   ATIGN                    [OK]
v1.4   ATPB 01 23               [OK]
v1.4   ATRD                     [OK]
v1.4   ATSD 01                  [OK]
v1.4   ATSI                     [OK]
v1.4   ATFI                     [OK]
v1.4   ATSS                     [OK]
v1.4   ATTA 12                  [OK]
v1.4   ATCSM1                 [FAIL]
v1.4   ATJHF1                 [FAIL]
v1.4   ATJTM1                 [FAIL]
v1.4   ATJTM5                 [FAIL]
v1.4b  ATCRA                  [FAIL]
v2.0   ATAMC                  [FAIL]
v2.0   ATAMT 20               [FAIL]
v2.1   ATCTM1                 [FAIL]
v2.1   ATCTM5                 [FAIL]
****************************************
*       RESETTING ELM
****************************************
*
*       ERRORS STATISTIC
*
* error_frame      =  0
* error_bufferfull =  0
* error_question   =  11
* error_nodata     =  0
* error_timeout    =  10
* error_rx         =  0
* error_can        =  0
*
*       RESPONSE TIME (Average)
*
* response_time    =  0.237
*
****************************************
 ATZ

OBDII  v1.5

>

Result: 76 from 95
 Max version:v1.4

Furtif, 2023-02-14:
Try https://github.com/Furtif/ddt4all or pr  opened https://github.com/cedricp/ddt4all/pull/779 is fine for me

ezshua, 2023-02-16:
Yes, it helps!

with one little remark:

ddt4all.py, 264 row
- self.screenlogfile = open("./logs/screens.txt", "at")
+self.screenlogfile = open("./logs/screens.txt", "at", encoding='utf-8')

its fix Unicode error on cyrillic codepages, like cp1251

closed 2023-02-16 by ezshua

Furtif, 2023-02-16:
> Yes, it helps! Thanx a lot!
> 
> with one little remark:
> 
> ddt4all.py, 264 row -- self.screenlogfile = open("./logs/screens.txt", "at") ++self.screenlogfile = open("./logs/screens.txt", "at", encoding='utf-8')
> 
> its fix Unicode error on cyrillic codepages, like cp1251

Is main.py not ddt4all.py