cedricp/ddt4all #780: QT5 issue on ubuntu 22.04.1 LTS

vbelloir, 2023-02-19:
Hi all,

I got an issue with QT 5 trying to lunch ddt4all on ubuntu.
First screen is OK, with choosing device (USB, BT, WIfi....), but even on demo or connected mode, I got: 

> python ddt4all.py 
> Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
> Utilisation de la base de donnée personnelle
> 1652 ECU chargées en base de donnée.
> Cannot mix incompatible Qt library (5.15.3) with this library (5.15.2)
> Abandon (core dumped)

Any idea?

vbelloir, 2023-02-19:
Here are all qt5 package installed:

```
apt list --installed | grep "qt5"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libaccounts-qt5-1/jammy,now 1.16-2 amd64  [installé, automatique]
libdbusmenu-qt5-2/jammy,now 0.9.3+16.04.20160218-2build1 amd64  [installé, automatique]
libmarblewidget-qt5-28/jammy,now 4:21.12.3-0ubuntu1 amd64  [installé, automatique]
libphonon4qt5-4/jammy,now 4:4.11.1-4 amd64  [installé, automatique]
libphonon4qt5-data/jammy,jammy,now 4:4.11.1-4 all  [installé, automatique]
libpolkit-qt5-1-1/jammy,now 0.114.0-2 amd64  [installé, automatique]
libpoppler-qt5-1/jammy-updates,jammy-security,now 22.02.0-2ubuntu0.1 amd64  [installé, automatique]
libqt5concurrent5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5core5a/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5dbus5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5designer5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5gui5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5help5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5keychain1/jammy,now 0.13.2-5 amd64  [installé, automatique]
libqt5multimedia5-plugins/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5multimedia5/jammy,now 5.15.3-1 amd64  [installé]
libqt5multimediagsttools5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5multimediawidgets5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5network5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé]
libqt5networkauth5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5opengl5-dev/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5opengl5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5positioning5/jammy,now 5.15.3+dfsg-3 amd64  [installé, automatique]
libqt5printsupport5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5qml5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5qmlmodels5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5qmlworkerscript5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5quick5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5quickcontrols2-5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5quicktemplates2-5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5quickwidgets5/jammy,now 5.15.3+dfsg-1 amd64  [installé, automatique]
libqt5script5/jammy,now 5.15.3+dfsg-1 amd64  [installé]
libqt5scripttools5/jammy,now 5.15.3+dfsg-1 amd64  [installé]
libqt5sensors5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5serialport5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5sql5-mysql/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5sql5-sqlite/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5sql5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5svg5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5test5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5texttospeech5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5waylandclient5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5waylandcompositor5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5webchannel5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5webengine-data/jammy,jammy,now 5.15.9+dfsg-1 all  [installé, automatique]
libqt5webengine5/jammy,now 5.15.9+dfsg-1 amd64  [installé, automatique]
libqt5webenginecore5/jammy,now 5.15.9+dfsg-1 amd64  [installé, automatique]
libqt5webenginewidgets5/jammy,now 5.15.9+dfsg-1 amd64  [installé, automatique]
libqt5webkit5/jammy,now 5.212.0~alpha4-15ubuntu1 amd64  [installé, automatique]
libqt5websockets5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5widgets5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
libqt5x11extras5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libqt5xml5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé]
libqt5xmlpatterns5/jammy,now 5.15.3-1 amd64  [installé, automatique]
libsignon-qt5-1/jammy,now 8.59+17.10.20170606-0ubuntu3 amd64  [installé, automatique]
phonon4qt5-backend-vlc/jammy,now 0.11.3-1 amd64  [installé, automatique]
phonon4qt5/jammy,now 4:4.11.1-4 amd64  [installé, automatique]
python3-pyqt5.qtwebkit/jammy,now 5.15.6+dfsg-1ubuntu3 amd64  [installé]
python3-pyqt5.sip/jammy,now 12.9.1-1build1 amd64  [installé, automatique]
python3-pyqt5/jammy,now 5.15.6+dfsg-1ubuntu3 amd64  [installé, automatique]
qt5-gtk-platformtheme/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
qt5-gtk2-platformtheme/jammy,now 5.0.0+git23.g335dbec-4build3 amd64  [installé, automatique]
qt5-qmake-bin/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
qt5-qmake/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64  [installé, automatique]
qt5-style-plugin-cleanlooks/jammy,now 5.0.0+git23.g335dbec-4build3 amd64  [installé, automatique]
qt5-style-plugin-motif/jammy,now 5.0.0+git23.g335dbec-4build3 amd64  [installé, automatique]
qt5-style-plugin-plastique/jammy,now 5.0.0+git23.g335dbec-4build3 amd64  [installé, automatique]
qt5-style-plugins/jammy,now 5.0.0+git23.g335dbec-4build3 amd64  [installé]
```

Furtif, 2023-02-19:
uses https://github.com/cedricp/ddt4all/pull/779 all fine.

vbelloir, 2023-02-19:
Hi Furtif, and thanks for your answer.

I clone your repo, but I got same issue.
```
$ python main.py
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

(process:22346): Gtk-WARNING **: 15:36:24.176: Locale not supported by C library.
	Using the fallback 'C' locale.
Cannot mix incompatible Qt library (5.15.3) with this library (5.15.2)
Abandon (core dumped)
```

Do I miss something?

Furtif, 2023-02-20:
Look ddt4all Comment for ubuntu

Furtif, 2023-02-20:
https://github.com/Furtif/ddt4all/blob/master/ddt4all.sh#L9 This

vbelloir, 2023-02-20:
Hi,
That's ok, it works.
Yestrerday, I didn't use correctly your repo.

Thanks for your job!

Furtif, 2023-02-20:
Fine

Furtif, 2023-02-20:
Close Issue 😉

vbelloir, 2023-02-20:
You're right!

closed 2023-02-20 by vbelloir