cedricp/ddt4all #783: HELP!!!    "Please add EOL settings" after Configuration Reset - PLEASE HELP!!!!

vvvvitaliu, 2023-03-01:
My car Renault Kangoo 2017    
MFD 5.4
pressing "Reset Configuration" button, but now the system is stuck at a warning screen saying "Unconfigured device, please add EOL settings"
not work... help me.. MFD 5.4

Dimonlipko, 2023-03-01:
Now you need go to Renault dealer workshop, and recovery setting in radio module. I have the same problem with Renault ZOE, newer push button "Reset Configuration")

vvvvitaliu, 2023-03-01:
> Теперь вам нужно отправиться в мастерскую дилера Renault и установить восстановление в радиомодуле. У меня такая же проблема с Renault ZOE, новее кнопка "Сброс конфигурации")

I read that with the help of ddt4all they somehow restored

shrlnm, 2023-03-01:
You may restore it by the pyren macro functionality. The macro is generating by the vin

vvvvitaliu, 2023-03-01:
> You may restore it by the pyren macro functionality. The macro is generated based on the vin

tell me how exactly to do it??? could you do it remotely???

vvvvitaliu, 2023-03-01:
> > You may restore it by the pyren macro functionality. The macro is generated based on the vin
> 
> tell me how exactly to do it??? could you do it remotely???
MY VIN VF1FW52R258540330

shrlnm, 2023-03-01:
https://cloud.mail.ru/public/7TgF/nUuZzdR8n

Инструкция как применять здесь
https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen

vvvvitaliu, 2023-03-01:
but I have r-link 1 and not 2, is it possible after all 2??

shrlnm, 2023-03-02:
@vvvvitaliu look in your mail box. You will find complete message without --link removed--. It is the link to the cmd file made especially for your r-link 1 and parameters according your vin number. 
You need to implement that macro by steps described in the instruction.

Zaiac5215273, 2023-03-02:
> "Unconfigured device, please add EOL settings"

Zaiac5215273, 2023-03-02:
> 
hi. help to restore R-link with the same problem. here is my vin VF1KT12C652410846      "Unconfigured device, please add EOL settings"

vvvvitaliu, 2023-03-03:
> @vvvvitaliu look in your mail box. You will find complete message without --link removed--. It is the link to the cmd file made especially for your r-link 1 and parameters according your vin number. You need to implement that macro by steps described in the instruction.

EOL its ok.
Radio non configure...

vvvvitaliu, 2023-03-03:
> VF1FW52R258540330

everything was successful. The EOL disappeared. now the radio is not configured on the dashboard. What should I do next?

vvvvitaliu, 2023-03-03:
shrlnm , maybe you have watsap??? for better communication?

shrlnm, 2023-03-04:
> now the radio is not configured on the dashboard. What should I do next?

What do you mean? I'm not an expert in r-link1. How it's look like?

vvvvitaliu, 2023-03-04:
[image]

vvvvitaliu, 2023-03-04:
is it possible to create a macro for mfd ti mіd?

AndyIreland, 2023-03-04:
> is it possible to create a macro for mfd ti mіd?

Dont forget vvvvitaliu if you want me to have a go at filling in the parameters like I mentioned the other day the offer is still there , you never got back to me . 

maybe all it needs now SHRLM has giving you the script is just the parameters filled in now your EOL message has gone .

shrlnm, 2023-03-04:
> is it possible to create a macro for mfd ti mіd

I sent you macro exact for your MFD. Are you sure that you have good ELM and all parameters form macro was applied? 
Please share a log

vvvvitaliu, 2023-03-06:
> > is it possible to create a macro for mfd ti mіd
> 
> I sent you macro exact for your MFD. Are you sure that you have good ELM and all parameters form macro was applied? Please share a log
[ecu_log.txt](https://github.com/cedricp/ddt4all/files/10897220/ecu_log.txt)

[elm_log.txt](https://github.com/cedricp/ddt4all/files/10897218/elm_log.txt)

shrlnm, 2023-03-06:
The log is fine. All commands were applied. 
What did you mean with "mfd **ti mіd**"?

vvvvitaliu, 2023-03-06:
if all the settings are applied, then why does it write that it is not configured? what to do next?

vvvvitaliu, 2023-03-06:
how to finish configuration? USB stick with the file tomtom.000 does not give any results.

AndyIreland, 2023-03-06:
@vvvvitaliu - did you try putting that tomtom.000 on a sd card then press home button 5 times on control panel to see if rlink rebooted and started installation??
if not , give me a shout and I will remotely boot your rlink with your card with tomtom.000 sd card in slot .... and then if that dont work I will look and see if any info needs adding in the ecu to get your rlink working.

shrlnm, 2023-03-11:
> how to finish configuration?

@vvvvitaliu Hi. Have you solved a problem. It is probably you also have to configure radio module additionally to MFD. If so then I may make a macro for radio too

shrlnm, 2023-03-29:
Hello. I'll very appreciate if someone share with me the dumps of RADIO and MFD made by pyren on R-Link 1. Thank you in advance.

AndyIreland, 2023-03-29:
> Hello. I'll very appreciate if someone share with me the dumps of RADIO and MFD made by pyren on R-Link 1. Thank you in advance.

hello @shrlnm  - just sent to you on drive2.ru message (I hope they are the right files you are looking for?)

bruvv, 2023-04-07:
Shit i clicked the wrong button now it also shows it. Running zoe 2015 rlink 1. Anyone can help? @AndyIreland ?

shrlnm, 2023-04-07:
@bruvv  for restoring all other parameters 
https://www.drive2.ru/b/643019689764615344/

AndyIreland, 2023-04-07:
> I managed to fix it by issuing a random username and password in the MFD > 14 config

well that was clever if that's all you had to do to get it working again - well done!

bruvv, 2023-04-07:
Umh after having the car rest for 8 hours (I was at work) I came back to the car and it showed unconfgured radio, so back to the drawing board :)

AndyIreland, 2023-04-07:
thats a shame - yeah sure, give this tomtom.000 a go and see if it fixes it, it might, it might not. 

try SD card way first. Format a 8gb SD card (not your map card!!) to fat32 and put tomtom.000 on the root.
If no work try the USB way  format a 8GB Usb stick to Fat32 and put tomtom.000 on the root 

keep car/engine running while its loading the file , it could take 20mins to load - after its finished loading manually reboot the rlink by pressing home button 5 times. 

https://we.tl/t-PKn3pjjISn

bruvv, 2023-04-08:
@AndyIreland thanks for providing the file, what do I do with the sd card, come in the car, unplug the original card and insert the original and reboot the MFD?

AndyIreland, 2023-04-08:
yes take out the map SD card and put in the card with tomtom.000 into the slot and turn on the rlink - yes it will be difficult to see the SD card reading because there is no light. Leave int for 30 mins (with engine running) and see what the screen is doing along the way see if it changes

bruvv, 2023-04-08:
Thanks for the quick reply, as soon as I enter the car the rlink 1 display shows unconfigured radio straight away, I eject the original renault sd, instert the new sd with your file. Hit the home button on the rlink 5 times, after waiting a bit I see "unconfigured radio" again within like 1 minute. So that means it is not working for me right?

AndyIreland, 2023-04-08:
try turning of the rlink (if you can do that with the eol, message) with the round button at side of screen , then remove map card. Then insert the fat32 card with tomtom.000 and then turn rlink on then press home button 5 times to reboot ... even if eol message leave rlink running for 30 mins just to see if it will read from SD card. if that dont work try the USB way

bruvv, 2023-04-08:
Tried it but it did not work. What else can I try ? The pyren telegram bot is offline I've noticed :(

AndyIreland, 2023-04-08:
thats a shame the tomtom.000 file did not fix it did you try both the SD way and the USB stick way? @bruvv 

I have a txt dump file but it would have my VIN number in it so I dont think it will work on your car with a different VIN number - then there would be the issue with the different radio code too I should imagine - if you talk nicely to @shrlnm and give him your VIN maybe he could make a dump .txt file for your particular car - other than that maybe you are missing some parameters in your configuration files for the MFD and radio I wouldnt know unless I could log on remotely over the internet and have a look at your parameters and see whats filled in

bruvv, 2023-04-08:
Thanks @AndyIreland for the support. Yep did both. USB with original SD inserted. Did nothing. Did only usb without any SD inserted. Nothing did anything.

I found this where you shared your dump before and the user reported it working: https://www.speakev.com/threads/zoe-wont-charge.160344/post-3044192

But I would prefer @shrlnm method tbh. But could try both anyways :)?

AndyIreland, 2023-04-09:
give your email address @bruvv  and i can send you the .txt dump file off my car if you like to see if it works. 

other than that try this 11.346xxx software update on a USB stick (if your rlink is reading from the USB socket ok) - try doing it by usb first, if not work try doing the SD way see if it loads:
https://we.tl/t-tjl51DKeAl

bruvv, 2023-04-09:
@AndyIreland no need! Used @shrlnm is method and got it back up again! The bot helped :) thanks for helping Andy! You rock

AndyIreland, 2023-04-09:
> @AndyIreland no need! Used @shrlnm is method and got it back up again! The bot helped :) thanks for helping Andy! You rock

great stuff glad you have it working again - now you have, check your rlink version is up to date at  11346xxx or 11.347xxx and if not you can install that file if you want to update it orgo the official way through the rlink site

bruvv, 2023-04-09:
Perfect Andy will look into tomorrow :) Happy easter.

AndyIreland, 2023-04-09:
> Perfect Andy will look into tomorrow :) Happy easter.

thank you same to you

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

runsren, 2023-10-01:
This thread got my Smart Media System working, thanks.

Alivelton, 2023-11-15:
I need help my car