cedricp/ddt4all #785: Start/Stop deactivation methods for Renault Clio IV - Ph2 -Dci - EDC gearbox

Trickykid1978, 2023-03-05:
Hello everybody, 

A complete newbie here ... while waiting for a reliable ELM to arrive (i bought a USB and a WIFI ELM327, both with PIC18F25K80 chips) , i started getting familiar with DDT4ALL in demo mode (i am using DDT4ALL Version 5.5.0 with a 2021 database, loading 3618 ECUs in total) . The first task as soon as i get connected, would be to deactivate the start/stop feature completely and for this reason, i watched tons of videos, read forum threads, etc . So far, i have found the following parameters that can disable start/stop but i am looking for the one that i will finally choose.  Have you tried any of these solutions and which one would you choose just to remain on the safe side of things ?  (the third solution comes from a user in Turkey who has posted a video in youtube explaining why this solution is better as it won't generate a SS signal in the dashboard saying No available etc) 
[image]
[image]
[image]

shrlnm, 2023-03-05:
Hi. I'am not an expert in disabling START/STOP function but I may suggest the way. IMHO it's may do a job. 
The X98 platform has an STASTO mtc parameter which is responsible for configuring this function. We may generate default configurations with and without this parameter and compare both sets of configurations.  The default configurations may be generated by the VIN

AndyIreland, 2023-03-05:
do some people say you can disable SS by just removing the extra cable on the 12v battery under the bonnet - I not sure because our Clio 4 has not got SS at all on ours. 

If you get SS error light on board you can always extinguish it by coding it out on the cluster 

[image]

also I dont know what year your ph2 is if over 2019 (but I think it would be called clio5 then) but if it is you will have a security gateway firewall on the OBD2 that you have to unlock with a token from renault before making any adjustments on DDT4ALL

shrlnm, 2023-03-05:
> do some people say you can disable SS by just removing the extra cable on the 12v battery under the bonnet

I suppose it is a battery current sensor. Removing it you will broke SOH procedure and may damage or get drained the battery one day

AndyIreland, 2023-03-05:
> > do some people say you can disable SS by just removing the extra cable on the 12v battery under the bonnet
> 
> I suppose it is a battery current sensor. Removing it you will broke SOH procedure and may damage or get drained the battery one day

ah right thanks, what is SOH - state of health?

shrlnm, 2023-03-05:
@AndyIreland yes It is. The car ECUs have quite complex algorithm of battery state assessment and the battery current sensor is one of main sensors for this. 
So it is quite logical that the car disable START/STOP function when it doesn't know the state of the battery.

Trickykid1978, 2023-03-05:
> Hi. I'am not an expert in disabling START/STOP function but I may suggest the way. IMHO it's may do a job. The X98 platform has an STASTO mtc parameter which is responsible for configuring this [...]

To be honest, i didn't get what you mean :)  I know that start-stop comes alive provided some conditions are met in the car . For example engine temperature (it won't start-stop when engine is still cold) , so it might be possible to set this temperature setting so high as to render start/stop useless as this temperature might never be met ...this is just a guess ...

Trickykid1978, 2023-03-05:
> do some people say you can disable SS by just removing the extra cable on the 12v battery under the bonnet
> 
> [image]
> 
>     * Im not too sure because our Clio 4 has not got SS at all on ours.
> 
> 
> If you get SS error light on board you can always extinguish it by coding it out on the cluster
> 
> [image]
> 
> also I dont know what year your ph2 is if over 2019 (but I think it would be called clio5 then) but if it is you will have a security gateway firewall on the OBD2 that you have to unlock with a token from renault before making any adjustments on DDT4ALL

Hi Andy,  i will be trying to deactivate it through DDT4ALL ... based on data available, i think it will be quite safe and easy to do but i would have felt better if i found someone who did it already and used a specific ecu/parameter to do it.  

My Clio is a X98ph2 - Clio IV Ph2  (this is the model in ddt4all) which is the facelifted Clio iv produced 2016-2019 (it is not firewalled or sth)

AndyIreland, 2023-03-05:
yeah - just make sure you choose the correct ecu(s) to do it. do a search for particular right ECU's in your car with the magnifying glass in DDT4ALL

KarelSvo, 2023-03-05:
Simply install a jumper on the hood open sensor connector.
Or is an open bonnet reported in the display?

Trickykid1978, 2023-03-05:
> Simply install a jumper on the hood open sensor connector. Or is an open bonnet reported in the display?

well, to be honest i would like to get a bit involved with ddt4all as there a few other things i might change if start/stop project goes well ...so, generally disabling start/stop feature is not the main/only issue but disabling it properly **with** ddt4all is :)  . In other renault cars' videos and instructions online, it appears as easy as four clicks (i mean it looks easier than the bonnet sensor that you proposed which is also a correct solution to deactivate ss . What puzzles me is which of the above three commands  would be better/safer to choose ...

Trickykid1978, 2023-03-05:
> yeah - just make sure you choose the correct ecu(s) to do it. do a search for particular right ECU's in your car with the magnifying glass in DDT4ALL

i haven't yet understood what green and red ECUs mean ... i mean if the command i want to change is in a ECU listed as red (after searching with the magnifying glass) , should i mess with it or not ?  So far, i have studied almost all ecus in demo mode (complete tree - all screens etc)  but i have zero experience of connected mode as i am waiting for reliable hardware - my mini and old blue ELM327 clone doesn't qualify as such). 

Anyway, after posting this, i found this thread https://github.com/cedricp/ddt4all/issues/728 and from what i understand, my above option No 2 (disabling through UPC-EMM) didn't work for this user with a ph2 clio iv . So for the time being, my two options will be no 1 and no 3 :)

AndyIreland, 2023-03-05:
in found ecu's if they are in RED then it means that DDT has found a ECU , but its not the one it has expected , so you can make changes but be extra careful with your changing of parameters and take extra caution just in case it bricks the ECU software... I mean thats what I have always taken the red one sto be - but that does not mean to say I have not changed parameters on those red ones , because I have and its been fine with me ... (so far! LOL)

shrlnm, 2023-03-06:
> To be honest, i didn't get what you mean :)

No problem. Sorry for my bad english. I'll try to explain in other words.
Changing anything thru ddt database (especially such complex  functions as start/stop) is not a right way because this function involve many parameters in many ECUs and you have to know all dependecies for doing right changes.
The DDT database is used by the manufacturer's engineers to create configuration templates, which are then used to configure the ECUs for a specific car (VIN) with an individual set of options. A set of options for a particular car is described by a set of MTC parameters. Based on MTC parameters and configuration templates, configurations are created to configure specific blocks. This procedure is done by REPROG software. 
For example, the start/stop function in MTC terms is controlled by two values (STOSTA = function enabled) and (SSTOST= function disabled)
I took an arbitrary VIN of one of the CLIO IV and generated configuration files with these two values. A different configuration was obtained in three ECUs - TDB, BCM and CLIMAT!!!???))). It is strange but Humidity sensor parameters somehow depends on start/stop )).
The main difference is offcourse in BCM. For the VIN  that I chose for the experiment, the following parameters received different values:

END_CLUTCH_USED_FOR_AUTOSTART_CF
UNDERHOOD_CF
E_B_STOPSTARTINHIBIT_CF
E_B_SOFINHIBITION_CF

In your case it may be different set of parameters and it why I mentioned your VIN in my first message

Trickykid1978, 2023-03-06:
> atabase (especially such complex functions as start/stop) is not a right way because this function involve many parameters in many ECUs and you have to know all dependecies for doing right changes. [...]

Hi Shrlnm and many thanks for your input ...  i understand what you mean . It makes sense that start stop is connected with clima as the operation of clima, might delay/postpone or even inhibit the start/stop function (meaning that if you have ambient temperature of 35 degrees and you set clima at 20 degrees, the compressor will be working at 100% thus inhibiting the SS function to operate until desired temperature is reached or almost reached) ... on the other hand, i have seen a lot of users canceling the SS without having (or without reporting) any issues or abnormalities on other systems... anyway my VIN , in case you can source sth more specific for my car,  is VF15RBJ0D57602094 ...also, please do check the above three possible solutions and let me know which one appears more harmless than the others (for some reason, my first choice would be solution no 3 (through Injection ECU) :)

shrlnm, 2023-03-06:
@Trickykid1978  For your VIN the difference is also in three ECUs

For disabling s/s

Cluster_X98_X87_X82_xHA_X98ph2_X87ph2_v7.4_20160112T110528.xml
    StopStartPresent_CF = 0

BCM_T4_VS_DDT_22_20160531T140853.xml
    END_CLUTCH_USED_FOR_AUTOSTART_CF = 0
    UNDERHOOD_CF = 0
    E_B_STOPSTARTINHIBIT_CF = 1
    E_B_SOFINHIBITION_CF = 1

ClimBoxGen3-v6.3_20161121T160703.xml
    CL_WithHumiditySensor = 1

Trickykid1978, 2023-03-06:
@shrlnm  

The cluster_X98 etc refers to S/S signals on dashboard. If this is set @ 0 , then no S/S related notifications will be shown at dashboard . If S/S is canceled but this setting is left @ 1 , then the ''S/S not available'' or sth will be reported at dash. 

The UNDERHOOD_CF refers to the ''Open hood'' sensor that disables the S/S anyway as @KarelSvo suggested earlier  (if you drive car with hood open or with this sensor disabled, then car will not allow S/S to operate ...

shrlnm, 2023-03-06:
@Trickykid1978 I don't know exact behaviour of these parameters (. I only know that these parameters depends on present s/s function

Trickykid1978, 2023-03-06:
@shrlnm  well it is a matter of dependencies i think ... as an example : if i cancel s/s through  BCM_T4_VS_DDT_22_20160531T140853.xml > E_B_STOPSTARTINHIBIT_CF = **1**  but leaving the UNDERHOOD_CF = **1** , the S/S will be canceled but the hood sensor will remain alive (but useless in regards to S/S function) . If i set UNDERHOOD_CF = 1, then this sensor will be canceled and this sensor might affect other vehicle operations besides S/S  etc etc (so the dependecies will never end)...  on the other hand, what would happen if i leave  E_B_STOPSTARTINHIBIT_CF = **0**  and set UNDERHOOD_CF = **0**  ?  From what i understand, the boot sensor will be canceled and , as a result, S/S will also be canceled ?

shrlnm, 2023-03-06:
@Trickykid1978 I do not know the exact behavior of all these parameters and their dependencies. But I know that the manufacturer's engineers decided to group these parameters and apply them simultaneously in exactly the same combination as I wrote. They know more than me and I tend to trust them).

Trickykid1978, 2023-03-06:
@shrlnm  yes , you are right but in any case, S/S can be officially deactivated by a physical, Manufacturer's button in the car (which i don't think that once pressed, three different ECU parameters change at once) ... so it is generally possible, acceptable and safe to drive the car without S/S (you just have to press the button everytime you put the ignition ON) .  It would be interesting to see ,once i get connected, if and which parameters change everytime i press the SS button :)

shrlnm, 2023-03-06:
@Trickykid1978 Yes it's really interesting. Please share the result of your experiment.

KarelSvo, 2023-03-06:
> ... so it is generally possible, acceptable and safe to drive the car without S/S (you just have to press the button everytime you put the ignition ON) . 

You can easily reverse the function of the button using a relay.  
[click on this example](https://www-dacianer-de.translate.goog/threads/start-stopp-automatisch-ausschalten-so-gehts.25348/?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp#post-465362)

WhimP990, 2023-03-15:
I am sorry for invaision, but where did you get DDT4ALL Version 5.5.0 with a 2021 database, i cant find it =(

AndyIreland, 2023-03-15:
you can get 5.5.0 with 2019 DB included from here - choose 'all cars exe' 
https://www.drive2.ru/b/498093336985338243/

Trickykid1978, 2023-03-16:
> I am sorry for invaision, but where did you get DDT4ALL Version 5.5.0 with a 2021 database, i cant find it =( I am not afraid to pay for it too.

Hello, i got mine from the link in this  video https://www.youtube.com/watch?v=K9hL-U6oUik  ... but i guess you will source them elsewhere as well if you search : '' Renault DDT2000 Database 11.2021 '' (you just need the ECU folder transferred in DDT4ALL folder).

WhimP990, 2023-03-16:
Thank you for responses! I already downloaded the 5.5 from drive2.ru, the car i am working is renault megane 2022 and it failed to recognise any ecu but uch, hope new database will help!

AndyIreland, 2023-03-16:
with a 2022 Renault car you do realise as well that it will surely have a Security Gateway firewall  that will have to be unlocked with a Renault Service Token as well .. definitely as well if you are looking to change any parameters @WhimP990  - well unless you can bypass the gateway by some way

AndyIreland, 2023-03-16:
> Thank you for responses! I already downloaded the 5.5 from drive2.ru, the car i am working is renault megane 2022 and it failed to recognise any ecu but uch, hope new database will help!
> 
> Found ECU [MIU] (no relevant ECU file found) :DIAGVERSION [08] SUPPLIER [LGE_VIETNA] SOFT [SW08.11] VERSION [SW08.11] Found ECU [ADAS-Sub] (no relevant ECU file found) :DIAGVERSION [04] SUPPLIER [A701] SOFT [R3.11212.3] VERSION [9.00] Found ECU [Navigation] (no relevant ECU file found) :DIAGVERSION [B1] SUPPLIER [BOSCH] SOFT [I_PRJ_RN_AIVI_21] VERSION [5805_211206] Found ECU [Cluster/TDB] (no relevant ECU file found) :DIAGVERSION [50] SUPPLIER [10VisteonPalmela-ICE ] SOFT [i60.14] VERSION [2044] ECU uses SDS 1003
> 
> and i need to connect to the cluster.

if its a larger ECU database you want though I think someone posted up (and I think it was up here on github) 'larger ecu.zip' - but I am damned if I can find it now, typical!

WhimP990, 2023-03-16:
> with a 2022 Renault car you do realise as well that it will surely have a Security Gateway firewall that will have to be unlocked with a Renault Service Token as well .. definitely as well if you [...]

I unlocked it with autel, it said that it was successful. So now i need to config it, hope the 2021/11 database will be enought.

KarelSvo, 2023-03-18:
> Found ECU [Cluster/TDB] (no relevant ECU file found) :DIAGVERSION [50] SUPPLIER [10VisteonPalmela-ICE ] SOFT [i60.14] VERSION [2044]

Cluster_C1A_RUN2_V2.12
Cluster_C1A_RUN2_V2.12_20211105T165123.xml

The identification line in the file has an bug, so it cannot be matched using automatic search.
      <AutoIdent DiagVersion="80" Supplier="10VisteonPalmela-ICE" Soft="V60.14" Version="2044"/>
change to
      <AutoIdent DiagVersion="80" Supplier="10VisteonPalmela-ICE" Soft="iV60.14" Version="2044"/>

WhimP990, 2023-03-18:
Thank you very much, i will try and write here, Thank you!

KarelSvo, 2023-03-19:
> I've noticed that DiagVersion is not the same, does it metter?

hex 50 = decimal 80

WhimP990, 2023-03-19:
Ok, i am noob, so hope you will forgive me. Maybe you know about the security gateway, i need to unlock it once or every time before i try to config something?

KarelSvo, 2023-03-19:
Reports from others say, after opening, the gateway remains open until the control devices have gone back to sleep.

WhimP990, 2023-03-20:
Thank you very much, i am waiting for renolink cable and i will try and let you know!

JABICHOCU, 2023-04-02:
still trying to de-activate  start-strop on my Renault Captur after trying many of the options here with no luck. I was wondering if anyone tried to do it  by raising the minimum percentage of battery needed for it to work (in my case and others its 80) . I found this on emm battery management options>>  C_pct_soczonetravail_min: 80  Do you think this can be changed to a higher number??

Trickykid1978, 2023-04-02:
> still trying to de-activate start-strop on my Renault Captur after trying many of the options here with no luck. I was wondering if anyone tried to do it by raising the minimum percentage of battery [...]

Have you tried Option no3 through Injection ecu ?  It didn't work ? 
[image]

JABICHOCU, 2023-04-03:
I have no such ecu on my car list...these are the ones ddt4all  identified on my car:

Cluster_X98_X87_X82_V2.4_20121001T113250 
T4_BCM_DDT2000_SW11_0_20130128T102714 
EMM_EDISON_DDT2000_SW10_V2_20121220T154401
EMM_EDISON_DDT2000_SW11_1_V1_6_3 
EDC17C42_1701_810_170A_400_V02_20150224T131230  
 EDC17C42_1701_810_170A_400_V02_20150224T131230 
 UPA_Transversal_V2.0_valeo_20121218T161808

AndyIreland, 2023-04-03:
Cluster is your dashboard
UPA is your parking sensors
I dont know why 2 EMM's showed up like that and with different SW - strange that.
I dont know why EDC is showing up 2 times like that , strange again. and not sure what EDC ECU is to be honest off the top of my head.

AndyIreland, 2023-04-03:
on my renault clio mk4 stop start inhibit is here on my BCM UCH module:
[image]

JABICHOCU, 2023-04-03:
I got this list from the "green" modules identified by ddt4all, red ones I discarded so as not to run unnecessary risks. Andy, I have tried to change this last one you posted  but no luck...

AndyIreland, 2023-04-03:
Yes i cannot see any bms setting or stop start entry on your BCM UCH SW11 at all.

Maybe the only thing you can do is take the stop and start wire off your 12v battery then and extinguish the warning light on your cluster dash with DDT4ALL then?

JABICHOCU, 2023-04-05:
That means leaving the battery control module completely useless...don´t think it is right, because the battery needs to be charged correctly.

AndyIreland, 2023-04-05:
yeah possibly, I don't know for certain because my clio mk4 hasn't even got stop/start feature at all on ours so i cannot test. 
Just going by what others have said about removing the (thinner) wire on the 12v battery has disabled it, but that they had the message/light on the dash .... but that can be turned off with DDT4ALL.

is that little wire not just a signal wire for the stop and start system to tell it that the battery is charged enough to activate stop and start? - its not to do with the charging as well is it?
I would have thought the charging up to the battery is done with the large black and red lead to the12v battery because you are talking over 45ah arent you from the alternator, that would be too much to send up that thin wire to the battery.

Trickykid1978, 2023-04-06:
> That means leaving the battery control module completely useless...don´t think it is right, because the battery needs to be charged correctly.

Maybe it would be better to try the ''hood sensor'' solution  https://www.youtube.com/watch?v=rjeImYk8cdo   I haven't done it but it looks safer than disconnecting the bms cable from battery.

shrlnm, 2023-04-06:
Hello @Trickykid1978 
> It would be interesting to see ,once i get connected, if and which parameters change everytime i press the SS button :)
Have you tried to compare parameters?

JABICHOCU, 2023-04-07:
> @Trickykid1978 For your VIN the difference is also in three ECUs
>
> For disabling s/s
>
> Cluster_X98_X87_X82_xHA_X98ph2_X87ph2_v7.4_20160112T110528.xml StopStartPresent_CF = 0 [...]

Hello shrlnm, can you provide a similar ecu list for Captur x87 , please?

shrlnm, 2023-04-07:
@JABICHOCU I need a VIN. The set of parameters may depends on it

JABICHOCU, 2023-04-07:
OK, this is what I get from ddt4all> VF12RFL1H49710343

shrlnm, 2023-04-07:
@JABICHOCU 
```
###################  SSTOST      #########################
------------------------------   Family :  3  :  INSTRUMENT PANEL
SSCU_PRESENT                                            ;0         

------------------------------   Family :  4  :  UCH
UNDERHOOD_CF                                            ;0         
INFOCLUTCH_DIAG_AUTO_DYN_CF                             ;0         
INFOCLUTCH_DIAG_AUTO_STATIC_CF                          ;0         
END_CLUTCH_USED_FOR_AUTOSTART_CF                        ;0         

------------------------------   Family :  8  :  AIR CONDITIONING
CL_WithHumiditySensor                                   ; 2

------------------------------   Family :  20  :  UPC
Nxx_EMM_SAC_CF                                          ;1         
Nxx_HW_FIh_CF 1                                         ;1         
Nxx_HW_FIh_CF 2                                         ;1         
Nxx_HW_FIh_CF 4                                         ;1         
Nxx_HW_FIh_CF 5                                         ;1         
Nxx_HW_FIh_CF 6                                         ;1         
Nxx_HW_FIh_CF 12                                        ;1         
Nxx_StopStart_ESM_FIh_CF 0                              ;1         
Nxx_DiagStopStart_CF 0                                  ;1         
Nxx_DiagStopStart_CF 1                                  ;1         
```

JABICHOCU, 2023-04-07:
OK, and so as to disable S/S what do you suggest,then?

shrlnm, 2023-04-07:
@JABICHOCU all these parameters  are for turning S/S off

AndyIreland, 2023-04-07:
well have you got the BCM_T4_VS_DDT_22 bcm uch module in your car @JABICHOCU  - set those parameters like that if you have - if not then try removing the connector off the battery or that lead by the bonnet catch and see what happens

JABICHOCU, 2023-04-07:
OK...thanks again for your help

shrlnm, 2023-04-07:
@JABICHOCU  these modules are installed in your car

```
#########   Family :  2  :  ABS
ABSESC___X10527998_Version_2.5.xml

#########   Family :  3  :  INSTRUMENT PANEL
Cluster_X98_X87_X82_V2.4_20121001T113250.xml

#########   Family :  4  :  UCH
T4_BCM_DDT2000_SW11_0.xml

#########   Family :  5  :  AIRBAG / PRETENSIONERS
RSAT4_ACU_eng_v11_20120925T103840.xml

#########   Family :  8  :  AIR CONDITIONING
X10_X98_X87_X52ClimBoxGen1_draft_C5_20110913T164836.xml

#########   Family :  18  :  POWER ASSISTED STEERING
X98_EPS_DDT2000_C1_020_b.02_20120329T150729.xml

#########   Family :  20  :  UPC
T4_EMM_DDT2000_SW8.4_V1.xml

#########   Family :  41  :  PARKING PROX CAN
UPA_Transversal_V2.0_valeo_20121218T161808.xml

#########   Family :  76  :  RADIO
ULC_v2.6.xml

#########   Family :  99  :  HANDS-FREE MODULE
T4_HFM_DDT2000_SW4_1_5.xml

```

JABICHOCU, 2023-04-07:
:) That´s more like it for newbies like me...thanx again!

shrlnm, 2023-04-07:
It's just for not to guess if ECU scanning was done well ))

Trickykid1978, 2023-04-07:
> Hello @Trickykid1978
> 
> > It would be interesting to see ,once i get connected, if and which parameters change everytime i press the SS button :)
> 
> Have you tried to compare parameters?

Hi, will do so this week and revert

JABICHOCU, 2023-04-30:
Well, I tried this option on the same Ecu ,which doesn´t appear on DDT4ALL as listed for my Captur. But I managed to find it on Ecutweker...although again, without no success. I click on "write" but an error message appears
[image]

Trickykid1978, 2023-04-30:
> T4ALL as listed for my Captur. But I managed to find it on Ecutweaker...although again, without no success. I click on "write" but an error message appears

Well, the injection ECU was not appearing in my scanned list of ecus either ... so in ddt4all, i just found it with car model drop down menu (i knew it was sth like EDC17...etc.) . I then loaded it up and noticed that it was a live ecu with data etc ... So i would suggest trying again with ddt4all.

Have you made any other changes with ECUtweaker ? i mean, are you certain it can edit parameters or is it working in reading mode only ?

[13 more replies]

closed 2023-06-02 by Furtif