cedricp/ddt4all #789: Renault Zoe Q210 Enable Rear camera and Rear parking assist

HristiyanKolevZoe, 2023-03-13:
Hello,
I own Renault Zoe Q210 with installed aftermarket Rear Parking Assist (with working/beeping sensors) and also aftermarket Rear Camera. How can I enable them in ddt4all, where I can find the settings? 
I want to switch the screen to this (like on the image) when I set reverse.
[image]

AndyIreland, 2023-03-13:
Thats strange did you get a model of Q210 with a rlink screen but without UPA sensors and reverse camera? - was it a ZEN model with the perfume in dash and air purifier and light interior?

AndyIreland, 2023-03-13:
first things first you need a specially wired OBD2 cable so that you can read and make changes through the car's OBD2 socket

HK-Zoe, 2023-03-13:
Thank for the fast answer.
Yes my Q210 is with R-Link screen (attached picture). Its not with light interior. I don't have the perfume in dash.
I already have the specially wired OBD2 cable. I only need to find the actual database for ddt4all and make the settings.
[image]

AndyIreland, 2023-03-13:
ah right - thats  good - download DDT4ALL from here and it has the 2019 database built in and a shortcut for the desktop: 
select 'all cars.exe'
good luck:
[https://www.drive2.ru/b/498093336985338243/](url)

HK-Zoe, 2023-03-16:
For my luck I just broke my elm327 with USB. The seller from who I buy it in ebay does not sell them anymore. From where I can buy new one that ships to Bulgaria? Any advices?

AndyIreland, 2023-03-16:
just do a search for one with a pic18f25k80 chip in it - how is your usb one broke?

HK-Zoe, 2023-03-20:
Everything is fine exept the camera. I have to replace it.
[Will Option D from this will work?](https://www.aliexpress.com/item/1005001602285189.html)

HK-Zoe, 2023-05-13:
Is there any way to make the rear camera full screen when I go on R?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

vicentmartinezmartinez, 2024-08-08:
> Hola, tengo Renault Zoe Q210 con asistente de estacionamiento trasero del mercado de accesorios instalado (con sensores de funcionamiento / pitidos) y también cámara trasera del mercado de accesorios. ¿Cómo puedo habilitarlos en ddt4all, donde puedo encontrar la configuración? Quiero cambiar la pantalla a esto (como en la imagen) cuando configuro la marcha atrás. [image]
YO ACABO DE CONSEGUIRLO. TIENES QUE MODIFICAR LOS PINES DEL OBD2.

[image]

vicentmartinezmartinez, 2024-08-10:
Acabo de activar la cámara con ddt4all.
El asistente para el aparcamiento funciona con sensores o  lo hace  con la cámara?
Tengo que comprar sensores?

AndyIreland, 2024-08-10:
> I just activated the camera with ddt4all. Does the parking assistant work with sensors or with the camera? Do I have to buy sensors?

Parking assist needs ultrasonic sensors in the bumper/fender to work

hgroupetech, 2025-08-11:
Does this work to fully disable the parking sensors ?