cedricp/ddt4all #791: unable to connect DDT4all correctly

Chnord59, 2023-03-17:
Hy guys,

I try to connect DDT4ALL (last release) under windows 10 in BT mode with an ELM 327 CHIP PIC18F25K80). It seems to work but when I try to write in expert mode, nothing happens. The messages are : 

Initialisation ELM a la vitesse 115200...
Tentative d'ouverture du portCOM7 @ 115200
V├®rfication vitesse de port: 38400
Vitesse port communication : 38400
1581 ECU charg├®es en base de donn├®e.
RadNav v1.5 Audio
QLayout: Cannot add null widget to QHBoxLayout/
QLayout: Cannot add null widget to QHBoxLayout/

and in the window at the bottom of the screen :

UCE utilise diagnostic session 1003
Envoie requête :StartDiagnosticSession.Extended
Delay 0 ms
Changement de mode session 10 03
Envoie requête :DataWrite.RadNav Configuration 1.1
Requête abandonée (verifiez les parametres en rouges): NO DATA

Any idea to fix that problem ?

Thank you

AndyIreland, 2023-03-17:
what year is your Renault car - could it be that the OBD port has security gateway firewall - the later Renaults all have gateway so you cannot change parameters until you unlock gateway with token

Chnord59, 2023-03-17:
26/12/2016, megane gt 1,6 205 hp
 
----------------De : Andy Robus ***@***.***>Envoyé : vendredi 17 mars 2023 16:00À : cedricp/ddt4all ***@***.***>Cc : Chnord59 ***@***.***>; Author ***@***.***>Objet : Re: [cedricp/ddt4all] unable to connect DDT4all correctly (Issue #791)
what year is your Renault car - could it be that the OBD port has security gateway firewall - the later Renaults all have gateway so you cannot change parameters until you unlock gateway with token

AndyIreland, 2023-03-17:
ah right I should not think 2016 car blocked by security gateway then - did you do a search for all ECU's in your car with the DDT4ALL magnifying glass first yes? 
You could also try increasing the CAN timeout to 600ms too and see if that helps

Chnord59, 2023-03-17:
I will try 
How to do to increase the CAN timeout ?

AndyIreland, 2023-03-17:
[image]

Chnord59, 2023-03-17:
Ok thank you
I did it
Here are the results

first test with 0 ms

Initialisation ELM a la vitesse 115200...
Tentative d'ouverture du portCOM7 @ 115200
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
1581 ECU charg├®es en base de donn├®e.
Warning, address 5D is not mapped
Scanning ECU INSULATOR
Scanning ECU ACC
Scanning ECU IDM
Scanning ECU BCM/UCH
Scanning ECU UPC-EMM
Scanning ECU 4WS
Scanning ECU CSHV
Scanning ECU HVAC
Warning, address 404 is not mapped
Warning, address 403 is not mapped
Warning, address 401 is not mapped
Warning, address 400 is not mapped
Warning, address 408 is not mapped
Warning, address 40A is not mapped
Scanning ECU Audio
Skipping CAN extended address (not supported yet)  2D
Scanning ECU UBP
Scanning ECU R RE
Scanning ECU ADP
Scanning ECU FRRAD
Scanning ECU Airbag
Scanning ECU Additional Heater
Scanning ECU MIU
Scanning ECU Navigation
Skipping CAN extended address (not supported yet)  16
Scanning ECU Cluster/TDB
Scanning ECU Tacho
Scanning ECU DLOCK
Scanning ECU 4WD
Scanning ECU TCU
Scanning ECU WCGS
Scanning ECU CAN Adapter
Scanning ECU VSP
Scanning ECU AAU
Scanning ECU C Box
Scanning ECU Suspension pilotee
Scanning ECU ABS/ESC
Scanning ECU EMCU
Scanning ECU HLS
Scanning ECU Power steering
Scanning ECU TPMS
Scanning ECU MC HEV FSCM
Scanning ECU EPS
Scanning ECU SVS
Scanning ECU SU
Scanning ECU SCRCM
Scanning ECU SCCM
Skipping CAN extended address (not supported yet)  E1
Skipping CAN extended address (not supported yet)  E0
Skipping CAN extended address (not supported yet)  E3
Skipping CAN extended address (not supported yet)  E2
Warning, address EE is not mapped
Scanning ECU Audio
Scanning ECU TCASE
Scanning ECU TPAD
Scanning ECU HU
Scanning ECU HFCK
Scanning ECU Frein de parking electrique
[... 131 lines cut ...]
Skipping CAN extended address (not supported yet)  18
Skipping CAN extended address (not supported yet)  DF
Warning, address DD is not mapped
Scanning ECU BCB-OBC_29b
Skipping CAN extended address (not supported yet)  DB
Warning, address DC is not mapped
Skipping CAN extended address (not supported yet)  DA
Skipping CAN extended address (not supported yet)  7B
Scanning ECU ECM/Injection
Scanning ECU Engineering
Scanning ECU SR
Scanning ECU LKS
Skipping CAN extended address (not supported yet)  41
Scanning ECU DDCM
Scanning ECU PBD
Scanning ECU LBC2
Scanning ECU DCM Renault
Skipping CAN extended address (not supported yet)  73
Scanning ECU SCU
Scanning ECU HSG - LNG
Scanning ECU DCM
Warning, address 48 is not mapped
Skipping CAN extended address (not supported yet)  D6
Skipping CAN extended address (not supported yet)  D2
Scanning ECU UDM/UPCU

AndyIreland, 2023-03-17:
what parameters are you specifically trying to change on your car? 

does your ECUS show up ok here after searching with the magnifying glass? 

[image]

and are the read parameters in red or not?

Chnord59, 2023-03-17:
[image]
I select

Chnord59, 2023-03-17:
i select XFB MEGANE IV
I try to activate the Race mode

Chnord59, 2023-03-17:
Is that the good way to select the car ?

AndyIreland, 2023-03-17:
no, did you choose that radnav v1.5 manually? - i would not do that. all those read parameters are in red - I would not try changing any parameters where the read in is red! - you could end up bricking ECU if you not careful

AndyIreland, 2023-03-17:
Plus have you got a specially wired up obd2 cable or rewired ELM327 cable so that you can read radio/nav through the OBD2 socket of the car???

Chnord59, 2023-03-17:
Yes I choose manually
I follow a specific tutorial on the french website "au bon coin du Rlink"
I have no specific cable

KarelSvo, 2023-03-17:
Show the files elm_ddt and ecu_ddt.
Show the link of the used ddt4all version.
Radnav transmits on Vehicle Can. No adapter cable necessary.

Chnord59, 2023-03-17:
where are these two files ?

Chnord59, 2023-03-17:
I've no files with ddt extension

Chnord59, 2023-03-17:
I have elm.py and elm.pyc and ecu.py and ecu.pyc

Chnord59, 2023-03-17:
[elm_ecu.zip](https://github.com/cedricp/ddt4all/files/11004592/elm_ecu.zip)

KarelSvo, 2023-03-17:
ddt4all/logs/elm_ddt
ddt4all/logs/ecu_ddt

Chnord59, 2023-03-17:
[logs.zip](https://github.com/cedricp/ddt4all/files/11004834/logs.zip)

Chnord59, 2023-03-17:
good version is here, don't use the previous one
[logs.zip](https://github.com/cedricp/ddt4all/files/11004913/logs.zip)

KarelSvo, 2023-03-17:
Your Elm interface is not suitable for this software.

Chnord59, 2023-03-17:
which one can I use ? Where is the error in the files ?
Thank you

AndyIreland, 2023-03-17:
you have a bluetooth ELM 327 OBD2 with CHIP PIC18F25K80 ? - I would have thought it should be compatible. - unless its got PIC18F25K80 written on the outside or in the listing, but really has not got a PIC18F25K80 - is there any way you can crack open the case and see if there really is a PIC18F25K80 chip on the main circuit board?

KarelSvo, 2023-03-17:
`############################################################
#[03/17/23 17:31:21.131] Init CAN
############################################################
>[17:31:21.131]AT WS
<[0.038]AT WS
?

>
>[17:31:21.169]AT E1
<[0.032]AT E1
?

>
>[17:31:21.201]AT S0
<[0.046]AT S0
?

>
>[17:31:21.247]AT H0
<[0.054]AT H0
?

>
>[17:31:21.301]AT L0
<[0.031]AT L0
?

>
>[17:31:21.332]AT AL
<[0.016]AT AL
?

>
>[17:31:21.348]AT CAF0
<[0.053]AT CAF0
?
`
The question marks are a negative answer to the commands for configuring your Elm interface.
Which country are you from and which sales platforms do you prefer?
Suitable Chinese Elm are becoming increasingly difficult to find.  
http://cvtz50.info/en/elm327/
Interfaces with original firmware, such as those from the Obdlink series, are the safest.

Chnord59, 2023-03-17:
Thank you for your help. Is it possible for you to morrow in the morning ?

KarelSvo, 2023-03-17:
> To go on : here are the errors when I start test ELM

`>[16:19:11.835]ATZ
<[0.914]ATZ
ü

OBDII  v1.5
`
Your Elm sends a non-ASCII compliant character (ü) in connection with the ATZ command. Therefore the test aborts.
You can find the list of commands for testing in the elm.py file after the line cmdb = '''.
In order for the test to work with your Elm, you must delete the ATZ test lines and then save the file.

AndyIreland, 2023-03-17:
> Thank you for your help. Is it possible for you to morrow in the morning ?

tomorrow in the afternoon after 2pm UK time I can do if any good?

Chnord59, 2023-03-17:
> 

I've just opened it, it strongly difficult to read but it's  a PIC 18F25K80

Chnord59, 2023-03-17:
> > Thank you for your help. Is it possible for you to morrow in the morning ?
> 
> tomorrow in the afternoon after 2pm UK time I can do if any good?

OK so 3 pm in France

Chnord59, 2023-03-17:
> 
Thank you for your help ! I've just opened the ELM to see the chip inside : it's a PIC18F25K80 like written on the box.
I'll see to morrow with Andy Ireland who will take control of my computer while connecting to the ECU.

AndyIreland, 2023-03-17:
Yes thats right - if you need me to have a look just let me know. I might not be of any help though, especially if it is an incompatible ELM327 . It is bluetooth adaptor is it?

Chnord59, 2023-03-17:
yes bluetooth, and I use Windows 10
The car is a MEGANE year 2016 model 1.6 GT 205 HP

Chnord59, 2023-03-17:
Does I need to install PC anywhere ?

AndyIreland, 2023-03-17:
anydesk from anydesk .com if OK

Chnord59, 2023-03-17:
it's done

Chnord59, 2023-03-17:
> > I've just opened the ELM to see the chip inside : it's a PIC18F25K80 like written on the box.
> 
> I had given you a link where you can see pictures with fake unsuitable PIC18F25K80. The Chinese also fake the writing on the microcontrollers.

These chinese...

AndyIreland, 2023-03-17:
amazing they go to all those lengths

Chnord59, 2023-03-17:
I just think that i won't be able to connect my PC to the wifi outside of my house.
So I will use a phone to create an access in order to connect the PC, but it won't be so fast (4G)

AndyIreland, 2023-03-17:
> I just think that i won't be able to connect my PC to the wifi outside of my house. So I will use a phone to create an access in order to connect the PC, but it won't be so fast (4G)

ah it might be ok tethered to your phone. We shall see, if its too bad connection we might  have to abort it. but I have done it before with others that have tethered to 4G and it werent too bad

KarelSvo, 2023-03-17:
> These chinese...

give me your elm.py file to edit.

Chnord59, 2023-03-17:
> > These chinese...
> 
> give me your elm.py file to edit.

[elm.zip](https://github.com/cedricp/ddt4all/files/11006268/elm.zip)

Chnord59, 2023-03-17:
> ah it might be ok tethered to your phone. We shall see, if its too bad connection we might have to abort it. but I have done it before with others that have tethered to 4G and it werent too bad

> ah it might be ok tethered to your phone. We shall see, if its too bad connection we might have to abort it. but I have done it before with others that have tethered to 4G and it werent too bad

if it doesn't work I'll buy a 30 meters RJ 45 to connect to the box !!

AndyIreland, 2023-03-17:
thats another way yep

KarelSvo, 2023-03-17:
[elm.zip](https://github.com/cedricp/ddt4all/files/11006312/elm.zip)

Replace file and run Elm test.

Chnord59, 2023-03-17:
> 

No change...

Chnord59, 2023-03-17:
> > 
> 
> No change...

Tentative d'ouverture du portCOM7 @ 115200
V├®rfication vitesse de port: 38400
Vitesse port communication : 38400
 ATWS                   ATWS
?

>
 ATE1                   ATE1
?

>
 ATL1                   ATL1
?

>
 ATL0                   ATL0
?

>
 ATI                    ATI
?

>
 AT@1                   AT@1
?

>
 ATRV                   ATRV
?

>
 ATPPS                  ATPPS
?

>
 ATS1                   ATS1
?

>
 ATS0                   ATS0
?

>
 ATCAF0                 ATCAF0
?

>
 ATCFC0                 ATCFC0
?

>
 ATH0                   ATH0
?

>
 ATR1                   ATR1
?

>
 ATKW0                  ATKW0
?

>
 ATD1                   ATD1
?

>
 ATD0                   ATD0
?

>
 ATAL                   ATAL
?

>
 ATSP 3                 ATSP 3
?

>
 ATSP 4                 ATSP 4
?

>
 ATIIA 01               ATIIA 01
?

>
 ATSI                   ATSI
?

>
 ATSP 5                 ATSP 5
?

>
 ATFI                   ATFI
?

>
 ATSP 6                 ATSP 6
?

>
 ATSP 7                 ATSP 7
?

>
 ATSP 8                 ATSP 8
?

>
 ATPC                   ATPC
?

>
 ATSH 012               ATSH 012
?

>
 ATSW 96                ATSW 96
?

>
 ATSW 34                ATSW 34
?

>
 ATWM 817AF13E          ATWM 817AF13E
?

>
 ATIB 10                ATIB 10
?

>
 ATST FF                ATST FF
?

>
 ATCRA 012              ATCRA 012
?

>
 ATCRA 01234567         ATCRA 01234567
?

>
 ATFC SH 012            ATFC SH 012
?

>
 ATFC SH 00112233       ATFC SH 00112233
?

>
 ATFC SD 300000         ATFC SD 300000
?

>
 ATFC SM 1              ATFC SM 1
?

>
ATAT0                   ATAT0
OK

>
ATAT1                   ATAT1
OK

>
 ATCM 123               ATCM 123
?

>
 ATCM 12345678          ATCM 12345678
?

>
 ATCF 123               ATCF 123
?

>
 ATCF 12345678          ATCF 12345678
?

>

Chnord59, 2023-03-17:
v1.0 ATWS [ECHEC]
v1.0 ATE1 [ECHEC]
v1.0 ATL1 [ECHEC]
v1.0 ATL0 [ECHEC]
v1.0 ATI [ECHEC]
v1.0 AT@1 [ECHEC]
v1.0 ATRV [ECHEC]
v1.1 ATPPS [ECHEC]
v1.3 ATS1 [ECHEC]
v1.3 ATS0 [ECHEC]
v1.0 ATCAF0 [ECHEC]
v1.0 ATCFC0 [ECHEC]
v1.0 ATH0 [ECHEC]
v1.0 ATR1 [ECHEC]
v1.2 ATKW0 [ECHEC]
v1.3 ATD1 [ECHEC]
v1.3 ATD0 [ECHEC]
v1.0 ATAL [ECHEC]
V1.0 ATSP 3 [ECHEC]
V1.0 ATSP 4 [ECHEC]
v1.2 ATIIA 01 [ECHEC]
v1.4 ATSI [ECHEC]
V1.0 ATSP 5 [ECHEC]
v1.4 ATFI [ECHEC]
V1.0 ATSP 6 [ECHEC]
V1.0 ATSP 7 [ECHEC]
V1.0 ATSP 8 [ECHEC]
v1.0 ATPC [ECHEC]
v1.0 ATSH 012 [ECHEC]
v1.0 ATSW 96 [ECHEC]
v1.0 ATSW 34 [ECHEC]
v1.0 ATWM 817AF13E [ECHEC]
v1.0 ATIB 10 [ECHEC]
v1.0 ATST FF [ECHEC]
v1.3 ATCRA 012 [ECHEC]
v1.3 ATCRA 01234567 [ECHEC]
v1.1 ATFC SH 012 [ECHEC]
v1.1 ATFC SH 00112233 [ECHEC]
v1.1 ATFC SD 300000 [ECHEC]
v1.1 ATFC SM 1 [ECHEC]
v1.2 ATAT0 [OK]
v1.2 ATAT1 [OK]
v1.0 ATCM 123 [ECHEC]
v1.0 ATCM 12345678 [ECHEC]
v1.0 ATCF 123 [ECHEC]
v1.0 ATCF 12345678 [ECHEC]
Adaptateur ARM incompatible 
Résultat: 2 réussis sur 46

For Media CAN (CAN Line 2) on STN1170, ELS27: Connect OBD pin 3 with 13 and 11 with 12.

Chnord59, 2023-03-17:
the ELM seems to be incompatible

AndyIreland, 2023-03-17:
Tomorrow I will try PyREN in DDT4ALL mode - it possible it may connect? - I dunno for sure

KarelSvo, 2023-03-17:
definitely not with this Elm

AndyIreland, 2023-03-17:
ah right - thank you @KarelSvo - it was just a thought

Chnord59, 2023-03-17:
Thank you both of you
I will try a new ELM asap
Great job !!

closed 2023-03-17 by Chnord59

KarelSvo, 2023-03-17:
https://www.ebay.fr/itm/374104232812

Chnord59, 2023-03-17:
So it's done 
Thank you for the tip 
I just wait for the postman to drop it !!
 
----------------De : KarelSvo ***@***.***>Envoyé : vendredi 17 mars 2023 22:53À : cedricp/ddt4all ***@***.***>Cc : Chnord59 ***@***.***>; State change ***@***.***>Objet : Re: [cedricp/ddt4all] unable to connect DDT4all correctly (Issue #791)
https://www.ebay.fr/itm/374104232812

AndyIreland, 2023-03-17:
you most probably wont need my remote help if you get one that connects @Chnord59  - but just in case you do here is a heads up that I would not be able to do it if you go for the wifi flavour OBD2 , only bluetooth or USB obd2 because you cannot connect the wifi bluetooth adaptor to widows laptop and internet at the same time I think? - I am not sure if you can connect to internet with LAN Ethernet and use wifi elm327 though? ... possibly, have never tried it.

Chnord59, 2023-03-17:
Well I hope it will work 
As you say, it will be impossible to be connected at the same time with two wifi..
 
----------------De : Andy Robus ***@***.***>Envoyé : vendredi 17 mars 2023 23:17À : cedricp/ddt4all ***@***.***>Cc : Chnord59 ***@***.***>; Mention ***@***.***>Objet : Re: [cedricp/ddt4all] unable to connect DDT4all correctly (Issue #791)
you most probably wont need my remote help if you get one that connects @Chnord59  - but just in case you do here is a heads up that I would not be able to do it if you go for the wifi flavour OBD2 , only bluetooth or USB obd2 because you cannot connect the wifi bluetooth adaptor to widows laptop and internet at the same time I think? - I am not sure if you can connect to internet with LAN Ethernet and use wifi elm327 though? ... possibly, have never tried it.

juancarlos002o, 2024-04-28:
> [image] I select

hellow  i have the same problem in the interface is appear no data  do you have any solution?

KarelSvo, 2024-04-28:
> > [image] I select
> 
> hellow i have the same problem in the interface is appear no data do you have any solution?

The files from the ddt4all/logs folder are missing for assessment.

AndyIreland, 2024-04-28:
> > [image] I select
> 
> hellow i have the same problem in the interface is appear no data do you have any solution?

Try increasing CAN Timeout to 600ms then hit refresh button, if that dont work then it could be wrong ECU selected or incompatible ELM327