cedricp/ddt4all #792: How to disable 4x4 mode for a MediaNav with DDT4All

MichaelPHolstein, 2023-03-19:
Good day, I am in possession of a Renault vehicle, and it appears that the 4x4 mode is currently enabled. This mode is designed specifically for certain Dacia models, and I would like to disable it.

Unfortunately, I have been unable to locate the appropriate setting to disable this mode. I can confirm that the ECU location for my MediaNav system is audio/ULC_gen4_v.1.4.

Would anyone happen to know the location of the setting to disable the 4x4 mode in ddt4all?

https://upload.forumfree.net/i/ff12373994/6B238B01_AAA0_442E_80B8_9E30A44F657E.jpeg

KarelSvo, 2023-03-19:
ECU Config2.ROAD

AndyIreland, 2023-03-19:
could you maybe disable using MICOM instead - I dont know if your medianav is medianav 1st edition or medianav evolution but here is my one - medianav evolution 9.1.3: 

[image]

https://youtu.be/c6Fdfhvcb9E

MichaelPHolstein, 2023-03-19:
> DataWrite.ECU Config2 ECU Config2.ROAD

Thank you this solves the issue!

closed 2023-03-19 by MichaelPHolstein