cedricp/ddt4all #793: Is there an option within DDT4all to enable Steering wheel mounted radio controls?

MichaelPHolstein, 2023-03-19:
Dear all,

I am the owner of a Renault Clio and I am experiencing an issue with the radio controls on my steering wheel. As shown in the attached image, the controls are as follows:

[image]

I previously had a MediaNav Evolition (V1) installed and the controls worked perfectly. However, since configuring a MediaNav Evolution V2, the MediaNav does not respond to the radio controls. I would like to inquire if there is an option within the ECU audio/ULC_gen4_v.1. to enable these controls.

Thank you in advance for your assistance.

AndyIreland, 2023-03-19:
try changing these settings swc:

[image]

KarelSvo, 2023-03-19:
DataWrite.ECU Config2
ECU Config2.SWRC2 type
No SWRC2

DataWrite.ECU Config
ECU Config.SWRC type
Type 2: Without Mute (TypreX52)

MichaelPHolstein, 2023-03-23:
Thank you! This solved the problem. I'll close the issue now

closed 2023-03-23 by MichaelPHolstein