cedricp/ddt4all #794: [Renault Clio IV ph2] Activate Android and Carplay

mkhaliid, 2023-03-21:
Hi,

I tried to activate Android and Carplay on my Clio IV ph2 R-Link but I cannot find relevant lines to activate.

I have DDT4ALL version 5.5 (07.2019) when I go under NAVIGATION I find only MFD4.3 and according to my understanding Carplay and android options can only be activated from MFD4.6...

On standard Clio IV I can find MFD5.3 but the writing does not work since mine is ph2.

Can someone help? What needs to be done in order to get MFD4.6 or newer for CLIO IV ph2?

Thanks alot

AndyIreland, 2023-03-21:
I wonder what radionav the ph2 has - I know the ph1 might have rlink1 or LG medianav. rlink1 you will only be able to activate Android Auto, it does not do carplay. 
Rlink2 should have AA and CP (or be able to be enabled using DDT4ALL) - Medianav only 2019 onwards done CP and AA. 

In any case you need a specially wired OBD2 adaptor or specially soldered ELM327 so that you can read/write parameters to the rlink or medianav through the car's OBD2 socket (I presume you have done this?)
MFD5.3 is for rlink1 radionav
What radio ECU's show up when you do a search in DDT4ALL with the magnifying glass?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif