cedricp/ddt4all #796: Issue reading ECU (Audio->ULC_gen4_v.1.4)

JakubUrbanowski, 2023-03-23:
Dear all,

I am experiencing issues with configuring the ULC_gen4_v.1.4 ECU. Currently, I have a MediaNav Evolution 2 where I can access the ECU configuration without any issues.

I have acquired two other MediaNav systems which are still virgin and need to be configured. When I remove my own system from my Clio and replace it with one of the two new (virgin) models, I receive the following message when attempting to retrieve the ECU configuration:

```py
Trying to open port /dev/cu.usbserial-12345678 @ 500000
Serial<id=0x7fc5a83cd250, open=True>(port='/dev/cu.usbserial-12345678', baudrate=500000, bytesize=8, parity='N', stopbits=1, timeout=5, xonxoff=False, rtscts=False, dsrdtr=False)
2038 ECU's loaded in database.
Traceback (most recent call last):
  File "/Users/Jakub/ddt4all/main.py", line 976, in changeScreen
    inited = self.paramview.init(screen, self.screenlogfile)
  File /Users/Jakub/ddt4all/parameters.py", line 628, in init
    scr_init = self.initScreen(screen)
  File "/Users/Jakub/ddt4all/parameters.py", line 988, in initScreen
    self.updateDisplays(True)
File "/Users/Jakub/ddt4all/parameters.py", line 1299, in updateDisplays
    self.updateDisplay(request_name, update_inputs)
  File "/Users/Jakub/ddt4all/parameters.py", line 1342, in updateDisplay
    value = ecu_data.getDisplayValue(elm_response, data_item, request.ecu_file.endianness)
  File "/Users/Jakub/ddt4all/ecu.py", line 667, in getDisplayValue
    return bytes.fromhex(value).decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
```

Retrieving other components such as tuning parameters and SPVR parameters works perfectly fine on these models.

Does anyone have an idea why I am able to retrieve the ECU configuration on my existing MediaNav MN3, but not on the virgin models? I have verified that the type numbers on the labels are identical to the model where I am able to log in.

Could it be a security or log file issue within DDT4ALL or something similar?

Additional info: I am using a MacBook Pro with M1 chip running macOS Monterey.

Thank you in advance for any responses.

Best,

Jakub

shrlnm, 2023-03-24:
@JakubUrbanowski 
try to edit the code a bit
open the file ddt4all/ecu.py  and edit the string 667
from
`return bytes.fromhex(value).decode('utf-8')`
replace on 
`return bytes.fromhex(value).decode('utf-8',errors="ignore")
`
Keep the number of spaces in the beginning of line

JakubUrbanowski, 2023-03-24:
Thank you! This solved the problem! I will close the issue. Strange that this error didn't occur on my current media nav.

closed 2023-03-24 by JakubUrbanowski